FreeOpcUa/opcua-asyncio #1045: create_session fails if server sends certificate chain

eikegu, 2022-09-21:
**Describe the bug** <br />
When the server sends the certificate chain with its certificate, these lines in create_session are triggered and raise an exception:
```
        elif self.security_policy.peer_certificate != response.ServerCertificate:
            raise ua.UaError("Server certificate mismatch")
```

**To Reproduce**<br />
Steps to reproduce the behavior incl code:
Try `await client.connect()` to a server which sends its certificate chain with the certificate.

Then try this:
```
await client.connect_socket()
await client.send_hello()
await client.open_secure_channel()
peer_cert = client.security_policy.peer_certificate
client.security_policy.peer_certificate = None
await client.create_session()
client.security_policy.peer_certificate = peer_cert
await client.activate_session()
```
This works because it circumvents the check.

**Expected behavior**<br />
create_session should only check against the certificate, not the entire chain.

**Version**<br />
Python-Version: 3.8.10<br />
opcua-asyncio Version (e.g. master branch, 0.9): 0.9.95 from pip.

schroeder-, 2022-09-22:
Which server you are connecting? Because it looks like a server side bug. 
Because as of [specs](https://reference.opcfoundation.org/v104/Core/docs/Part4/5.6.2/) 
the server should return his certificate not a chain:

<html>
<body>

serverCertificate | ApplicationInstanceCertificate | The Application Instance Certificate issued to the Server.A Server shall prove possession by using the private key to sign the Nonce provided by the Client in the request. The Client shall verify that this Certificate is the same as the one it used to create the SecureChannel.The ApplicationInstanceCertificate type is defined in 7.2.If the securityPolicyUri is NONE and none of the UserTokenPolicies requires encryption, the Client shall ignore the ApplicationInstanceCertificate.
-- | -- | --

</body>
</html>

eikegu, 2022-09-23:
Thanks for your reply. It is a proprietary server. I will try to contact the developers. Until then, the workaround described in my initial message works for me.

schroeder-, 2023-03-29:
Fix via #1250 and #1260

closed 2023-03-29 by schroeder-