FreeOpcUa/opcua-asyncio #1058: Serialize/deserialize of generate dataclasses with recursive structs with load_data_type_definitions fails

bitkeeper, 2022-09-29:
Suggestion or direction how to fix the (de)serializer for the case below are more than welcome.

**Version:**
master d2b48b5a76cb874437651c3894db41a22a25f7a2 (Wed Sep 28 15:11:06 2022 +0200)

**Case:**
Given a nodeset that contains resursive constructions. Like typical tree structures.

Example:
```xml
   <UADataType NodeId="ns=1;i=1001" BrowseName="1:MyParameterType">
        <DisplayName>MyParameterType</DisplayName>
        <References>
            <Reference ReferenceType="HasSubtype" IsForward="false">i=22</Reference>
        </References>

        <Definition Name="1:MyParameterType">
            <Field DataType="MyParameterType" ValueRank="1" ArrayDimensions="0" Name="Subparameters"/>
        </Definition>
    </UADataType>
```

Trying to use the dataclasses generated by `server.load_data_type_definitions`, with method calls and method callbacks.

**Test to validate:**
Add the following test to `tests/test_common.py`:
```python
from asyncua.ua.ua_binary import struct_to_binary
async def test_custom_struct_recursive_serialize(opc):
    nodes = await opc.opc.import_xml("tests/custom_struct_recursive.xml")
    await opc.opc.load_data_type_definitions()

    param =ua.MyParameterType()
    bin = struct_to_binary(param)
```

**Problem:**
When MyParameterType is received or send the serialize/deserialize fails due recursive method calls:
```python
  File "d:\work\opcua-asyncio\asyncua\ua\ua_binary.py", line 308, in <listcomp>
    encoding_functions = [(f.name, field_serializer(f.type)) for f in data_fields]
  File "d:\work\opcua-asyncio\asyncua\ua\ua_binary.py", line 262, in field_serializer
    return create_list_serializer(type_from_list(uatype))
  File "d:\work\opcua-asyncio\asyncua\ua\ua_binary.py", line 372, in create_list_serializer
    type_serializer = create_type_serializer(uatype)
  File "d:\work\opcua-asyncio\asyncua\ua\ua_binary.py", line 355, in create_type_serializer
    return create_dataclass_serializer(uatype)
  File "d:\work\opcua-asyncio\asyncua\ua\ua_binary.py", line 308, in create_dataclass_serializer
    encoding_functions = [(f.name, field_serializer(f.type)) for f in data_fields]
  File "d:\work\opcua-asyncio\asyncua\ua\ua_binary.py", line 308, in <listcomp>
    encoding_functions = [(f.name, field_serializer(f.type)) for f in data_fields]
  File "d:\work\opcua-asyncio\asyncua\ua\ua_binary.py", line 262, in field_serializer
    return create_list_serializer(type_from_list(uatype))
  File "d:\work\opcua-asyncio\asyncua\ua\ua_binary.py", line 372, in create_list_serializer
    type_serializer = create_type_serializer(uatype)
  File "d:\work\opcua-asyncio\asyncua\ua\ua_binary.py", line 355, in create_type_serializer
    return create_dataclass_serializer(uatype)
  File "d:\work\opcua-asyncio\asyncua\ua\ua_binary.py", line 308, in create_dataclass_serializer
    encoding_functions = [(f.name, field_serializer(f.type)) for f in data_fields]
  File "d:\work\opcua-asyncio\asyncua\ua\ua_binary.py", line 308, in <listcomp>
```

**Cause:**

The issue for the serializer is located in `ua_binary.py::create_list_serializer`:

```python
functools.lru_cache(maxsize=None)
def create_list_serializer(uatype) -> Callable[[Any], bytes]:
    """
    Given a type, return a function that takes a list of instances
    of that type and serializes it.
    """
    if hasattr(Primitives1, uatype.__name__):
        data_type = getattr(Primitives1, uatype.__name__)
        return data_type.pack_array
    type_serializer = create_type_serializer(uatype)
    none_val = Primitives.Int32.pack(-1)

    def serialize(val):
        if val is None:
            return none_val
        data_size = Primitives.Int32.pack(len(val))
        return data_size + b''.join(type_serializer(el) for el in val)
    return serialize

```

Because the serializer are generated on the fly, it will call resursive `create_type_serializer` of the type of which it is already part.

schroeder-, 2022-09-29:
This one is rly hard. Can't think of getting this working with lru_cache. We would need something like lazy evaluation.

bitkeeper, 2022-09-29:
In my use case (https://raw.githubusercontent.com/OPCFoundation/UA-Nodeset/latest/ISA95-JOBCONTROL/opc.ua.isa95-jobcontrol.nodeset2.xml) the construction isn't even used (list is kept empty).

An improvement already would be if this situation could be detected and the `create_type_serializer` doesn't dive deeper if already has created a serializer for the uatype on the _'stack'_. Or stop after a certain level depth.

That at least would prevent never ending recursive call wich ends with an Exception.

schroeder-, 2022-09-29:
I think i managed something for the encoding, will share something soon.

schroeder-, 2022-09-29:
See #1060 got encoding/decoding working. 
UAExpert looks a bit strange:
[image]
But shows the correct in the edit window.

closed 2022-10-02 by schroeder-