FreeOpcUa/opcua-asyncio #1061: xmlimporter loops infinitely when sorting nodes by parentid, when parentid is not found

gauss345, 2022-09-29:
**Describe the bug** <br />

When setting up a simple `server-minimal.py` example, by importing a nodeset, that is generated by TIA portal, v15.1, SIEMENS/SIMANTIC - s7-1500, the `xmlimporter.py` goes into a infinite loop, as the parent node is not found. 

No output information is given  about this. 

in `python-opcua` ( **not** `opcua-asyncio` ), we get output with the same nodeset that states parent node if does not refer to a valid node

```
    raise UaStatusCodeError(self.value)
opcua.ua.uaerrors._auto.BadParentNodeIdInvalid: "The parent node id does not to refer to a valid node."(BadParentNodeIdInvalid)

```

**To Reproduce**<br />
```
import asyncio
import logging
import os
from asyncua import Server, ua
from asyncua.common.methods import uamethod
import traceback
from datetime import datetime

build_date = datetime(2022, 6, 15, 17, 00)
BASE_DIR = os.path.dirname(os.path.dirname(os.path.abspath(__file__)))

async def main():
    _logger = logging.getLogger("asyncua")
    # setup our server
    server = Server()
    await server.init()
    await server.set_build_info(
        product_uri="https://",
        product_name="test asyncio",
        manufacturer_name="company",
        software_version="alpha",
        build_number="202206151700",
        build_date=build_date,
    )

    server.set_security_policy([
        ua.SecurityPolicyType.NoSecurity,
    ])
    server.set_security_IDs([
        "Anonymous",
    ])
    server.set_endpoint("opc.tcp://0.0.0.0:4840/freeopcua/server/")
    print("Setup done!")

    print(f"current base dir: {BASE_DIR}")

    try:
        nodes1 = await server.import_xml(os.path.join(BASE_DIR, "examples", "nodesets", "custom", "V03_21.00.01.PLC_1.OPCUA_20220329.xml"))
      
    except Exception as e:
        just_the_string = traceback.format_exc()
        print(f" Caught exception loading nodeset: {just_the_string}")

    _logger.info("Starting server!")
    async with server:
        while True:
            await asyncio.sleep(1)

if __name__ == "__main__":
    logging.basicConfig(level=logging.DEBUG)
    logging.getLogger("asyncio").setLevel(logging.DEBUG)
    asyncio.run(main(), debug=True)

```

Unfortunately, due to company policy, I cannot attach the nodeset xml.

```
    def _sort_nodes_by_parentid(self, ndatas):
        """
        Sort the list of nodes according their parent node in order to respect
        the dependency between nodes.

        :param nodes: list of NodeDataObjects
        :returns: list of sorted nodes
        """

        sorted_ndatas = []
        sorted_nodes_ids = set()
        all_node_ids = set(data.nodeid for data in ndatas)
        cnt = 0
        while len(sorted_nodes_ids) < len(ndatas):
            for ndata in ndatas:
                print(f"ndata: {ndata} : {cnt}")
                cnt = cnt + 1
                if ndata.nodeid in sorted_nodes_ids:
                    continue
                elif ndata.parent is None or ndata.parent not in all_node_ids:
                    sorted_ndatas.append(ndata)
                    sorted_nodes_ids.add(ndata.nodeid)
                else:
                    # Check if the nodes parent is already in the list of
                    # inserted nodes
                    if ndata.parent in sorted_nodes_ids:
                        sorted_ndatas.append(ndata)
                        sorted_nodes_ids.add(ndata.nodeid)
        return sorted_ndatas

```

The above method goes into an infinite loop as no `ndata` is added to the `sorted_ndatas` or `sorted_nodes_ids` lists.
Thus the `len(sorted_nodes_ids)` is always less than `len(ndatas)`

**Expected behavior**<br />

I expect the Nodeset to be loaded, and the server be accessible via uaexpert, such that I can browse.

**Screenshots**<br />
[image]
Screenshot of debugging the `xmlimporter.py` and the method as show above.

[image]
so `all_node_ids` has the nodeId's but it does not contain the parent id

[image]
output for `ndata` and the values for the `nodeid` and `parent`

**Version**<br />
Python-Version: python 3.8.13<br />
opcua-asyncio Version:  asyncua==0.9.98

gauss345, 2022-09-29:
```
 <UAObjectType NodeId="ns=3;i=1002" BrowseName="3:SimaticDeviceType">
        <DisplayName>SimaticDeviceType</DisplayName>
        <Description Locale="en-US">This type represents one simatic opcua device.</Description>
        <References>
            <!--Reference ReferenceType="HasSubtype" IsForward="false">ns=2;i=1002</Reference-->
            <Reference ReferenceType="HasComponent">ns=3;i=5063</Reference>
            <Reference ReferenceType="HasComponent">ns=3;i=5064</Reference>
            <Reference ReferenceType="HasComponent">ns=3;i=5062</Reference>
            <Reference ReferenceType="HasComponent">ns=3;i=5061</Reference>
            <Reference ReferenceType="HasComponent">ns=3;i=5059</Reference>
            <Reference ReferenceType="HasComponent">ns=3;i=5060</Reference>
            <Reference ReferenceType="HasComponent">ns=3;i=5065</Reference>
            <Reference ReferenceType="HasProperty">ns=3;i=6070</Reference>
            <Reference ReferenceType="HasProperty">ns=3;i=6071</Reference>
            <Reference ReferenceType="HasProperty">ns=3;i=6072</Reference>
        </References>
    </UAObjectType>
```

uncommenting the reference type , gets the parent node correctly, but then `len(sorted_node_ids)` still less than `len(ndatas)`,
because
```
NFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(Identifier=15957, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>), requested parent node NumericNodeId(Identifier=11715, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(Identifier=15958, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>), requested parent node NumericNodeId(Identifier=15957, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(Identifier=15959, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>), requested parent node NumericNodeId(Identifier=15957, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(Identifier=15960, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>), requested parent node NumericNodeId(Identifier=15957, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(Identifier=15961, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>), requested parent node NumericNodeId(Identifier=15957, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(Identifier=15962, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>), requested parent node NumericNodeId(Identifier=15957, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(Identifier=15963, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>), requested parent node NumericNodeId(Identifier=15957, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(Identifier=15964, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>), requested parent node NumericNodeId(Identifier=15957, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(Identifier=16134, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>), requested parent node NumericNodeId(Identifier=15957, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(Identifier=16135, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>), requested parent node NumericNodeId(Identifier=15957, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>) does not exists
INFO:asyncua.server.address_space:add_node: while adding node NumericNodeId(Identifier=16136, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>), requested parent node NumericNodeId(Identifier=15957, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>) does not exists
DEBUG:asyncio:Close <_UnixSelectorEventLoop running=False closed=False debug=True>

```

the above is not in our nodeset.

schroeder-, 2022-09-30:
Without a nodeset there isn't much I can do to help here. My guess is that ns=2;i=1002 is missing or is not inserted because of another error.

gauss345, 2022-09-30:
> Without a nodeset there isn't much I can do to help here. My guess is that ns=2;i=1002 is missing or is not inserted because of another error.

Thanks for reply, 

So ns=2;i1002 is in the nodeset, and is added into the list, 

[image]

[image]
the `refs`  list in the object `NodeData`, 

but in the `all_node_ids`, the

[image]
the identifier is a string, 

which is then compared to the nodeid as a int

[image]

```
 <UAObjectType NodeId="ns=3;i=1002" BrowseName="3:SimaticDeviceType">
        <DisplayName>SimaticDeviceType</DisplayName>
        <Description Locale="en-US">This type represents one simatic opcua device.</Description>
        <References>
            <!--Reference ReferenceType="HasSubtype" IsForward="false">ns=2;i=1002</Reference-->
            <Reference ReferenceType="HasComponent">ns=3;i=5063</Reference>
            <Reference ReferenceType="HasComponent">ns=3;i=5064</Reference>
            <Reference ReferenceType="HasComponent">ns=3;i=5062</Reference>
            <Reference ReferenceType="HasComponent">ns=3;i=5061</Reference>
            <Reference ReferenceType="HasComponent">ns=3;i=5059</Reference>
            <Reference ReferenceType="HasComponent">ns=3;i=5060</Reference>
            <Reference ReferenceType="HasComponent">ns=3;i=5065</Reference>
            <Reference ReferenceType="HasProperty">ns=3;i=6070</Reference>
            <Reference ReferenceType="HasProperty">ns=3;i=6071</Reference>
            <Reference ReferenceType="HasProperty">ns=3;i=6072</Reference>
        </References>
    </UAObjectType>
<UAObject NodeId="ns=3;s=PLC" BrowseName="3:PLC_1" ParentNodeId="ns=2;i=5001">
        <DisplayName>PLC_1</DisplayName>
        <Description Locale="en-US">The PLC instance which supports you with OPC UA functionality.</Description>
        <References>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=2;i=5001</Reference>
            <Reference ReferenceType="Organizes" IsForward="false">i=85</Reference>

            <Reference ReferenceType="HasTypeDefinition">ns=3;i=1002</Reference>

            <Reference ReferenceType="HasProperty">ns=3;s=DeviceManual</Reference>
            <Reference ReferenceType="HasProperty">ns=3;s=DeviceRevision</Reference>
            <Reference ReferenceType="HasProperty">ns=3;s=EngineeringRevision</Reference>
            <Reference ReferenceType="HasProperty">ns=3;s=HardwareRevision</Reference>
            <Reference ReferenceType="HasProperty">ns=3;s=Manufacturer</Reference>
            <Reference ReferenceType="HasProperty">ns=3;s=Model</Reference>
            <Reference ReferenceType="HasProperty">ns=3;s=OperatingMode</Reference>
            <Reference ReferenceType="HasProperty">ns=3;s=OrderNumber</Reference>
            <Reference ReferenceType="HasProperty">ns=3;s=RevisionCounter</Reference>
            <Reference ReferenceType="HasProperty">ns=3;s=SerialNumber</Reference>
            <Reference ReferenceType="HasProperty">ns=3;s=SoftwareRevision</Reference>
        </References>
    </UAObject>
<UAObject NodeId="ns=3;s=DataBlocksGlobal" BrowseName="3:DataBlocksGlobal" ParentNodeId="ns=3;s=PLC">
        <DisplayName>DataBlocksGlobal</DisplayName>
        <Description Locale="en-US">Holding all global available datablocks.</Description>
        <References>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=3;s=PLC</Reference>
            <Reference ReferenceType="HasTypeDefinition">ns=2;i=1005</Reference>
        </References>
    </UAObject
<UAObject NodeId="ns=3;s=&quot;DB_ScvsInterface01&quot;" BrowseName="3:DB_ScvsInterface01" ParentNodeId="ns=3;s=DataBlocksGlobal">
        <DisplayName>DB_ScvsInterface01</DisplayName>
        <References>
            <Reference ReferenceType="Organizes" IsForward="false">ns=3;s=DataBlocksGlobal</Reference>
            <Reference ReferenceType="HasTypeDefinition">i=58</Reference>
        </References>
    </UAObject>
    <UAVariable NodeId="ns=3;s=&quot;DB_ScvsInterface01&quot;.&quot;IN&quot;" BrowseName="3:IN" ParentNodeId="ns=3;s=&quot;DB_ScvsInterface01&quot;" DataType="ns=3;s=DT_&quot;UDT_ScvsCommIn&quot;" AccessLevel="3">
        <DisplayName>IN</DisplayName>
        <Description Locale="de-DE">Daten an SCVS</Description>
        <References>
            <Reference ReferenceType="HasTypeDefinition">ns=3;s=VT_"UDT_ScvsCommIn"</Reference>
            <Reference ReferenceType="HasComponent" IsForward="false">ns=3;s="DB_ScvsInterface01"</Reference>
        </References>
    </UAVariable>

```
And extract above of our nodeset .... I hope this is helpful, sorry I cant provide the full one :(

closed 2023-03-31 by schroeder-