FreeOpcUa/opcua-asyncio #1064: "Error in watchdog loop" after v0.9.98 bump

thetanz-geoff, 2022-10-03:
**Describe the bug** <br />
In a fresh build with no other changes apart from an upgrade of the asyncua dependency from 0.9.97 to 0.0.98, previously working OPC comms fail with "Error in watchdog loop".

**To Reproduce**<br />
Update asyncua dependency from 0.9.97 to 0.9.98; rebuild and enjoy.

**Expected behavior**<br />
Comms to complete successfully.

~**Screenshots**~ How about **Logs**?<br />
```
2022-10-02 17:28:53,428 | opcUA                | DEBUG | read_node            | Read value: 3.0 from node: XYZ_ABC.XYZ123.XY123_Ab1_NodeNodeNode\5ec6d85-fa3-10441 Error in watchdog loop
Traceback (most recent call last):
   File "/usr/local/lib/python3.7/site-packages/asyncua/client/client.py", line 454, in _monitor_server_loop
     await asyncio.sleep(timeout)   File "/usr/local/lib/python3.7/asyncio/tasks.py", line 595, in sleep
     return await future concurrent.futures._base.CancelledError
Error while renewing session Traceback (most recent call last):
   File "/usr/local/lib/python3.7/site-packages/asyncua/client/client.py", line 472, in _renew_channel_loop
     await asyncio.sleep(duration)
   File "/usr/local/lib/python3.7/asyncio/tasks.py", line 595, in sleep
     return await future concurrent.futures._base.CancelledError
```

**Version**<br />
Python-Version: `3.7.14`<br />
opcua-asyncio Version (e.g. master branch, 0.9): tag `v0.9.98`

Best I can see, the issue is caused by this change here:
[image]
which removes two lines that swallow an `asyncio.CancelledError` and instead let the exception be raised.

Happy to submit a PR to fix but would prefer input first to be sure I'm not undoing something worthy.

oroulet, 2022-10-03:
but when do you get that issue? tests do not fail and we are using asyncua everyday here and we did not see any crash.
Can you send a step by step scenerio to reproduce the issue?

oroulet, 2022-10-03:
I see you are using python 3.7, maybe there is something special with that old python version, although we have tests and they pass

thetanz-geoff, 2022-10-03:
Thanks for the jet-fast response.

As noted above, the issue occurs when:
`requirements.txt` has:
```
asyncua~=0.9.95
```

and does not occur when:
`requirements.txt` has:
```
asyncua==0.9.97 #0.9.98 breaks watchdog loop
```

There is no other difference between the builds.

Normal output:
```
2022-10-03 08:32:18,929 | opcUA                | DEBUG | read_node            | Read value: 6.0 from node: XYZ_ABC.XYZ123.XY123_Ab1_NodeNodeNode\5ec6d85-fa3-10441
2022-10-03 08:32:18,930 | opcReporter          | DEBUG | recordResult         | Finished read_node for HeadNumber - report id [1448]. Result: [6.0].
```

thetanz-geoff, 2022-10-03:
I'll provide a usage snippet forthwith, but this code has been working a-ok in Production environments for the better part of 2 years until the revision bump.

oroulet, 2022-10-03:
I wrote that change, it should really not change anything for users.. and did not break any test, but obviously it is breaking something for you .
Generally CancelledError should not be catched, it only happens when we want to close a task

thetanz-geoff, 2022-10-25:
Hi @oroulet and team.

We've continued to investigate this issue. From what we can tell, it is benign - the node is still read and the data is captured, but the exception is raised and logged while we are disconnecting.

Here's evidence of this:
[image]

And here's the calling code:
[image]

What we've yet to determine is:
a. Why this occurs consistently, yet only for one of the nodes we read, `HeadNumber`. We also read a boolean flag `XyzAccumulating` (at the same frequency) but that occurs consistently without any exception being raised. As highlighted in green in the above screenshot.
b. A workaround or means of suppressing this error, which is recorded every 9 seconds or so in our scenario.

Note that we have wrapped our OPC comms with the following code. Since encountering this error, we've updated the code to add locking, but it has made no difference to the issue reported in this thread.

```
import asyncio
import os
import threading
from os.path import basename
from typing import Any, Dict, List, Optional, Set, Tuple

from asyncua import Client, ua
from asyncua.common.node import Node

from common import Logger as setLogger

log = setLogger(basename(__file__))
logger = log.setStreamHandler()

class OpcUAComms:
    """ 
    Manages Communication with the OPC UA server
    """

    def __init__(self, opcServerURL: str):
        """ 
        Initialise object using the provided URL
        :param str opcServerURL: URL to the OPC UA Server, "opc.tcp://10.x.x.x:xxxx"
        """
        self.opcServerURL = opcServerURL
        self.opcClient = Client(url = self.opcServerURL, timeout=10)
        self.syncLock = threading.Lock()

    async def writeValue(self, value, node, type: ua.VariantType = ua.VariantType.Double):
        data_value = ua.DataValue(ua.Variant(value, type))
        try:
            await node.set_value(data_value)
        except Exception as e:
            logger.exception(e)

    def getNode(self, node_name:str, namespaceidx = 2) -> Node:
        """ 
        Return the actual OPC UA Node object from the node name
        """
        return self.opcClient.get_node(ua.NodeId(node_name, NamespaceIndex = namespaceidx))

    async def read_node(self, node: str) -> Any:
        """ 
        Read the provided name, gets the matching OPC UA node and return its value
        """
        if node is None or node == "":
            raise ValueError

        self.syncLock.acquire()
        try:
            await self.opcClient.connect()

            node_image_cognition_generic = self.getNode(node)
[... 87 lines cut ...]

        """
        self.syncLock.acquire()
        try:
            await self.opcClient.connect()

            for node in payload:
                current_node = self.getNode(node['name'])
                if 'type' in node:
                    specific_type = ua.VariantType.Double
                    try:
                        specific_type = ua.VariantType[type]
                    except KeyError as e:
                        logger.exception(e)
                    await self.writeValue(node['value'],current_node, type=specific_type)
                else:
                    # Assumes Double
                    await self.writeValue(node['value'],current_node)
                logger.debug(f"Wrote value {node['value']} in node {node['name']} ")
        finally:
            await self.opcClient.disconnect()
            self.syncLock.release()
```

Any insights very welcome, please. Confirming that this issue still goes away if we peg the version of `opcua-asyncio` to `0.9.97` or lower.

oroulet, 2022-10-25:
@AndreasHeine looks like something similar to your issue. But as long as none of you can provide a way to reproduce that error we cannot do much.

AndreasHeine, 2022-10-25:
> @AndreasHeine looks like something similar to your issue. But as long as none of you can provide a way to reproduce that error we cannot do much.

did you guys know when the python whileloop checks the condition? at the end of each iteration or constantly while awaiting a task like sleep?

thetanz-geoff, 2022-10-26:
Hi @AndreasHeine; I'm unaware of your issue but a rough Googling suggests it might be this?
https://bugs.python.org/issue44795

I also came across this:
https://www.pythonfixing.com/2022/01/fixed-how-to-shutdown-gracefully-on.html

> If your task is performing blocking work while being cancelled, you won't be able to catch the exception within the task itself. The safe bet here is to register a done callback using `add_done_callback` on your `asyncio.Task`.

Which suggests that `add_done_callback` might be necessary to gracefully handle certain cancellations, a la:
```
def done_cb(task: asyncio.Task):
    name = task.get_name()
    try:
        task.exception()
    except asyncio.CancelledError:
        print(f"Done: Task {name} was cancelled")
        pass
    except Exception as e:
        print(f"Done: Task {name} didn't handle exception { e }")
    else:
        print(f"Done: Task {name} is simply done")
```

thetanz-geoff, 2022-11-06:
Upgrading to Python 3.8 (specifically 3.8.15) appears to have resolved the issue.

closed 2022-11-06 by thetanz-geoff

oroulet, 2022-11-06:
@AndreasHeine did you also use python 3.7?

AndreasHeine, 2022-11-06:
yes till eol! https://devguide.python.org/versions/

we can drop 3.7 next year 😉

oroulet, 2022-11-06:
Ok. Then there is something with python3.7. probably s bug there. It is not the first time we are hit by a python bug

pizzo00, 2022-11-29:
Had the same issue with 3.7 resolved with 3.8