FreeOpcUa/opcua-asyncio #1075: Feature request / Question: How to make a Client use the XmlExporter to generate an xml file from an external opcua server

CraigPersonal100, 2022-10-10:
I have used `asyncua` to create a python simulator of a system, and it exposes the same (in theory) opcua interface as the real system. The real system is being created by another organisation and they have sent me a PLC that I can power up and connect to so I can see the opcua interface.

My server uses the `asyncua.common.XmlImporter` to create the nodes, and then it iterates through them and stores them in a python dictionary of type `Dict[NODE_NAME_ENUM, asyncua.Node]`, where it uses the display name of the imported Node to determine the enum, and raises an exception if there is no corresponding enum.

To be certain that my simulator is exposing the same interface as the real system, and to make any future changes in the real system easier to reflect in my simulator, I would like to replace my XML config that I semi-handwrote with one that is generated from the PLC that I have been sent. I can then run my simulator and be certain that it exposes the same interface.

I do not have access to the source code of the real system as it uses an opcua plugin for the programming environment of the PLC (I think TwinCat).

Is it possible to write a client that can connect to the PLC's opcua interface and create the XML file for me, so then I can load it into my simulator? Alternatively, is it likely I can export the XML file from TwinCAT? What is the name of this format so I know what to search for, or is it `asyncua` specific? If neither of those options are possible, how can achieve what I am trying to do?

Thanks!

oroulet, 2022-10-11:
Can't you already do that? The export_xml method works from client side

CraigPersonal100, 2022-10-11:
Ah, I was looking at the sync client, and I didn't see that the normal one had an export_xml function. The arguments for the XmlExporter class are `def __init__(self, server)` which made me think that it only worked with the server.

The `build_etree` function accepts a list of nodes as it's argument, will I have to browse the node tree with a recursive function calling `get_children`, and then flattening the result into a list, or will it work with just `exp.build_etree(self.get_objects_node().get_children())` ?

You can close this issue. Sorry, I didn't see it already existed.

closed 2023-03-29 by schroeder-