FreeOpcUa/opcua-asyncio #1079: @uamethod decorator --> DOES Automatically convert to and from variants ?

dibyendumca, 2022-10-12:
### Discussed in https://github.com/FreeOpcUa/opcua-asyncio/discussions/1078

<div type='discussions-op-text'>

<sup>Originally posted by **dibyendumca** October 12, 2022</sup>
SCREEN SHOTS ...
#@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
[image]
[image]
[image]

PLEASE REFER THE CODE SNIPPET BELOW...
############################################################################################
@uamethod
    def Multiply(parent, x, y):
        # Function Multiply
        msgLog = messageLogger()
        res = 0.0
        msgLog.WriteLog( LogErr.INF, "Multiply method called with parameters: " + str(x) + ", " + str(y))
        try:
            res = x * y
        except Exception as e:
            msgLog.WriteLog(
                LogErr.ERR, "Oops!" + sys.exc_info()[0] + "occurred.")
            msgLog.WriteLog(
                LogErr.ERR, "Oops!" + e.__class__ + "occurred.")
            print()
        else:
            msgLog.WriteLog(LogErr.INF, "Result: " + str(res))
            return res

NOW ADDING THE FUNCTION **Multiply**(x, y) ...
# ##########################################################################
    inargx = ua.Argument()
    inargx.Name = "x"
    inargx.DataType = ua.NodeId(ua.ObjectIds.Double)
    inargx.ValueRank = -1
    inargx.ArrayDimensions = []
    inargx.Description = ua.LocalizedText("First number x")

    inargy = ua.Argument()
    inargy.Name = "y"
    inargy.DataType = ua.NodeId(ua.ObjectIds.Double)
    inargy.ValueRank = -1
    inargy.ArrayDimensions = []
    inargy.Description = ua.LocalizedText("Second number y")

    outarg = ua.Argument()
    outarg.Name = "Result"
    outarg.DataType = ua.NodeId(ua.ObjectIds.Double)
    outarg.ValueRank = -1
    outarg.ArrayDimensions = []
    outarg.Description = ua.LocalizedText("Multiplication result")

    multiplyNode = await ParamMethods.add_method(spaceNodeId, "Multiply", CustomFeatures.Multiply, [inargx, inargy], [outarg])
    # #####################################################################

WHENEVER I AM CALLING THE FUNCTION Multiply(x, y) WITH PARAMETERS x=2 AND y=3 from FreeOpcUa Client tool, 
I AM GETTING THE FOLLOWING ERROR ...
###############################################################################################
Multiply method called with parameters: Variant(Value=2.0, VariantType=<VariantType.Double: 11>, Dimensions=None, is_array=False), 3.0
Error executing method call CallMethodRequest(ObjectId=NodeId(Identifier=2, NamespaceIndex=2, NodeIdType=<NodeIdType.FourByte: 1>), MethodId=NodeId(Identifier=6, NamespaceIndex=2, NodeIdType=<NodeIdType.FourByte: 1>), InputArguments=[Variant(Value=2.0, VariantType=<VariantType.Double: 11>, Dimensions=None, is_array=False), Variant(Value=3.0, VariantType=<VariantType.Double: 11>, Dimensions=None, is_array=False)]), an exception was raised:
Traceback (most recent call last):
  File "...\CustomFeaturesStub.py", line 54, in Multiply
    res = x * y
TypeError: unsupported operand type(s) for *: 'Variant' and 'float'

AFTER MODIFYING THE CODE OF THE FUNCTION Multiply(x, y) ...
[image]
[image]

CODE SNIPPET ....
# ################################################################################
@uamethod
    def Multiply(parent, x, y):
        # Function Multiply
        msgLog = messageLogger()
        x = x.Value
        y = y.Value
        res = 0.0
        msgLog.WriteLog(
            LogErr.INF, "Multiply method called with parameters: " + str(x) + ", " + str(y))
        try:
            res = x * y
        except Exception as e:
            msgLog.WriteLog(
                LogErr.ERR, "Oops!" + sys.exc_info()[0] + "occurred.")
            msgLog.WriteLog(
                LogErr.ERR, "Oops!" + e.__class__ + "occurred.")
            print()
        else:
            msgLog.WriteLog(LogErr.INF, "Result: " + str(res))
            return res
# ######################################################################

ERROR ...
Error executing method call CallMethodRequest(ObjectId=NodeId(Identifier=2, NamespaceIndex=2, NodeIdType=<NodeIdType.FourByte: 1>), MethodId=NodeId(Identifier=6, NamespaceIndex=2, NodeIdType=<NodeIdType.FourByte: 1>), InputArguments=[Variant(Value=2.0, VariantType=<VariantType.Double: 11>, Dimensions=None, is_array=False), Variant(Value=3.0, VariantType=<VariantType.Double: 11>, Dimensions=None, is_array=False)]), an exception was raised:
Traceback (most recent call last):
  File "C:\ProgramData\Anaconda3\lib\site-packages\asyncua\server\address_space.py", line 583, in _call
    result = await self._run_method(node.call, method.ObjectId, *method.InputArguments)
  File "C:\ProgramData\Anaconda3\lib\site-packages\asyncua\server\address_space.py", line 602, in _run_method
    res = await asyncio.get_event_loop().run_in_executor(self._pool, p)
  File "C:\ProgramData\Anaconda3\lib\concurrent\futures\thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "C:\ProgramData\Anaconda3\lib\site-packages\opcua\common\methods.py", line 69, in wrapper
    result = func(self, parent, *[arg.Value for arg in args])
  File "D:\MADI04\PROJECTS\PYTHON\ASYNCUA\DacXpertM\CustomFeaturesStub.py", line 51, in Multiply
    y = y.Value
AttributeError: 'float' object has no attribute 'Value'

#########################################################################################
**IS THIS A BUG ???**, @uamethod --> Converting only the 2nd parameter from VARIANT to FLOAT, but not the 1st parameter

</div>

schroeder-, 2022-10-12:
The paramerters are converted from Variants. Your code looks like copied from https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/server-methods.py , which i tried with the current  release and is working correct. Are you using the latest pip version?
The only other possibility is that the client is sending a Variant(Variant(3.0)). We currently doesn't check if the parameters are matching the input args.

dibyendumca, 2022-10-12:
THANK YOU SO MUCH..........
I HAD AN OLD PIP VERSION

closed 2022-10-12 by dibyendumca