FreeOpcUa/opcua-asyncio #1095: client asyncua to kepware

pedramjazayeri, 2022-10-23:
Hi, 
I wrote a code with asyncua in python. I could not synchronize it with KEPSERVER.
Also, I read thee example "client-kepware" but yet the problem remains. What should I do?
I think it is because the username and password.
I would be thankful if you help me.
thanks.

AndreasHeine, 2022-10-23:
whats the issue? error?

pedramjazayeri, 2022-10-23:
The code is:

from opcua import Client
url='opc.tcp://localhost:49320'   
client=Client(url)
cert = "cert.der"
key = "key.pem"
client.set_security_string("Basic256Sha256, SignAndEncrypt, cert.der, key.pem") 
client.connect() 
_______________________
The error is:
AttributeError:  SignAndEncrypt
_______________________________________

and also, I am doubtful from where I can drive certificate and private key. please guide me
thanks

AndreasHeine, 2022-10-23:
can you provide the stacktrac?

pedramjazayeri, 2022-10-23:
did you mean below?
_________________________________

runfile('C:/Users/P.e.d.r.a.M/untitled0.py', wdir='C:/Users/P.e.d.r.a.M')
Traceback (most recent call last):

  File "C:\Users\P.e.d.r.a.M\newanaconda3\lib\site-packages\spyder_kernels\py3compat.py", line 356, in compat_exec
    exec(code, globals, locals)

  File "c:\users\p.e.d.r.a.m\untitled0.py", line 4, in <module>
    client.set_security_string("Basic256Sha256, SignAndEncrypt, cert.der, key.pem")

  File "C:\Users\P.e.d.r.a.M\newanaconda3\lib\site-packages\opcua\client\client.py", line 190, in set_security_string
    mode = getattr(ua.MessageSecurityMode, parts[1])

  File "C:\Users\P.e.d.r.a.M\newanaconda3\lib\enum.py", line 429, in __getattr__
    raise AttributeError(name) from None

AttributeError:  SignAndEncrypt

pedramjazayeri, 2022-10-23:
Generally for synchronizing asyncua in kepserver 6.6, what steps should I follow?

AndreasHeine, 2022-10-23:
> Generally for synchronizing asyncua in kepserver 6.6, what steps should I follow?

not sure what you mean with syncronize...

but first you need to connnect which fails in setsecuritysring... it somehow does not find the securitymode SignAndEncrypt in the enum...

did you have a fresh installation of the current release? which version did you use?

AndreasHeine, 2022-10-23:
https://github.com/FreeOpcUa/opcua-asyncio/blob/master/asyncua/client/client.py#L159

alternativly you could try to set the securitysettings the advanced way!

pedramjazayeri, 2022-10-23:
I meant I wanna make connection between them.
I wrote this code in spyder v5.3.3 and kepserver.EX 6.6.
I can connect kepserver with software "UaClient" but I can not connect my code with kepserver. This is the problem.

> not sure what you mean with syncronize...
>
> but first you need to connnect which fails in setsecuritysring... it somehow does not find the securitymode SignAndEncrypt in the enum...
>
> did you have a [...]

pedramjazayeri, 2022-10-23:
sir, how I can get certificate and private key?

> https://github.com/FreeOpcUa/opcua-asyncio/blob/master/asyncua/client/client.py#L159
> 
> alternativly you could try to set the securitysettings the advanced way!

AndreasHeine, 2022-10-23:
you need to create both for your application for example with openssl!
the subjectAltName must match the client.application_uri propertie

> openssl req -x509 -newkey rsa:4096 -sha256 -keyout my_private_key.pem -out my_cert.pem -days 3650 -nodes -addext "subjectAltName = URI:urn:example.org:FreeOpcUa:python-opcua"

> openssl x509 -outform der -in my_cert.pem -out my_cert.der

https://github.com/FreeOpcUa/opcua-asyncio/tree/master/examples
https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/generate_certificate.sh

AndreasHeine, 2022-10-23:
did you have spaces in the security string?
"Basic256Sha256,_SignAndEncrypt,_cert.der,_key.pem"

its not working if so!

pedramjazayeri, 2022-10-23:
> did you have spaces in the security string? "Basic256Sha256,_SignAndEncrypt,_cert.der,_key.pem" <-- marked them with _
> 
> its not working if so!

I've changed spaces with "_" but still not working and the error is:

runfile('D:/BMN/OPC/1401-08/1401-08-01/untitled0.py', wdir='D:/BMN/OPC/1401-08/1401-08-01')
Traceback (most recent call last):

  File "C:\Users\pedra\anaconda3\lib\site-packages\spyder_kernels\py3compat.py", line 356, in compat_exec
    exec(code, globals, locals)

  File "d:\bmn\opc\1401-08\1401-08-01\untitled0.py", line 7, in <module>
    client.set_security_string('Basic256Sha256,_SignAndEncrypt,_8c458df4c321c8e3e7d10c74e276cc4751712f0f.der,_key.pem')

  File "C:\Users\pedra\anaconda3\lib\site-packages\opcua\client\client.py", line 190, in set_security_string
    mode = getattr(ua.MessageSecurityMode, parts[1])

  File "C:\Users\pedra\anaconda3\lib\enum.py", line 429, in __getattr__
    raise AttributeError(name) from None

AttributeError: _SignAndEncrypt

AndreasHeine, 2022-10-23:
no! no spaces or _ it just insert them to show you the positions where you added spaces... which is not allowed

pedramjazayeri, 2022-10-24:
> no! no spaces or _ it just insert them to show you the positions where you added spaces... which is not allowed

excuse me sir, I remove spaces, now the error changed into this:
UaError: No matching endpoints: 3, http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256

AndreasHeine, 2022-10-24:
the error tells you that the opc ua server does not expose a endpoint with Basic256Sha256!

You can lookup the avalible endpoint by open the kepware configuration or with uaexpert and one more advanced way would be to call client.connect_and_get_server_endpoints() and connect with the propper client settings

pedramjazayeri, 2022-10-24:
dear AndresHeine, Another question is that if I want to connect my code to kepserver without any username,password or any security, like code below:

from opcua import Client
url="opc.tcp://127.0.0.1:49320"   #"IP address:port number"
client=Client(url)
client.connect() 
_______________________________

I receive the error:
 BadSessionNotActivated: "The session cannot be used because ActivateSession has not been called."(BadSessionNotActivated)
_______________
for this error what is the solution?
thanks

> the error tells you that the opc ua server does not expose a endpoint with Basic256Sha256!
>
> You can lookup the avalible endpoint by open the kepware configuration or with uaexpert and one more [...]

AndreasHeine, 2022-10-24:
client.connect() calls ActivateSession... see in sourcecode:
https://github.com/FreeOpcUa/opcua-asyncio/blob/ca19970e5c2c31c51dd659197a3a759cf28b2a45/asyncua/client/client.py#L269
[image]

did you have a clean fresh installation of asyncua? and the latest version?

pedramjazayeri, 2022-10-24:
> client.connect() calls ActivateSession... see in sourcecode:
> 
> https://github.com/FreeOpcUa/opcua-asyncio/blob/ca19970e5c2c31c51dd659197a3a759cf28b2a45/asyncua/client/client.py#L269
> 
> [image]
> 
> did you have a clean fresh installation of asyncua? and the latest version?

asyncua version 0.9.98
opcua  0.98.13

AndreasHeine, 2022-10-24:
opcua 0.98.13 <--- Deprecated and not maintained anymore and not part of this repository!

currently hard to help you... you post only snipits not real used code... so i can only review library code which is correct and works if you use it the way its ment to be!

Kepware Opc Ua Endpoints:
[image]
Verion:
[image]

UaExpert GetServerEndpoints (no security and no password):
[image]

basic client template:
```python
import asyncio
import logging
from asyncua import Client, ua

logging.basicConfig(level=logging.WARNING)
_logger = logging.getLogger('asyncua')

async def main():
    client = Client(url="opc.tcp://127.0.0.1:49320", timeout=4)
    # client.application_uri = "..."
    # client.session_timeout = 30000 # requested value will be negotiated between client and server
    # client.description = "..."
    # client.product_uri = "..."
    # client.name = "..."
    # client.max_chunkcount = 0 # requested value will be negotiated between client and server
    # client.max_messagesize = 0 # requested value will be negotiated between client and server
    # client.set_user = "user"
    # client.set_password = "pw"
    # await client.set_security_string("Basic256Sha256,SignAndEncrypt,cert.pem,key.pem")

    # catching all networkrequests is some times important
    try:
        await client.connect()
    except Exception as e: # catching all exceptions is not always wise!
        print(e)
        return

    '''
    clientcode!
    '''

    obj = client.get_objects_node()

    children = await obj.get_children()

    for child in children:
        print("Objects-Node-Child: ", child)

    await asyncio.sleep(2)

    try:
        await client.disconnect()
    except Exception as e:  # catching all exceptions is not always wise!
        print(e)
        return

if __name__ == "__main__":
    asyncio.run(main())
```

output in my case was:
```
Objects-Node-Child:  i=2253
Objects-Node-Child:  ns=2;s=_AdvancedTags
Objects-Node-Child:  ns=2;s=_ConnectionSharing
Objects-Node-Child:  ns=2;s=_CustomAlarms
Objects-Node-Child:  ns=2;s=_DataLogger
Objects-Node-Child:  ns=2;s=_EFMExporter
Objects-Node-Child:  ns=2;s=_IDF_for_Splunk
Objects-Node-Child:  ns=2;s=_IoT_Gateway
Objects-Node-Child:  ns=2;s=_LocalHistorian
Objects-Node-Child:  ns=2;s=_Redundancy
Objects-Node-Child:  ns=2;s=_Scheduler
Objects-Node-Child:  ns=2;s=_SecurityPolicies
Objects-Node-Child:  ns=2;s=_SNMP Agent
Objects-Node-Child:  ns=2;s=_System
Objects-Node-Child:  ns=2;s=_ThingWorx
Objects-Node-Child:  ns=2;s=Beispiele für Datentyp
Objects-Node-Child:  ns=2;s=Beispiele für Simulation
Objects-Node-Child:  ns=2;s=Kanal1
```

if the Kepware Server does look different in configuration you need to adopt the client settings!

feel free to look in the examples there is almost everything available:
https://github.com/FreeOpcUa/opcua-asyncio/tree/master/examples

pedramjazayeri, 2022-10-24:
My problem has not solved yet but thank you for you guidance.

pedramjazayeri, 2022-10-24:
Thank you, but my problem has not solved yet and the last question:

I run the code and faced with this error:
WARNING:asyncua.uaprotocol:updating client limits to: TransportLimits(max_recv_buffer=65536, max_send_buffer=65536, max_chunk_count=5000, max_message_size=16777216)

Can I solve this?

pedramjazayeri, 2022-10-24:
> its just a warning not an error!

yes that's true, but I can not access to the data on kepserver. You mean that now, my client code is connected to kepserver?

closed 2023-03-14 by AndreasHeine