FreeOpcUa/opcua-asyncio #1100: asyncua\sync.py line 394 bug

shenenlutc, 2022-10-28:
sync.py  line 394:
          return SyncNode(sync_node.tloop, node.Node(sync_node.aio_obj.**server**, nodeid))
should be:
          return SyncNode(sync_node.tloop, node.Node(sync_node.aio_obj.**session**, nodeid))

**Describe the bug** <br />
Traceback (most recent call last):
  File "C:\Users\15068\develop\code\python\opc\opcua-client-gui-master\venv\lib\site-packages\uawidgets\tree_widget.py", line 242, in fetchMore
    self._fetchMore(parent)
  File "C:\Users\15068\develop\code\python\opc\opcua-client-gui-master\venv\lib\site-packages\uawidgets\tree_widget.py", line 252, in _fetchMore
    self.add_item(desc, parent)
  File "C:\Users\15068\develop\code\python\opc\opcua-client-gui-master\venv\lib\site-packages\uawidgets\tree_widget.py", line 210, in add_item
    item[0].setData(new_node(parent_node, desc.NodeId), Qt.UserRole)
  File "C:\Users\15068\develop\code\python\opc\opcua-client-gui-master\venv\lib\site-packages\asyncua\sync.py", line 394, in new_node
    return SyncNode(sync_node.tloop, node.Node(sync_node.aio_obj.server, nodeid))
AttributeError: 'Node' object has no attribute 'server'

**To Reproduce**<br />
run opcua-client-gui app.py, click connect

**Expected behavior**<br />
can't connect to opc server.

**Version**<br />
Python-Version:<br />
asyncua Version (1.0.0):

AndreasHeine, 2022-10-28:
@shenenlutc 
for now use a older version of asyncua <1.0 (API Changes) till we update the client gui!

oroulet, 2022-10-28:
seems to be a forgotten rename in your PR @AndreasHeine ;-)

oroulet, 2022-10-28:
I made a PR, merging as soon as tests pass

AndreasHeine, 2022-10-28:
> seems to be a forgotten rename in your PR @AndreasHeine ;-)

missing tests....