FreeOpcUa/opcua-asyncio #1115: opcua-client

gchen001, 2022-11-07:
Hi there

Recently I found that when I upgrade opcua-asynico to the latest version, and i could not connect the server by this library using opcua-client gui, even when I tried to connect the server example. Could anyone helpe me? thanks a lot.

server:
--------------
import asyncio
import logging

from asyncua import Server, ua
from asyncua.common.methods import uamethod

@uamethod
def func(parent, value):
    return value * 2

async def main():
    _logger = logging.getLogger("asyncua")
    # setup our server
    server = Server()
    await server.init()
    server.set_endpoint("opc.tcp://0.0.0.0:4840/freeopcua/server/")

    # setup our own namespace, not really necessary but should as spec
    uri = "http://examples.freeopcua.github.io"
    idx = await server.register_namespace(uri)

    # populating our address space
    # server.nodes, contains links to very common nodes like objects and root
    myobj = await server.nodes.objects.add_object(idx, "MyObject")
    myvar = await myobj.add_variable(idx, "MyVariable", 6.7)
    # Set MyVariable to be writable by clients
    await myvar.set_writable()
    await server.nodes.objects.add_method(
        ua.NodeId("ServerMethod", idx),
        ua.QualifiedName("ServerMethod", idx),
        func,
        [ua.VariantType.Int64],
        [ua.VariantType.Int64],
    )
    _logger.info("Starting server!")
    async with server:
        while True:
            await asyncio.sleep(1)
            new_val = await myvar.get_value() + 0.1
            _logger.info("Set value of %s to %.1f", myvar, new_val)
            await myvar.write_value(new_val)

if __name__ == "__main__":
    logging.basicConfig(level=logging.DEBUG)
    asyncio.run(main(), debug=True)

-------
[image]

oroulet, 2022-11-07:
was a bug which was fixed a few days ago. Doing a release now so that everything works again

gchen001, 2022-11-08:
Thanks for your explaination. I guess so, because when i tested it in the uaexpert, everything works well

closed 2022-11-08 by gchen001