FreeOpcUa/opcua-asyncio #1122: Unexpected behavior finding registered server

alejandro-zuleta-globant, 2022-11-10:
**Describe the bug** <br />
`Client.find_servers` method is not working, at least, as my common sense expects it. I have two registered servers in a additional discovery server. When I try to get the registered servers from the discovery server using a client instance I get only one server in the response. Presumably the last registered server.

**To Reproduce**<br />

Run each function using `asyncio.run(main())`

Discovery server

```py
async def main():
    server = Server()
    await server.init()

    server.set_endpoint("opc.tcp://0.0.0.0:4841/freeopcua/server/")

    uri = "http://examples.freeopcua.github.io"
    idx = await server.register_namespace(uri)

    async with server:
        logger_.info("Discovery server is running")
        while True:
            await asyncio.sleep(10)
```

Registered server 1
```py
async def main():
    server = Server()
    await server.init()

    server.set_endpoint("opc.tcp://0.0.0.0:4842/freeopcua/server/")

    uri = "http://examples1.freeopcua.github.io"
    idx = server.register_namespace(uri)

    async with server:
        await server.register_to_discovery(url="opc.tcp://localhost:4841", period=10)
        logger_.info("Registered server 1 is running")
        while True:
            await asyncio.sleep(10)
```
Registered server 2

```py
async def main():
    server = Server()
    await server.init()

    server.set_endpoint("opc.tcp://0.0.0.0:4843/freeopcua/server/")

    uri = "http://examples.freeopcua.github.io"
    idx = server.register_namespace(uri)

    async with server:
        await server.register_to_discovery(url="opc.tcp://localhost:4841", period=10)
        logger_.info("Registered server 2 is running")
        while True:
            await asyncio.sleep(10)
```
Discovery client
```py
async def main():
    client = Client(url="opc.tcp://localhost:4841", timeout=10)

    async with client:
        logger_.info("Discovery client is running")
        
        while True:
            await asyncio.sleep(3)
            servers = await client.find_servers()
            for server in servers:
                logger_.info(f"Servers: {servers}")
```

**Expected behavior**<br />
The `Client.find_servers` should return a list of registered server, however it returns only one server, as shown bellow
```
INFO:DISCOVERY_CLIENT:Servers: [ApplicationDescription(ApplicationUri='urn:freeopcua:python:server', ProductUri='urn:freeopcua.github.io:python:server', ApplicationName=LocalizedText(Locale=None, Text='FreeOpcUa Python Server'), ApplicationType_=<ApplicationType.ClientAndServer: 2>, GatewayServerUri=None, DiscoveryProfileUri=None, DiscoveryUrls=['opc.tcp://0.0.0.0:4842/freeopcua/server/'])]
```
Sometimes the response alternates between one server or the other. I think it is because of the re-register scheduler but I am not sure.

[image]

**Version**<br />
Python-Version: 3.10.7
asyncua==0.9.98

AndreasHeine, 2022-11-11:
not sure if that stuff works correct...

[image]

connect creates a session... but discovery does not work on that layer...

@oroulet 
from my knowledge it just opens a secure-channel for the register request!

oroulet, 2022-11-11:
I wrote that code 10 years ago and never tested it. So i do not remember anything. Not sure it works at all

AndreasHeine, 2023-02-05:
@alejandro-zuleta-globant https://github.com/FreeOpcUa/opcua-asyncio/pull/1193

alejandro-zuleta-globant, 2023-02-20:
@AndreasHeine I just installed `asyncua` from the latest commit in the master branch but it seems to behave the same way, as the `client.find_servers()` function returns only the last registered server and not all.

AndreasHeine, 2023-02-20:
i fixed one clear bug that the server uses session context to register which is from opcua spec. wrong. havent looked into find servers yet but let me have a look on it!