FreeOpcUa/opcua-asyncio #1125: How to delete value from Node?

Flock1993, 2022-11-17:
I have server:
```
import asyncio
import logging

from asyncua import Server, ua

async def main():
    _logger = logging.getLogger("asyncua")
    server = Server()
    await server.init()
    server.set_security_policy([
        ua.SecurityPolicyType.NoSecurity,
        ua.SecurityPolicyType.Basic256Sha256_SignAndEncrypt,
        ua.SecurityPolicyType.Basic256Sha256_Sign])
    server.set_endpoint("opc.tcp://localhost:4840/freeopcua/server/")
    uri = "http://localhost"
    idx = await server.register_namespace(uri)
    await server.nodes.objects.add_object(idx, "Object")
    await server.nodes.objects.add_method(
        ua.NodeId("ServerMethod", idx),
        ua.QualifiedName("ServerMethod", idx),
        [ua.VariantType.Int64],
        [ua.VariantType.Int64],
    )
    async with server:
        while True:
            await asyncio.sleep(1)

if __name__ == "__main__":
    logging.basicConfig(level=logging.DEBUG)
    asyncio.run(main(), debug=True)
```

After starting Server i write values to Node Object with this code

```
import asyncio

from asyncua import Client as OPClient
from asyncua.common.node import Node

URL = "opc.tcp://admin@localhost:4840/freeopcua/server/"
NAMESPACE = "http://localhost"
TEST_TAGS = {"test_tag1": 1, "test_tag2": 2}

"""Adding values to Node OPC UA Server"""
print(f"Connecting to OPC UA Server {URL} ...")
async with OPClient(url=URL) as opc_client:
    namespace_index: int = await opc_client.get_namespace_index(NAMESPACE)
    print(f"Namespace Index for '{NAMESPACE}': {namespace_index}")
    tag_obj: Node = await opc_client.nodes.root.get_child(
        ["0:Objects", f"{namespace_index}:TagsObject"]
    )
    for tag, value in TEST_TAGS.items():
        var: Node = await tag_obj.add_variable(namespace_index, tag, value)
        print(str(tag) + '-' + str(await var.read_value()))
```

I want to delete wroten values of Node. how i can do it?
I found only the function to delete node from address space.
Server restart is not my variant too.
Thanks you in advance! And sorry if theme of my question dont match.

oroulet, 2022-11-17:
a variable node has always a  value. some node type can have value None. so you can trey to write None

Flock1993, 2022-11-17:
Thank you for answers
Now i think the idea with server restrart is more elegant. Because i am using it for testing.
Is there a method for do it?

closed 2022-12-05 by Flock1993