FreeOpcUa/opcua-asyncio #1136: Example program not quitting if ConnectionRefused

km-manzm, 2022-12-01:
Hello Devs,
I tried to run the example in the examples/sync/client-minimal.py.
If it can't connect to the server, because no OPC-UA Server is running at given host address, the program will never quit (but it should)

If i send a SIGINT signal to the program (CTRL+C) the following error is showing:

```
^CException ignored in: <module 'threading' from '/usr/local/lib/python3.8/threading.py'>
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/threading.py", line 1388, in _shutdown
    lock.acquire()
KeyboardInterrupt: 
```

(Using the async verion of the example works.)

**To Reproduce**<br />
install asyncua in latest version, run the example without any OPC UA Server running.
Program will not stop and quit itself.

**Expected behavior**<br />
Program will stop and quit itselt

**Version**<br />
Python-Version: 3.8 and 3.11 (only these two have been tested)
opcua-asyncio Version 0.9.8, 0.9.98, 1.0.0 (only these versions tested)

suCRR, 2023-02-15:
Hi,
I'm facing the same issue with version 1.0.1.

schroeder-, 2023-02-19:
I can reproduce it, but i looks like a bug in either python or asyncio error. As I am no expert in this area, I cannot help here. Maybe @oroulet  knows anything about it.

oroulet, 2023-02-19:
interesting. I can reproduce too. The async version works so it must be something with the wrappers

oroulet, 2023-02-19:
OK I have been bit by this one before. when an exceptin is raised in `__enter__` then `__exit__`is not called.
see https://github.com/FreeOpcUa/opcua-asyncio/pull/1218

closed 2023-03-29 by schroeder-