FreeOpcUa/opcua-asyncio #1139: tloop sometimes won't stop after connection failed

km-manzm, 2022-12-02:
**Describe the bug** <br />
In my code if the connection to the OPC UA server breaks in any way (e.g.  Server crashed) i will try to reconnect to it every 60 seconds. However sometimes it can happen that the tloop thread of one client will be running forever. 

After some time (4-5 hours) i can see many tloop threads running (about 50-100) which will sooner or later lead to exceptions that no new Threads can be spawned anymore.

**To Reproduce**<br />

`from asyncua.sync import Client
import time

connected = False
while not connected:
  try:
    _opcua_client = Client("tcp.opc://127.0.0.1:4842/")
    _opcua_client.connect()
    connected = True
  except Exception:
    print("could not connect")
    time.sleep(60)

**Expected behavior**<br />
if connection failed, there is no need for the tloop to run. So that thread should be quit as soon as possible and not running forever.

**Version**<br />
Python-Version:3.8<br />
opcua-asyncio Version (e.g. master branch, 0.9): 1.0.0

km-manzm, 2022-12-02:
I've found a workaround for my code that works:

if there was a client before, i call the _opcua_client.tloop.stop() methode.

```
if _opcua_client:
  # if there was a client before, stop the tloop due to bug
  _opcua_client.tloop.stop()
```

After implementing that line, there is always only one tloop thread running.

oroulet, 2022-12-02:
your code over is not correct, you should always call disconnect().
I admit that starting the Threadloop in the background is a bad idea. I should never have done that API. but generally you never know what a connect method does(). Connection can fail at any internal step and you can end up having to do some cleaning. So your code should be something like:

```

connected = False
while not connected:
    _opcua_client = Client("tcp.opc://127.0.0.1:4842/")
  try:
    _opcua_client.connect()
    connected = True
  except Exception:
    print("could not connect")
    time.sleep(60)
   finally:
       _opcua_client.disconnect()
   
```
or even better use explicit tloop as shown in the examples:
```
    with ThreadLoop() as tloop:
        with Client("opc.tcp://localhost:4840/freeopcua/server/", tloop=tloop) as client:
```
or 
```
      tloop = ThreadLoop()
    try: 
  
        tloop.start()
        c = Client("opc.tcp://localhost:4840/freeopcua/server/", tloop=tloop) 
        ....whatever....
     finally :
        tloop.stop()
```

closed 2023-03-29 by schroeder-