FreeOpcUa/opcua-asyncio #1142: Best practice of storing data in a database from a client

unregistered33, 2022-12-07:
Not issue, just "how to".

AndreasHeine, 2022-12-07:
the SubscriptionHandler is Triggered vor each DataChangeNotification in the Publishresponse of the Subscription! So in essence each Subscription consists of a number of MonitoredItems (like a Group of Nodes/Variables) and for each DataChangeNotification in that "Group" the SubHandler.datachange_notification(...) gets executed! I would recommend using an asyncio queue which gets filled by the "datachange_notification" callback with a queue consuming task which delegates the notification to a "DataBaseConnectionPool" which finally makes the "SQL Insert".

note: the subhandler methods can be async! 

https://github.com/FreeOpcUa/opcua-asyncio/blob/e680d7ee57591e08f0a31d75a3c5b1cfbec4e315/asyncua/common/subscription.py#L47
```python
class SubHandler:
    """
    Subscription Handler. To receive events from server for a subscription
    This class is just a sample class. Whatever class having these methods can be used
    """

    def datachange_notification(self, node: Node, val, data: DataChangeNotif):
        """
        called for every datachange notification from server
        """
        pass

    def event_notification(self, event: ua.EventNotificationList):
        """
        called for every event notification from server
        """
        pass

    def status_change_notification(self, status: ua.StatusChangeNotification):
        """
        called for every status change notification from server
        """
        pass

```

unregistered33, 2022-12-08:
Thank you, Andreas!

closed 2022-12-08 by unregistered33