FreeOpcUa/opcua-asyncio #1143: Simatic S7 1500 - connection issues

nervaccio, 2022-12-07:
Hello,
I'm pretty sure that is not a problem of the library, but I can't find another place where ask for help or some direction.

I have some issue to connect to a Simatic S7 1500, without use the client that supplier supply "UA Client 1500". There is a User & Password, its the only things I've hidden.

[image]

[image]

[image]

[image]

Until here its all fine, but I would to connect through python to export the data in a programmatically way.

This is my first try with **client-minimal-auth.py** where I've changed **url, client.set_user ,client.set_password**
[image]

The modals that I accept with the UA Client 1500 its related to the certificate? I need to ask for the certificate to the supplier?
I've found this on stackoverflow [https://stackoverflow.com/questions/50328537/generate-certificate-for-opc-client](https://stackoverflow.com/questions/50328537/generate-certificate-for-opc-client) 
there is this bash script to generate certificates, but I have no luck with:
* client-with-encryption.py (example)
* certificates in the examples folder or self generate certificates

```bash
#!/bin/bash
openssl genrsa -out default_pk.pem 2048
openssl req -new -key default_pk.pem -out cert.csr -subj "/C=IT/ST=RM/L=RM/O=Procosmet/OU=Dev/CN=172.16.1.1"
openssl x509 -req -days 3650 -extfile extensions.cnf -in cert.csr -signkey default_pk.pem -out public.pem
openssl x509 -in public.pem -inform PEM -out public.der -outform DER
```

I really appreciate every tips or direction, its my first time with this kind of integration.

PS: I've tried with the GUI client, the issue is the same.

Thanks
Giovanni

AndreasHeine, 2022-12-07:
ok the error appears on create session but the server does not return a statuscode... so there might be an issue with the user, certificate or there is no free session on the server... another thing could be that the server needs the client certificate in the trusted list but its just guessing...

nervaccio, 2022-12-07:
Next week I'll try to contact the supplier. 
Are there anything that I can ask about the opc ua server setting?

What does mean "there is no free session on the server" ?

I want to try with Wireshark and see opcua-asyncio and their client (that work) if I see something different. Until I get in touch with the supplier.

Thank your for your time, this friday I'll do more tests.

If anyone else have some others tips its welcome ❤

nervaccio, 2022-12-09:
[UPDATE]
I've tried with a third party library, UaExpert 1.6, just to be sure there isn't a strange configuration on OPC UA server.
Even with UaExpert a modal appear and ask to accepet server(?) certificate to continue.

So probably I make some mistake on the script. I'll check if there is some api to "accept server's certificate".

[image]

AndreasHeine, 2022-12-09:
we trust server certificates by default

dremsol, 2023-02-17:
Please try using `client.application_uri = "urn:freeopcua:client"` in your client application and set the urn to be equal to your generated client certificate.

closed 2023-03-14 by AndreasHeine