FreeOpcUa/opcua-asyncio #1146: Logging: How to log data changes by client in server

INI-MED, 2022-12-09:
###

Hi, i have a server and at least 2 clients.
I found out how to log which client is trying to connect to my server

`class UserManager:`

    def __init__(self, log_queue: Queue):
        self.logger = QueueLogger("connections", log_queue)
        self.users = []

    def get_user(self, iserver: InternalServer, username=None, password=None, certificate=None) -> User:
        if username and iserver.allow_remote_admin and username in ("admin", "Admin"):
            user = User(role=UserRole.Admin)
        else:
            user = User(role=UserRole.User)
        user.name = username
        self.users.append(user)
        self.logger.warning(f"New user has been connected: {user.name}, {user.role}")
        return user

Also, I found out how to get data changes notifications

`class SubHandler:`

    def __init__(self, sub_queue):
        self.logger = QueueLogger('nodes', sub_queue)

    def datachange_notification(self, node, val, data):
        var = f'ns={node.nodeid.NamespaceIndex};s={node.nodeid.Identifier}'
        self.logger.info(var + ' = ' + str(val))

now I want to log on server side variable change with the name of client who did the change

---
_**The problem is**:_ I can't get any info about variable change source

Thanks in advance

---

Python-Version:<br /> 3.9
opcua-asyncio Version :<br /> 1.0.1

AndreasHeine, 2022-12-09:
what do you mean with "variable change source"?
"data" containes all the datachangenotification data!

AndreasHeine, 2022-12-09:
no the subscribed node does not provide such details!

AndreasHeine, 2022-12-09:
its not part of OPC UA Spec. to be precise

INI-MED, 2022-12-10:
> its not part of OPC UA Spec. to be precise

So, it's impossible to know which client is changing variables on my server?

oroulet, 2022-12-10:
There is a monitoring api somewhere. Look at all example. I do not remember what it is called.

INI-MED, 2022-12-10:
> There is a monitoring api somewhere. Look at all example. I do not remember what it is called.

In the examples on github or somewhere in the library files?

Kantiran91, 2022-12-20:
> There is a monitoring api somewhere. Look at all example. I do not remember what it is called.

Do you mean the Audit Events?  [ see Part 3](https://reference.opcfoundation.org/Core/Part3/v105/docs/9.5)