FreeOpcUa/opcua-asyncio #1147: Function set_writable() seems not to be working

ivan2033, 2022-12-15:
**Describe the bug** <br />

I wanted to change a variable node writable atribute in order to be writable or not depending on some criteria. However, using the set_writable(False) function does nothing and I am able to change the variable value both inside the server and using an external client.

The only thing that works well is using different roles (admin/user) for every variable node in the server.

**To Reproduce**<br />

myvar = await dsObj.add_variable("ns=%d;i=%d" %(idx, varIdCnt), varname, 0, varianttype=uatypee , datatype= None)
await myvar.set_writable(False)

**Expected behavior**<br />

I would expect a variable node not to be writable using an external client.

**Version**<br />
Python Version: 3.9.12
opcua-asyncio Version: (master branch, 1.0)

AndreasHeine, 2022-12-15:
works as expected:

```python
import asyncio
import logging
from datetime import datetime
from asyncua import Server, ua

logging.basicConfig(level=logging.WARNING)
_logger = logging.getLogger('asyncua')

async def main():
     # Serversetup
    server = Server()
    await server.init()
    server.set_endpoint("opc.tcp://0.0.0.0:4840")

    async with server:

        idx = await server.register_namespace("my/test/ns")

        myvar = await server.nodes.objects.add_variable(f"ns={idx};i=1000", "writeable", 0, varianttype=ua.VariantType.Float , datatype= None)
        await myvar.set_writable()

        myvar2 = await server.nodes.objects.add_variable(f"ns={idx};i=2000", "notwritable", 0, varianttype=ua.VariantType.Float , datatype= None)
        await myvar2.set_writable(writable=False)

        while 1:
            await asyncio.sleep(1)

if __name__ == "__main__":
    asyncio.run(main())
```

tested with UaExpert!

[image]
[image]

ivan2033, 2022-12-16:
Thanks for your answer. I have created a variable node and the AccessLevel and UserAccessLevel attributes are only CurrentRead. However, if I try to change the variable value using UAExpert Data Access View tab, the value change is possible.

[image]

AndreasHeine, 2022-12-16:
my script is tested with python 3.11 and asyncua latest release 1.0.0 and it works as it should!

-[image]

are you able to write in my sample as well?

ivan2033, 2022-12-20:
Yes, I have tried your example and works as excepted for an anonymous user. But when I add a user/admin using diferent ways with a JSON file or a CertificateManager the variable RD/WR status does not matter anymore. An admin can write every variable, and what I would expect is only to be able to write variables that has WR status, not everything.

Thanks for your support.

`    

    import sys
    import asyncio
    import logging
    sys.path.insert(0, "..")
    from asyncua import Server
    from asyncua import ua

    from asyncua.server.user_managers import CertificateUserManager

    logging.basicConfig(level=logging.INFO)

    async def main():

        cert_user_manager = CertificateUserManager()
        await cert_user_manager.add_admin("Client/Certificates/peer-certificate-example-1.der", name='admin')
        await cert_user_manager.add_user("Client/Certificates/peer-certificate-example-2.der", name='user')  

        server = Server(user_manager=cert_user_manager)

        await server.init()

        server.set_endpoint("opc.tcp://0.0.0.0:4840/freeopcua/server/")
        server.set_security_policy([ua.SecurityPolicyType.Basic256Sha256_SignAndEncrypt],
                                permission_ruleset=None)
        # load server certificate and private key. This enables endpoints
        # with signing and encryption.

        await server.load_certificate("Server/Certificates/certificate.der")
        await server.load_private_key("Server/Certificates/key.pem")

        idx = 0

        # populating our address space
        myobj = await server.nodes.objects.add_object(idx, "MyObject")
        myvar_rd = await myobj.add_variable(idx, "MyVariable_RD", 0.0)
        myvar_wr = await myobj.add_variable(idx, "MyVariable_WR", 0.0)
        await myvar_rd.set_writable(False)  # Set MyVariable to be writable by clients
        await myvar_wr.set_writable(True)  # Set MyVariable to be writable by clients

        # starting!

        async with server:
            while True:
                await asyncio.sleep(1)
                current_val = await myvar_rd.get_value()
                count = current_val + 0.1
                await myvar_rd.write_value(count)

    if __name__ == "__main__":
        logging.basicConfig(level=logging.INFO)
        asyncio.run(main())
`

AndreasHeine, 2022-12-20:
ok thanks for the followup!

founde some thing in:
https://github.com/FreeOpcUa/opcua-asyncio/blob/bb03fed7eaef4eedb43bc078b72e740c7982e653/asyncua/server/address_space.py#L83

if i read that correct admins always can write:
```python

 async def write(self, params: ua.WriteParameters, user: User = User(role=UserRole.Admin)) -> List[ua.StatusCode]:
        # self.logger.debug("write %s as user %s", params, user)
        res: List[ua.StatusCode] = []
        for writevalue in params.NodesToWrite:
            if user.role != UserRole.Admin:
                if writevalue.AttributeId != ua.AttributeIds.Value:
                    res.append(ua.StatusCode(ua.StatusCodes.BadUserAccessDenied))
                    continue
                al = self._aspace.read_attribute_value(writevalue.NodeId, ua.AttributeIds.AccessLevel)
                ual = self._aspace.read_attribute_value(writevalue.NodeId, ua.AttributeIds.UserAccessLevel)
                if (
                    al.StatusCode is None
                    or al.Value is None
                    or ual.Value is None
                    or not al.StatusCode.is_good()
                    or not ua.ua_binary.test_bit(al.Value.Value, ua.AccessLevel.CurrentWrite)
                    or not ua.ua_binary.test_bit(ual.Value.Value, ua.AccessLevel.CurrentWrite)
                ):
                    res.append(ua.StatusCode(ua.StatusCodes.BadUserAccessDenied))
                    continue
            if writevalue.AttributeId == ua.AttributeIds.Value and self._aspace.force_server_timestamp:
                dv = dataclasses.replace(writevalue.Value, ServerTimestamp=datetime.utcnow(), ServerPicoseconds=None)
            else:
                dv = writevalue.Value
            res.append(
                await self._aspace.write_attribute_value(writevalue.NodeId, writevalue.AttributeId, dv)
            )
        return res
```

not sure if me need to move that part up in the code:
```python

                if (
                    al.StatusCode is None
                    or al.Value is None
                    or ual.Value is None
                    or not al.StatusCode.is_good()
                    or not ua.ua_binary.test_bit(al.Value.Value, ua.AccessLevel.CurrentWrite)
                    or not ua.ua_binary.test_bit(ual.Value.Value, ua.AccessLevel.CurrentWrite)
                ):
                    res.append(ua.StatusCode(ua.StatusCodes.BadUserAccessDenied))
                    continue
```

so we check writable before user!?
@oroulet any thoughts? do we get troulbe with writing from serverside? i mean a NOT writable node should be wwritable from serverside but not from clientside!?
maybe we need to add the "internal" role: "server"!?

schroeder-, 2022-12-21:
I think a internal role is the way. Because we don't want the user to change the value, but the server still want to update the value.

ivan2033, 2023-01-24:
I believe that the checking done in the "write" function inside "address_space.py" is to check if the user is able to write or not depending on certain conditions. But when I tried to debug, the write request does not reach that part of code because another check is done before when processing the message:

`2023-01-24 12:00:19,923 - PID  7692 - WARNING  - User(role=<UserRole.User: 3>, name='myuser') attempted to do something they are not permitted to do - uaprocessor - process_message - line:145`

I think it could be because the permission ruleset in which the User role does not have write permissions.

[image]

I do not know which is the best approach, but I can think about 2 different ways (I ask you to correct me if I am wrong):

1. Add a new role type for the server (InternalRole) for letting it write without restrictions internally. The AdminRole could potentially be able to write variables, if conditions are met
2. Add a new role type for an advanced user (AdvanceUser) who could write variables in case the proper conditions are met. The Admin role will remain for the server for internal use