FreeOpcUa/opcua-asyncio #1148: ValueError: error parsing asn1 value: ParseError { kind: ExtraData }

Alurith, 2022-12-16:
---
**Describe the bug** <br />
I tried to connect to a machine which needs Basic256Sha256 and SignAndEncrypt as security settings. I've generated my private key and the certificate using the following commands

`openssl genrsa -out key.pem 2048`
`openssl.exe req -x509 -days 365 -new -out certificate.pem -key key.pem -config ssl.conf`

This is the code I am running
```
import logging
import asyncio

from asyncua import Client

url = "opc.tcp://192.168.1.24:4840"

async def main():
    client = Client(url)
    client.set_user("the_username")
    client.set_password("the_password")

    await client.load_client_certificate("cert.pem")
    await client.load_private_key("key.pem")
    await client.set_security_string("Basic256Sha256,SignAndEncrypt,cert.pem,key.pem")

    client.session_timeout = 2000
    client.application_uri = "urn:opcua:python:server"

    async with client:
        print("async")
        root = client.nodes.root
        print(f"root -> {root}")
        objects = client.nodes.objects
        while True:
            print("childs og objects are: ", await objects.get_children())
            await asyncio.sleep(1)

if __name__ == "__main__":
    logging.basicConfig(level=logging.WARN)
    asyncio.run(main())

```
This is the traceback
```
(venv) PS D:\workspace\opcua_test> python.exe .\main.py
WARNING:asyncua.uaprotocol:updating client limits to: TransportLimits(max_recv_buffer=65536, max_send_buffer=65536, max_chunk_count=256, max_message_size=16777216)
Traceback (most recent call last):
  File "D:\workspace\opcua_test\main.py", line 33, in <module>
    asyncio.run(main())
  File "C:\Python310\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Python310\lib\asyncio\base_events.py", line 646, in run_until_complete
    return future.result()
  File "D:\workspace\opcua_test\main.py", line 16, in main
    await client.set_security_string("Basic256Sha256,SignAndEncrypt,cert.pem,key.pem")
  File "D:\workspace\opcua_test\venv\lib\site-packages\asyncua\client\client.py", line 167, in set_security_string
    return await self.set_security(
  File "D:\workspace\opcua_test\venv\lib\site-packages\asyncua\client\client.py", line 197, in set_security
    server_certificate = uacrypto.x509_from_der(endpoint.ServerCertificate)
  File "D:\workspace\opcua_test\venv\lib\site-packages\asyncua\crypto\uacrypto.py", line 37, in x509_from_der
    return x509.load_der_x509_certificate(data, default_backend())
  File "D:\workspace\opcua_test\venv\lib\site-packages\cryptography\x509\base.py", line 528, in load_der_x509_certificate
    return rust_x509.load_der_x509_certificate(data)
ValueError: error parsing asn1 value: ParseError { kind: ExtraData }
```

I've also tried using the certificate and the private key generated by UaExpert but it's the same.

**Version**<br />
Python-Version: 3.10.6
opcua-asyncio Version: 1.0.1

schroeder-, 2022-12-16:
If I am not wrong you need a DER formated public certificate.

Alurith, 2022-12-16:
@schroeder- I've also tried with a .der certificate created using this command: `openssl x509 -outform der -in cert.pem -out cert.der`, but it behaves the same.

schroeder-, 2022-12-16:
From the backtrack the server certificate seems to trigger the error. Maybe there is a bug in the  cryptography library which is used for x509. You could try to update the lib or open a issue on their github.

PR #1822 (2025-04-19): Include support for x509 certificate chains in DER format
# Fix certificate chain handling in x509_from_der function

## Description
This PR enhances the x509_from_der function to properly handle certificate chains by extracting and using the first certificate when a chain is detected.

When connecting to certain OPC UA servers that send certificate chains, the library currently fails with the error:

```ValueError: error parsing asn1 value: ParseError { kind: ExtraData }```

This happens because the cryptography library's `load_der_x509_certificate` function expects a single certificate, but  [... cut]
commit e3e02e3c (2025-07-22): Include support for x509 certificate chains in DER format (#1822)

* Fix certificate chain handling in x509_from_der function

This change enhances the x509_from_der function to properly handle certificate chains
by extracting and using the first certificate when a chain is detected.

When a certificate chain is received, the cryptography library raises a ValueError
with 'extradata' in the error message. This fix catches that specific error and
implements DER parsing to extract the first certificate from the chain.

This allows the library to work with certificate chains that some OPC UA serve [... cut]
closed 2022-12-29 by Alurith