FreeOpcUa/opcua-asyncio #1152: ValueError: 26 is not a valid VariantType

unregistered33, 2022-12-20:
Hi! When i try to run client_minimal_auth.py
I get this error in function browse_node:
  File "C:\Users\unreg\AppData\Local\Programs\Python\Python311\Lib\enum.py", line 1111, in __new__
    raise ve_exc
ValueError: 26 is not a valid VariantType

How i can fix it? Thanks! 
ua-server Prosys Sim Server

schroeder-, 2022-12-20:
A complete stacktrace would be nice.
I noticed that we don't fully comply to the spec: https://reference.opcfoundation.org/Core/Part6/v104/docs/5.2.2.16

> The Built-in Type Ids 26 through 31 are not currently assigned but may be used in the future. Decoders shall accept these IDs, assume the Value contains a ByteString and pass both onto the application. Encoders shall not use these IDs.

unregistered33, 2022-12-20:
Thanks for answer!
I found all posible types in uatypes.py
```
class VariantType(Enum):
    """
    The possible types of a variant.

    :ivar Null:
    :ivar Boolean:
    :ivar SByte:
    :ivar Byte:
    :ivar Int16:
    :ivar UInt16:
    :ivar Int32:
    :ivar UInt32:
    :ivar Int64:
    :ivar UInt64:
    :ivar Float:
    :ivar Double:
    :ivar String:
    :ivar DateTime:
    :ivar Guid:
    :ivar ByteString:
    :ivar XmlElement:
    :ivar NodeId:
    :ivar ExpandedNodeId:
    :ivar StatusCode:
    :ivar QualifiedName:
    :ivar LocalizedText:
    :ivar ExtensionObject:
    :ivar DataValue:
    :ivar Variant:
    :ivar DiagnosticInfo:
    """

    Null = 0
    Boolean = 1
    SByte = 2
    Byte = 3
    Int16 = 4
    UInt16 = 5
    Int32 = 6
    UInt32 = 7
    Int64 = 8
    UInt64 = 9
    Float = 10
    Double = 11
    String = 12
    DateTime = 13
    Guid = 14
    ByteString = 15
    XmlElement = 16
    NodeId = 17
    ExpandedNodeId = 18
    StatusCode = 19
    QualifiedName = 20
    LocalizedText = 21
    ExtensionObject = 22
    DataValue = 23
    Variant = 24
    DiagnosticInfo = 25
    **Unknown = 26     # added by me
    Unknown2 = 27     # added by me
    Unknown3 = 28     # added by me
    Unknown4 = 29     # added by me
    Unknown5 = 30**  # added by me
```

The error has certainly disappeared, but I understand that this solution is incorrect

schroeder-, 2022-12-22:
Created a fix in #1158

closed 2023-02-26 by schroeder-