FreeOpcUa/opcua-asyncio #1160: Tests are broken?

ratara, 2022-12-28:
Hi,
I think it is most likely a failure on my site but because the minimal server example works fine, it is a little bit strange.
When I run ./run_tests.sh most tests fail due to missing attributes.
`~/Dokumente/opcua-asyncio$ ./run-tests.sh 
============================================================= test session starts =============================================================
platform linux -- Python 3.10.6, pytest-7.2.0, pluggy-1.0.0
rootdir: /home/ralf/Dokumente/opcua-asyncio, configfile: pytest.ini, testpaths: tests
plugins: mock-3.10.0, asyncio-0.20.3, cov-4.0.0
asyncio: mode=strict
collected 581 items                                                                                                                           

tests/test_callback_service.py .                                                                                                        [  0%]
tests/test_client.py FFFF.FFFF                                                                                                          [  1%]
tests/test_common.py FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF [ 21%]
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF.                                                                             [ 31%]
tests/test_connections.py FF                                                                                                            [ 31%]
tests/test_crypto_connect.py FFFFFFFFssFFFFFFFFFFFFFF.                                                                                  [ 36%]
tests/test_custom_structures.py ..FFFsFFFF..FFFFFFFFFsFF                                                                                [ 40%]
tests/test_ha_client.py FFFFFFFFFFFFFFFF                                                                                                [ 43%]
tests/test_history.py FFFFFFFFFFFFFFFFFFFFFFFF                                                                                          [ 47%]
tests/test_history_events.py FFFFFFFFFFFFFFFFFF                                                                                         [ 50%]
tests/test_history_limits.py FFFFFF                                                                                                     [ 51%]
tests/test_permissions.py FFF                                                                                                           [ 51%]
tests/test_server.py FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF.FFFF                                                                        [ 59%]
tests/test_standard_address_space.py s                                                                                                  [ 59%]
tests/test_subscriptions.py FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF.                                 [ 72%]
tests/test_sync.py .............                                                                                                        [ 74%]
tests/test_tools.py FF                                                                                                                  [ 74%]
tests/test_uaerrors.py ...                                                                                                              [ 75%]
tests/test_uautils.py .                                                                                                                 [ 75%]
tests/test_unit.py ................................................................                                                     [ 86%]
tests/test_xml.py FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF                                        [100%]

================================================================== FAILURES ===================================================================
_____________________________________________________________ test_service_fault ______________________________________________________________

server = <async_generator object server at 0x7f09ca3f41c0>, admin_client = <async_generator object admin_client at 0x7f09c7ed0640>

    async def test_service_fault(server, admin_client):
        request = ua.ReadRequest()
        request.TypeId = ua.FourByteNodeId(999)  # bad type!
        with pytest.raises(ua.UaStatusCodeError):
>           await admin_client.uaclient.protocol.send_request(request)
E           AttributeError: 'async_generator' object has no attribute 'uaclient'

tests/test_client.py:17: AttributeError
___________________________________________________________ test_objects_anonymous ____________________________________________________________

server = <async_generator object server at 0x7f09ca3f41c0>, client = <async_generator object client at 0x7f09c6ab4240>

    async def test_objects_anonymous(server, client):
>       objects = client.nodes.objects
E       AttributeError: 'async_generator' object has no attribute 'nodes'
`
.. alot more errors like this ..

`FAILED tests/test_xml.py::test_xml_byte[server] - AttributeError: 'async_generator' object has no attribute 'opc'
FAILED tests/test_xml.py::test_xml_required_models_fail[server] - AttributeError: 'async_generator' object has no attribute 'opc'
FAILED tests/test_xml.py::test_xml_required_models_pass[server] - AttributeError: 'async_generator' object has no attribute 'opc'
=========================================== 485 failed, 91 passed, 5 skipped, 14 warnings in 55.78s ===========================================
`

Any ideas?

I installed dev_requirements.txt and requirements.txt.

ratara, 2023-01-14:
Well, it looks like the test setup is not performed properly because the client is never completely initialised.
I also did not find any setup for client that is run through. Any ideas?
[image]

oroulet, 2023-01-14:
Tests run on CI so most probably you are missing some dependencies. Look at dev requirements file

ratara, 2023-01-14:
Hi @oroulet,
thank you for answering. I was working on the project some time ago, so it worked there.
I reinstalled the virtual environment now multiple times and installed requirements.txt and dev_requirements.txt afterwards.

`python -m venv venv`
`source ./venv/bin/activate`
`pip install -r requirements.txt`
`pip install -r dev_requirements.txt`

Then I am running the tests by:
`./run_tests.sh`

I always get the same behaviour. Is there a special python version required? I am using 3.10.6.
I have no idea why I am stuck here.

oroulet, 2023-01-14:
Still looks like the same issue. Check pytest-asyncio. Really looks like that one is missing

ratara, 2023-01-14:
[image]

ratara, 2023-01-15:
Ok, ok. I think I solved the riddle. Obviously, it was my bad as you already stated. I discovered today in my merge request that my fork was far away from up to date. Sorry for that, I thought it was.
If this is solved after I fixed the conflicts, I will close this issue.

ratara, 2023-01-15:
It is.

closed 2023-01-15 by ratara