FreeOpcUa/opcua-asyncio #1170: Use opcua-asyncio client to communicate with Rockwell rslinx classic opc server

rwang0417, 2023-01-05:
Greetings!

I am trying to write an OPC client for controllogix PLCs. Can this library be used to write an OPC client in Python that can communicate with the OPC server configured using rslinx?

The main functions I want to achieve are searching for available OPC servers and connecting and reading data from the PLC. I tried to use the client example but don't know how to set the URL for rslinx OPC server. 

Any help or guidance is greatly appreciated!

Koseng, 2023-01-06:
Hi,
you need to find out whether it is an **opc** server (older protocol) or an **opc ua** server (newer protocol). If it is an opc ua server this is the right library. If it is an older opc server, then it is not.

rwang0417, 2023-01-06:
Thank you so much for pointing this out. It turns out that rslinx is opc server and factory talk is opc ua. Thanks again!

closed 2023-01-06 by rwang0417