FreeOpcUa/opcua-asyncio #1171: Discover local servers from client

rwang0417, 2023-01-10:
Greetings! 

Sorry to ask a basic question. Is it possible to search for servers on the same PC from the client? In the server configuration, the Discovery Service URL is opc.tcp://localhost:4840. The server URL that I am trying to find is opc.tcp://localhost:4990.

Here is what I tried and it gives me a ConnectionError. If I change the port to 4990 then it works and prints the Discovery URLs for all the endpoints. Is there a way to use the Discovery Service URL (port 4840) to find all the server URLs from the client on the same PC?
```
async def main():
    async with Client(url="opc.tcp://localhost:4840", timeout=10) as client:
        servers = await client.connect_and_find_servers()
        print(servers)
```

Thanks in advance!

schroeder-, 2023-01-11:
There is nothing you can do. If the server doesn't server on port 4840 you can't get the urls.

The only other way is to set up a LocalDiscoveryServer (LDS) or GlobalDiscoveryServer (GDS) and get the server to register it to a discovery server. This also only works if the server support this.

rwang0417, 2023-01-11:
Thanks so much for the reply! Here is a screenshot of the server configuration in FactoryTalk Linx. I can only find the Discovery service running but not sure if it is the LDS. 
[image]

I wanted to do this to achieve a search function similar to UAExpert. As in the screenshot below, it can automatically find servers running on my local PC. Is there a way to do this using this library and which function should I use? 
[image]
I would greatly appreciate any help or guidance you can give me! Cheers!

schroeder-, 2023-01-11:
Ok this should work if you target 4840. Maybe the log output of your code can give a glue.

rwang0417, 2023-01-11:
Hi @schroeder, here are the code and log output. If I use 4840, it gives me the ErrorMessage below. If I change to port 4990, then it can print the correct endpoint URLs. I am new to OPC UA and not sure if I used the correct functions to do the discovery. Please kindly give me some guidance on how to get it to work. Thank you in advance!
```
import asyncio
from asyncua import Client

# ======================== Class for OPC UA Client =========================
class OPCUAClient(object):
    async def get_endpoint_list(self):
        """find all endpoint urls using the discover url"""
        async with Client(url="opc.tcp://localhost:4840") as client:
            servers = await client.find_servers()
            for s in servers:
                print(s.DiscoveryUrls)

async def main():
    opcua = OPCUAClient()
    await opcua.get_endpoint_list()
```

```
updating client limits to: TransportLimits(max_recv_buffer=65536, max_send_buffer=65536, max_chunk_count=0, max_message_size=16777216)
Received an error: ErrorMessage(Error=StatusCode(value=2148204544), Reason=None)
Received an error: ErrorMessage(Error=StatusCode(value=2148204544), Reason=None)
close_secure_channel was called but connection is closed
disconnect_socket was called but connection is closed
Exception in callback <lambda>(<Task finishe...n is closed')>) at c:\rockwell\test\opcua\test_opcua1.py:86
handle: <Handle <lambda>(<Task finishe...n is closed')>) at c:\rockwell\test\opcua\test_opcua1.py:86>
Traceback (most recent call last):
  File "C:\Users\rwang15\Anaconda3\envs\TuneAIWebEnv\lib\asyncio\events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "c:\rockwell\test\opcua\test_opcua1.py", line 86, in <lambda>
    lambda t: print(f'Task done with result={t.result()}  << return val of main()'))
  File "c:\rockwell\test\opcua\test_opcua1.py", line 15, in main
    await opcua.get_endpoint_list()
  File "c:\rockwell\test\opcua\test_opcua1.py", line 8, in get_endpoint_list
    async with Client(url="opc.tcp://localhost:4840") as client:
  File "C:\Users\rwang15\Anaconda3\envs\TuneAIWebEnv\lib\site-packages\asyncua\client\client.py", line 86, in __aenter__
    await self.connect()
  File "C:\Users\rwang15\Anaconda3\envs\TuneAIWebEnv\lib\site-packages\asyncua\client\client.py", line 280, in connect
    await self.create_session()
  File "C:\Users\rwang15\Anaconda3\envs\TuneAIWebEnv\lib\site-packages\asyncua\client\client.py", line 413, in create_session
    response = await self.uaclient.create_session(params)
  File "C:\Users\rwang15\Anaconda3\envs\TuneAIWebEnv\lib\site-packages\asyncua\client\ua_client.py", line 328, in create_session
    data = await self.protocol.send_request(request)
  File "C:\Users\rwang15\Anaconda3\envs\TuneAIWebEnv\lib\site-packages\asyncua\client\ua_client.py", line 163, in send_request
    raise ConnectionError("Connection is closed") from None
ConnectionError: Connection is closed
```

schroeder-, 2023-01-12:
Errorcode 2148204544 -> ServiceNotSupport. So there is a limited opc ua server running. You could try to use find_servers_on_network instead of find_servers . 
If I find time I can try to test our server against the officel LDS from the OPC UA Foundation.

rwang0417, 2023-01-12:
Hi @schroeder, many thanks for the guidance! I tested both find_servers_on_network and find_servers on 4840 and they both gave the Errorcode that indicates ServiceNotSupport. My guess is that either the opc ua server I am using (FatoryTalk Linx) does not support this or I didn't configure it correctly. I'll look into that direction and if I find anything I'll comment here. Thanks again for all the help!

closed 2023-01-12 by rwang0417