FreeOpcUa/opcua-asyncio #1172: Declare ExtensionObject array variable

nakr94, 2023-01-11:
Hello,
i recently worked with ExtensionObjects arrays but i encountered a problem i am unable so solve.
I followed the [example ](https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/server-custom-structures-and-enums.py) and modified it to a certain extend. In the following ill only show parts of the program that are very specific or differ from the example. I can not provide the entirety of the code because of privacy issues.

Besides setting up the server just like the example i generated the new type RobotDataEntry:
`        robot_data_entry_node, encoding_nodes = await new_struct(
          server,
          namespace_node,
          "RobotDataEntry",
          [
              new_struct_field("name", ua.VariantType.String),
              new_struct_field("dataType", ua.VariantType.Int16),
              new_struct_field("stringValue", ua.VariantType.String),
              new_struct_field("intValue", ua.VariantType.Int32),
              new_struct_field("floatValue", ua.VariantType.Float),
          ],
      )
`

Next i created a variable with this type:
`
        default_value = construct_opcua_array(SimulationStatus.READY, "", "", "")
        variant = ua.Variant(default_value, ua.VariantType.ExtensionObject, is_array=True)

        node_id = Variable.READVAR.get_node_id(namespace_node)
        Variable.READVAR.node = await server.nodes.objects.add_variable(node_id, node_id, variant)
        await Variable.READVAR.node.write_value_rank(1)
        await Variable.READVAR.node.write_array_dimensions([N_REQUEST_FIELDS])
        await Variable.READVAR.node.set_writable(True)
`
where `N_REQUEST_FIELDS` is a constant set to 100. The `node_id` variable is just a string.

Writing is done with this code:
`
    async def set_value(self, value, array=False) -> None:
        await self.node.set_data_value(ua.DataValue(Value=ua.Variant(value, ua.VariantType.ExtensionObject, is_array=array)))
        
    await Variable.READVAR.set_value(construct_opcua_array(), array=True)
`
with `construct_opcua_array()` creating a list of `ua.RobotDataEntry` entries.

My question now is: Is this the proper way of defining and updating ExtensionObject arrays? Because clients connecting to the server holding the defined variable are only receiving an empty array which size of 100. The client is written in node-opcua.
I encountered a similar issue described in [this issue](https://github.com/mikakaraila/node-red-contrib-opcua), maybe it is related because the error on the client is the same saying "Internal Error: we are not expecting this dataType to be processed already".
[Here](https://github.com/node-opcua/node-opcua/blob/master/packages/node-opcua-client-dynamic-extension-object/source/resolve_dynamic_extension_object.ts) is a reference to the related code section in line 47.

If you need any further information or code segments, let me know.
Thanks

schroeder-, 2023-01-11:
Did you try to read the value with UAExpert? This should be the first step, If you UAExpert can read this, your code should be correct.

Also can the client read a scalar value with your datatype? Maybe the client can't read datatype description. You could try the pre 1.04 structs see the example: https://github.com/FreeOpcUa/opcua-asyncio/blob/master/examples/server-custom-structures-before-1.04.py

nakr94, 2023-01-11:
I tried browsing this server with the FreeOpcua client and it worked as intended. I tried to rollback to the pre 1.04 delcaration but it did not work as well. Even the FreeOpcua client has an issue showing the values of the structure fields when i used the declaration pre 1.04.

schroeder-, 2023-01-11:
I tried this code and it works with UAExpert:

```Python
   await new_struct(server, idx, "TestStruct", [
        new_struct_field("MyBool", ua.VariantType.Boolean),
        new_struct_field("MyUInt32", ua.VariantType.UInt32),
    ])
    await server.load_data_type_definitions()
    v = [ua.TestStruct(True, 1), ua.TestStruct(False, 2), ua.TestStruct(False, 3)]
    var = await server.nodes.objects.add_variable(idx, "array_node", ua.Variant(v, ua.VariantType.ExtensionObject))
    await var.write_value_rank(1)
    await var.write_array_dimensions([3])
    await var.set_writable()
    async with server:
        while True:
            await asyncio.sleep(1)
```

nakr94, 2023-01-12:
Is there any limitations on the size of the array?

nakr94, 2023-01-12:
And maybe another question: Is it possible to store JSON data directly in OPC UA?

schroeder-, 2023-01-12:
There is a limit for networkmessages, this depends on the client/server if your structure excesses 10 MB you should consider making each value a independent node.

If you want to store JSON you can just store it as String.

nakr94, 2023-01-12:
Is there no native support for JSON? If i write the message as string, i have to parse it on the receiving end.

schroeder-, 2023-01-12:
Yes. I theory ExtensionObjects can be encoded/decoded as Json but we don't support that and most other OPC UA implementations. Also you still have to express your JSON content as OPC UA Structure.

nakr94, 2023-01-12:
ok thank you, i think this is related to node opcua. I will create a related issue in their issue section.

closed 2023-01-12 by nakr94