FreeOpcUa/opcua-asyncio #1175: Data type for object nodes

rwang0417, 2023-01-11:
Greetings!

Is it possible to read the data type of object nodes? I am trying to find all the nodes with the type 'PID_Enhanced'. I have some tags(UDT) with this type, but as they are UDTs, they have member variables. When I browse the nodes using opc ua, These PIDE nodes are object nodes so they do not have the data type attribute. I was wondering if the type 'PID_Enhanced' can be read from PLC? If so, how can I read it?

Here is what I see from the PLC code. My target is to get the data type PID_Enhanced using opc ua.
[image]

Thanks in advance!

schroeder-, 2023-01-12:
Sry I miss read you want the type description of a node. 

``` Python
nid = await node.read_type_definition()  # NodeId of the type 
# get the name display name of the NodeId
dt_node = client.get_node(self, nid)
name = await dt_node.read_display_name()
```

rwang0417, 2023-01-12:
Hi @schroeder, thank you so much for your help! I didn't know I can treat the type as a node as well. 

Unfortunately, it turns out that all the UDTs with different data types in PLC have the same type node as shown below. For example, the UDT with data type PID_ENHANCED and another UDT with data type Timer have the same NodeId 61 and the same display name FolderType.
[image]

I checked the Types folder in the Address Space using UAExpert, I couldn't find the PID_ENHANCED type here. Actually none of the UDT data types in PLC can be found here. Does this mean that my OPC UA server (FactoryTalk Linx) does not get the UDT types and there is no way to read it from the client?
[image]

My settings: PLC(CompactLogix) -> OPCUA server(FactoryTalk Linx) -> OPCUA client(asyncua/UAExpert).

rwang0417, 2023-01-12:
Hi @schroeder, I think I know what I missed. I need to enable the tag list access in the FactoryTalk Linx Gateway Configuration server configuration and add the tags including UDTs in the tag group.
[image]

Now both of the code snippets you shared above work. The difference is that the first one shows PID_ENHANCEDDataType while the second one shows PID_ENHANCEDVariableType. Both can achieve my target. Many many thanks for the great help and for the awesome library!

closed 2023-01-12 by rwang0417