FreeOpcUa/opcua-asyncio #1176: Catch exception when server shutdown

unregistered33, 2023-01-12:
Hi!
I have this code:
```
    try:
        subscription = await uaclient.create_subscription(period, handler)
        await subscription.subscribe_data_change(nodes_dict)
        while True:
            await uaclient.check_connection()  # Throws a exception if connection is lost
            data = await queue.get()
            print(data)
            queue.task_done()
    except (ConnectionError, ua.UaError):
        print('Something wrong')
```  
        

All works fine, but if i manually shutdown the server i'v got this error:

```
ServiceFault (BadNoSubscription, diagnostics: DiagnosticInfo(SymbolicId=None, NamespaceURI=None, Locale=None, LocalizedText=None, AdditionalInfo=None, InnerStatusCode=None, InnerDiagnosticInfo=None)) from server received  in response to PublishRequest
Exception calling status change handler
Traceback (most recent call last):
  File "C:\Workspace\Python\Katla_micros\venv\Lib\site-packages\asyncua\client\client.py", line 457, in _monitor_server_loop
    _ = await self.nodes.server_state.read_value()
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Workspace\Python\Katla_micros\venv\Lib\site-packages\asyncua\common\node.py", line 179, in read_value
    result = await self.read_data_value()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Workspace\Python\Katla_micros\venv\Lib\site-packages\asyncua\common\node.py", line 190, in read_data_value
    return await self.read_attribute(ua.AttributeIds.Value, None, raise_on_bad_status)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Workspace\Python\Katla_micros\venv\Lib\site-packages\asyncua\common\node.py", line 304, in read_attribute
    result = await self.session.read(params)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Workspace\Python\Katla_micros\venv\Lib\site-packages\asyncua\client\ua_client.py", line 397, in read
    data = await self.protocol.send_request(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Workspace\Python\Katla_micros\venv\Lib\site-packages\asyncua\client\ua_client.py", line 163, in send_request
    raise ConnectionError("Connection is closed") from None
ConnectionError: Connection is closed

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Workspace\Python\Katla_micros\venv\Lib\site-packages\asyncua\common\subscription.py", line 173, in _call_status
    if asyncio.iscoroutinefunction(self._handler.status_change_notification):
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'SubscriptionHandler' object has no attribute 'status_change_notification'
```

How i can catch this?
Ideally, I need to try to reconnect to the server after some time. Thanks for reply!

oroulet, 2023-01-12:
I thought the second one was fixed. but add that method to your handler and will not get that warning.
The first exception is normal since you got some unexpected stop of server

unregistered33, 2023-01-12:
Im little confused
My handler:
```
class SubscriptionHandler:
    def datachange_notification(self, node: Node, val, data):
            queue.put_nowait({'node': node.nodeid.to_string(),
                          'value': val,
                          'timestamp': str(data.monitored_item.Value.SourceTimestamp),
                          'status_code': data.monitored_item.Value.StatusCode.value})
```
point me, please, where i must add method?

schroeder-, 2023-01-12:
If you don't want to upgrade add this to your handler:
 
```Python
    def status_change_notification(self, status: ua.StatusChangeNotification):
        """
        called for every status change notification from server
        """
        pass
```

unregistered33, 2023-01-12:
Thanks for reply! im using 1.0.1. After adding your code, the error stopped appearing. But the process still works. Where i can handle this error, for example, output a message and abort the process? My main code:
```
async def ua_process(client: UAClient):
    if not client.ua_client:
        return
    handler = SubscriptionHandler()

    nodes_list = client.nodes.split(',')
    nodes_dict = []
    for n in nodes_list:
        try:
            node = client.ua_client.get_node(n)
            nodes_dict.append(node)
        except ua.UaError:
            print(f'Node <{n}> not found. Ignored.')

    try:
        subscription = await client.ua_client.create_subscription(client.period, handler)

        await subscription.subscribe_data_change(nodes_dict)
        while True:
            await client.ua_client.check_connection() 
            data = await queue.get()
            print(data)
            queue.task_done()
    except (ConnectionError, ua.UaError):
        raise KeyboardInterrupt
```
This code running in loop:
```
    try:
        loop = asyncio.new_event_loop()
        asyncio.set_event_loop(loop)
        loop.run_until_complete(ua_connect(client))
        loop.run_until_complete(ua_process(client))
    except KeyboardInterrupt:
        pass
    finally:
        if client.ua_client:
            print(loop.run_until_complete(ua_disconnect(client)))
        else:
            return 'No client connected. Process halted'
```

closed 2023-01-16 by unregistered33