FreeOpcUa/opcua-asyncio #1178: Reading values from queue

unregistered33, 2023-01-14:
Hi, guys!
I'm trying to put the received values in a queue and then get them in a separate thread. This is necessary in order to call the read function from FastApi. But when I try to stop the thread via an event (stop_read() function), the process freezes on the 'reading_thread.join()' (stop_read() function). I noticed that the problem is in the line 'item = _queue.get()' (read_data(queue) function). If i comment it out, then the stream ends normally. Any hint will be helpful. Thanks!

```
from queue import Queue
from threading import Thread, Event

queue = Queue()
event = Event()
reading_thread = None

class SubscriptionHandler:
    def datachange_notification(self, node: Node, val, data):
        queue.put_nowait({'node': node.nodeid.to_string(),  'value': val})

    def status_change_notification(self, status: ua.StatusChangeNotification):
        pass

def read_data(_queue): # thread function
    while not event.is_set():
        item = _queue.get() # when i comment this
        # print(f'just strimg') # and uncommet this, the thread terminates normally
        
    queue.task_done()
    client_shell.status = Status.CONNECTED
    print('Reading process stopped.')

async def read():  # calling from FastApi
    handler = SubscriptionHandler()
    node = client_shell.ua_client.get_node('ns=3;s=1008')
    try:
        subscription = await client_shell.ua_client.create_subscription(client_shell.period, handler)
        await subscription.subscribe_data_change(node)

        global reading_thread
        reading_thread = Thread(target=read_data, args=(queue,))
        reading_thread.start()
       return 'Process started'
    except (ConnectionError, ua.UaError) as e:
        await disconnect_from_server()
        return 'Disconnect from server. ' + str(e)

def stop_read():  # try to stop reading thread. Calling from FastApi
    event.set()
    reading_thread.join() # wait while thread finished

```

unregistered33, 2023-01-14:
I think this is happening because subscription continues to work, is there a way to stop the subscription?

AndreasHeine, 2023-01-14:
unsubscribe or delete the subscription!

unregistered33, 2023-01-14:
Sorry, i was not attentive. Im add this lines to my read() function to see quick reaction and have this error and process freez:
       ```
 
         try:
        subscription = await client_shell.ua_client.create_subscription(client_shell.period, handler)
        await subscription.subscribe_data_change(nodes_dict)

        await client_shell.ua_client.check_connection()
        global reading_thread
        reading_thread = Thread(target=read_data, args=(queue,))
        reading_thread.start()
        await asyncio.sleep(10)
        await subscription.unsubscribe(handler) # with await subscription.delete() the same result
    except (ConnectionError, ua.UaError) as e:
        await disconnect_from_server()
        return 'Disconnect from server. ' + str(e)
```

ua_binary.py > _Primitive1 > pack_array > data: <micros.ua_app.ua_micro.SubscriptionHandler object at 0x00000122D58D88D0> is not a instance of "list"!
ServiceFault (BadNothingToDo, diagnostics: DiagnosticInfo(SymbolicId=None, NamespaceURI=None, Locale=None, LocalizedText=None, AdditionalInfo=None, InnerStatusCode=None, InnerDiagnosticInfo=None)) from server received  in response to DeleteMonitoredItemsRequest

Upd. I replaced the queue with a list, and everything seems to be working fine. Have no idea why with queue its not working

AndreasHeine, 2023-01-14:
you should not mix asyncio and threading... use asyncio queue instead and no threading

unregistered33, 2023-01-15:
Yes, you're right. I changed the code to asynchronous reading from async queue, but when I tried to stop the task, the process still hung. The problem was that first I deleted the subscription, and then I set an event to complete the reading task. The read code was stuck when accessing the queue (queue.get()). If i first wait for the reading task to finish, and then delete the subscription, then everything is fine.  Thanks!

AndreasHeine, 2023-01-15:
create two tasks
one for consume from queue and
one timeout task 

then asyncio.wait(tasks) with first complete!

AndreasHeine, 2023-01-15:
another way is to while shutdown event is not set await queue.get()

unregistered33, 2023-01-16:
> another way is to while shutdown event is not set await queue.get()

This is what I'm using now:
```
    async def read_data(self, target=None):
        self.status = Status.PROCESSED
        while not self.event.is_set():
            item = await self.queue.get()
            target(item)
            self.queue.task_done()
        self.status = Status.CONNECTED
        self.event.clear()
```