FreeOpcUa/opcua-asyncio #1180: WARNING asyncua.uaprotocol: updating server limits to...

rnestler, 2023-01-16:
**Describe the bug** <br />
After an upgrade to `0.9.98` I see the following warnings:
```
2023-01-16 14:33:01,068 WARNING  asyncua.uaprotocol: updating server limits to: TransportLimits(max_recv_buffer=65535, max_send_buffer=65535, max_chunk_count=1601, max_message_size=104857600)
2023-01-16 14:33:01,077 WARNING  asyncua.uaprotocol: updating server limits to: TransportLimits(max_recv_buffer=65535, max_send_buffer=65535, max_chunk_count=1601, max_message_size=104857600)
...
```

**To Reproduce**<br />
 * Start a server (`python examples/server-minimal.py`)
 * Connect with a client to the server (`python examples/client-minimal.py`)
 

**Expected behavior**<br />

I don't expect to see warning for the regular usage.

**Version**
I tested with two sets of versions
Python-Version: 3.9.15
opcua-asyncio Version: 0.9.98

Python-Version: 3.10.9
opcua-asyncio Version: 2c78916fccef7cc72e38ad5c6308400222e0e2aa (current master)

AndreasHeine, 2023-01-16:
thats the result of the hello / ack negotiation between client and server - you can ignore that its just telling you that the requested settings have been revised

closed 2023-01-20 by oroulet

ZionDeng, 2023-02-12:
Hi, is there a way to hide this? I need to connect and disconnect very frequently but I don't want to see the log in console that often.

AndreasHeine, 2023-02-12:
set the loglevel to info

PR #1181 (2023-01-20): Use info log level for client and server limit update message
Fixes #1180
commit d93a9652 (2023-01-20): Use info log level for client and server limit update message

Fixes #1180