FreeOpcUa/opcua-asyncio #1184: Data loss while reading value by a fixed rate using subscription

rwang0417, 2023-01-19:
Greetings!

I have a tag in PLC that increases by 1 every 10ms as shown below. My target is to read this tag(OPC_Count) every 1000ms with opc ua client. I was able to read the value but saw some data loss. I am using FactoryTalk Linx as the opc ua server and I wrote a client using this library.
[image]

I created a subscription to this node and set the publishing interval to 1000ms. Below is the data loss I observed. As the value increases by 1 every 10ms, I expect to see the number increase by 100 every second. However, sometimes it increases by 200 between two prints. When I sleep for 100 seconds, I can only get 90 ish data instead of 100 data.
[image]

Is there any method to fix this? Here is the code I am using. I hope to make the actual sampling interval as accurate as possible. Thanks in advance!
```
async with Client(url=endpoint_url) as client:
    handler = SubscriptionHandler()
    node = client.get_node("ns=2;s=TagGroup01#::[CompactLogix]Program:MainProgram.OPC_Count")
    subscription = await client.create_subscription(1000, handler)
    nodes = [node] 
    await subscription.subscribe_data_change(nodes=nodes, queuesize=1)
    await asyncio.sleep(100)
    print(len(handler.data_history))

class SubscriptionHandler:
    def __init__(self):
        self.data_history = []

    def datachange_notification(self, node: Node, value, data):
        self.data_history.append(value)
        print('OPC Count: ', value)
```

AndreasHeine, 2023-01-19:
i have tested it a while ago and there is no dataloss... your subscription settings or monitored items settings are not correct! i mean the client just tells the server how to sample the monitored items and then it sends cyclic publishrequests and get the queued changes (in a nutshell) you should increase queuesize to 1000ms/10ms = 100 so at least the queue fits all expected changes in one second maybe thats the issue server will discard changes if queue is full ...

AndreasHeine, 2023-01-19:
you should not use a publishing interval smaller 500ms its just the transport intervall of queued changes

the sampling rate is just how often the server should look for changes if you use 0 the server does it as fast as possible

the queue should always be greater then the expected number of changes according to the samplerate of the monitored item otherwise the server discard changes and you have data loss

another thing to point out is that the subscription setting you choose might be not supported by the server and in those cases the server revises them!

rwang0417, 2023-01-20:
Thank you so much Andreas! I consulted some internal resource and it turns out that the controller itself has limit on the sampling rate. In my case, 1000ms sampling rate should be the best choice and with the new settings I hardly see any data loss at this rate. Thanks again for all the kind help!

closed 2023-03-14 by schroeder-