FreeOpcUa/opcua-asyncio #1198: Read_Raw_History Continuation limit of 10000 needs to be increased to 1000000 or make this an optional

ImerysOTDataAndArchitecture, 2023-02-07:
Hello, 

We are working with Read_Raw_History but we see a continuation point with limit of 10000 which is not sufficient for us. 

Is there a way to make this optional or iterate through the full list to get full data set from Read Raw History. 

Example : Let OPCUA gives a dataset of 50000 records but Read Raw History is giving us only 10000 data points that 50000 data set. Is there a way to loop through the records and get all the 50000 records in Python Client ?

AndreasHeine, 2023-02-07:
there is the concept of ServerCapabilities and OperationalLimits which can be adjusted!

ImerysOTDataAndArchitecture, 2023-02-07:
Hello,

Is there any documentation or code references you can help me with?

Thanks
Nandan Chapa
Industrial IT - Global Data & Architecture
Office : +1 7706453505 Mobile : +1 316-821-6061

<https://www.surveymonkey.com/r/BMR8H92?Employee_ID=[NandanChapa]>

-- 
*The contents of this email message and any attachments are intended solely 
for the addressee**(s). It m**ay contain confidential and/or privileged 
information and may be legally protected from further disclosure. If you 
are not the intended recipient of this message or their agent, or if this 
message has been addressed to you in error, please immediately alert the 
sender by reply email and then delete this message and any attachments. If 
you are not the intended recipient, you are hereby notified that any use, 
dissemination, copying, or storage of this message or its attachments is 
strictly prohibited and may be unlawful. *

AndreasHeine, 2023-02-07:
What is the Server? python aswell?

ImerysOTDataAndArchitecture, 2023-02-07:
Hello,

We are using OPCUA Server. It is not Python though

Thanks
Nandan Chapa
Industrial IT - Global Data & Architecture
Office : +1 7706453505 Mobile : +1 316-821-6061

<https://www.surveymonkey.com/r/BMR8H92?Employee_ID=[NandanChapa]>

-- 
*The contents of this email message and any attachments are intended solely 
for the addressee**(s). It m**ay contain confidential and/or privileged 
information and may be legally protected from further disclosure. If you 
are not the intended recipient of this message or their agent, or if this 
message has been addressed to you in error, please immediately alert the 
sender by reply email and then delete this message and any attachments. If 
you are not the intended recipient, you are hereby notified that any use, 
dissemination, copying, or storage of this message or its attachments is 
strictly prohibited and may be unlawful. *

schroeder-, 2023-02-07:
From the opc ua side it is possible to call ReadHistoryRaw with the ContinueationPoint and get the next 10000 records. I don't know if how we provide it on the client side, but using uaclient it should be possible todo.

AndreasHeine, 2023-02-07:
If the OPC UA Server sets its limit to 10k you cant change it from OPC UA Client Side... sorry!

you need to check those values:
[image]

Reference:
https://reference.opcfoundation.org/v104/Core/docs/Part5/6.3.11/

The [MaxNodesPerHistoryReadData](https://reference.opcfoundation.org/search/21?t=MaxNodesPerHistoryReadData) [Property](https://reference.opcfoundation.org/search/21?t=Property)indicates the maximum size of the nodesToRead array when a [Client](https://reference.opcfoundation.org/search/21?t=Client)calls the HistoryRead [Service](https://reference.opcfoundation.org/search/21?t=Service)using the historyReadDetails RAW, PROCESSED, MODIFIED or ATTIME.

AndreasHeine, 2023-02-07:
continuation points are provided by the server but you can use the timestamp of the first value to get older values! so you need to make multiple requests due to the OperationalLimits!

ImerysOTDataAndArchitecture, 2023-02-07:
Andreas,

Below is the screenshot from Server Capabilities _OperationLimits.

[image: image.png]

Is there a code reference on how to retrieve the next set of data from a
Python Asyncio client ?

Thanks
Nandan Chapa
Industrial IT - Global Data & Architecture
Office : +1 7706453505 Mobile : +1 316-821-6061

<https://www.surveymonkey.com/r/BMR8H92?Employee_ID=[NandanChapa]>

-- 
*The contents of this email message and any attachments are intended solely 
for the addressee**(s). It m**ay contain confidential and/or privileged 
information and may be legally protected from further disclosure. If you 
are not the intended recipient of this message or their agent, or if this 
message has been addressed to you in error, please immediately alert the 
sender by reply email and then delete this message and any attachments. If 
you are not the intended recipient, you are hereby notified that any use, 
dissemination, copying, or storage of this message or its attachments is 
strictly prohibited and may be unlawful. *

AndreasHeine, 2023-02-07:
well @schroeder- is right! if make the historyread you will recv a continuationpoint ;)

https://reference.opcfoundation.org/Core/Part4/v104/docs/5.10.3

<html><body>
<a name="Table56">

ContinuationPoint | This parameter is used only if the number of values  to be returned is too large to be returned in a single response or if  the timeout provided as hint by the Clientis close to expiring and not all nodes have been processed.                         When this parameter is not used, its value is null.                         Serversshall support at least one continuation point per Session. Serversspecify a max history continuation points per Sessionin the Servercapabilities Objectdefined in OPC 10000-5. A continuation point shall remain active until the Clientpasses the continuation point to HistoryReador the Sessionis closed. If the max continuation points have been reached the oldest continuation point shall be reset.
-- | --

</a>
</body>
</html>

schroeder-, 2023-02-07:
This is all done of the function **read_raw_history**. Otherwise the server does handle something wrong or our code is wrong.
``` Python
node = client.get_node(....)
results = await node.read_raw_history(starttime, endtime, numvalues=0, return_bounds=False)
```

AndreasHeine, 2023-02-07:
@schroeder- depends on if you call it from ua_client.py its not ^^

ImerysOTDataAndArchitecture, 2023-02-07:
Hello,

Below is my code which only gets 10000 records whereas the OPC UA Server
Logs state 58772+ records

        client.connect()
        myvar = client.get_node('ns=60;i=6031')
        #print(myvar)
        start_time = datetime.now() - timedelta(days=30)
        end_time = datetime.now()
        print("Start_Time:",start_time)
        print("End_Time:",end_time)
        print("Gathering the Data...")
        data = myvar.read_raw_history(start_time, end_time)

*Output:*

D:\Python\Python39\python.exe
D:/Nandan/PythonScripts/Read_Raw_History_OPCUA_Asyncio.py
Requested session timeout to be 3600000ms, got 600000ms instead
Start_Time: 2023-01-08 11:29:01.762148
End_Time: 2023-02-07 11:29:01.762148
Gathering the Data...
Count of Data: 10000
Numpy Array: [19.92241287 19.92586136 19.99489021 ... 22.58795166
22.80921555
 23.02400398]
Numpy Array Sum: 535248.6299600601
Numpy Array Mean: 53.524862996006014
Numpy Array Min Value: 0.0
Numpy Array Max Value: 33107.1875

*OPC UA Logs:*

2023-02-07T16:29:10.210Z SERVER_ERROR
CSessionServer::ProcessQueuedHistoryReadMessage>VfiHistoryReadEx Succeeded
in 1735ms
2023-02-07T16:29:10.211Z SERVER_ERROR
CSessionServer::ProcessQueuedHistoryReadMessage>VfiHistoryReadEx return
58772 values for ns=60;i=6031
2023-02-07T16:29:10.212Z SERVER_ERROR
CSessionServer::ProcessQueuedHistoryReadMessage>VfiHistoryReadEx create a
CHistoryContinuationPoint from index 10000-->20000
2023-02-07T16:29:10.214Z SERVER_ERROR
CSessionServer::ProcessQueuedHistoryReadMessage>VfiHistoryReadEx create a
CHistoryContinuationPoint from index 20001-->30001
2023-02-07T16:29:10.215Z SERVER_ERROR
CSessionServer::ProcessQueuedHistoryReadMessage>VfiHistoryReadEx create a
CHistoryContinuationPoint from index 30002-->40002
2023-02-07T16:29:10.217Z SERVER_ERROR
CSessionServer::ProcessQueuedHistoryReadMessage>VfiHistoryReadEx create a
CHistoryContinuationPoint from index 40003-->50003
2023-02-07T16:29:10.219Z SERVER_ERROR
CSessionServer::ProcessQueuedHistoryReadMessage>VfiHistoryReadEx create a
CHistoryContinuationPoint from index 50004-->58772

Thanks
Nandan Chapa
Industrial IT - Global Data & Architecture
Office : +1 7706453505 Mobile : +1 316-821-6061

<https://www.surveymonkey.com/r/BMR8H92?Employee_ID=[NandanChapa]>

-- 
*The contents of this email message and any attachments are intended solely 
for the addressee**(s). It m**ay contain confidential and/or privileged 
information and may be legally protected from further disclosure. If you 
are not the intended recipient of this message or their agent, or if this 
message has been addressed to you in error, please immediately alert the 
sender by reply email and then delete this message and any attachments. If 
you are not the intended recipient, you are hereby notified that any use, 
dissemination, copying, or storage of this message or its attachments is 
strictly prohibited and may be unlawful. *

ImerysOTDataAndArchitecture, 2023-02-08:
> Which version of asyncua are you using? The behavior was added in Version
> v0.9.90.

Hello,

Is there a way to find the version of the Asyncio library ? I used Pycharm
to get the library installed with the Python 3.9 interpreter.

As per Github V1.0.0 is the latest.

Thanks
Nandan Chapa
Industrial IT - Global Data & Architecture
Office : +1 7706453505 Mobile : +1 316-821-6061

<https://www.surveymonkey.com/r/BMR8H92?Employee_ID=[NandanChapa]>

-- 
*The contents of this email message and any attachments are intended solely 
for the addressee**(s). It m**ay contain confidential and/or privileged 
information and may be legally protected from further disclosure. If you 
are not the intended recipient of this message or their agent, or if this 
message has been addressed to you in error, please immediately alert the 
sender by reply email and then delete this message and any attachments. If 
you are not the intended recipient, you are hereby notified that any use, 
dissemination, copying, or storage of this message or its attachments is 
strictly prohibited and may be unlawful. *