FreeOpcUa/opcua-asyncio #1200: no events in alarms and conditions example

dremsol, 2023-02-10:
When running `client_to_uanet_alarm_conditions.py` using the example alarms and conditions server `opc.tcp://opcua.demo-this.com:62544/Quickstarts/AlarmConditionServer` no events are coming in. 

When subscribing to the Server object in UaExpert this works without a problem. Could i be missing something or should this example work out of the box?

AndreasHeine, 2023-02-10:
it should work out of the box...

does it work with this server? opc.tcp://opcua.umati.app:4843 <- online avalible!

dremsol, 2023-02-11:
Thanks for the quick reply. Your suggested server indeed works with the example script (although i cant connect with UaExpert, it's trying to resolve discovery URL to localhost.) 

It's still unclear to me why the other server doesn't work. It should catch all events based on conditionType right?

AndreasHeine, 2023-02-11:
the example subscribes all alarmconditions so a subtype of conditiontype

```python
handle = await sub.subscribe_alarms_and_conditions(client.nodes.server, alarmConditionType) 
```

AndreasHeine, 2023-02-11:
some servers dont emit via the server object... usually you have a notifier chain via references
[image] 
but this is just a guess that the chain is incomplete and ua expert somehow gets it still...

dremsol, 2023-02-11:
Thanks for the suggestion, although following the chain from `Server` node down the tree using `HasNotifier` and `HasEventSource` seems good?

I also have the same issues with a Beckhoff OPCUA A&C server.

[image]

AndreasHeine, 2023-02-11:
does it work if you subscribe the actaual alarm object instead of server?

dremsol, 2023-02-11:
I cant subscribe to either `SouthMotor` or `WestTank` in the example above. I'm only able to move the 'Blue' object into event view

dremsol, 2023-02-13:
Hi @AndreasHeine, I've figured out the reason why I did not receive the A&C from Beckhoff. I've subscribed on the wrong types... Thanks for your help pointing me in the right direction. I will close this issue!

closed 2023-02-13 by dremsol