FreeOpcUa/opcua-asyncio #1204: UaStatusCodeError superclass confuses typecheckers

loehnertj, 2023-02-14:
**Describe the bug** <br />

In https://github.com/FreeOpcUa/opcua-asyncio/blob/master/asyncua/ua/uaerrors/_base.py, ``UaStatusCodeError`` is defined as:

``class UaStatusCodeError(_AutoRegister("Meta", (UaError,), {})): ...``

This confuses my typechecker (PyLance / VSCode). It yells at me that I cannot except on e.g. ``BadTooManyOperations`` because it thinks that it is not a proper ``Exception`` subclass. I get ugly red squiggly lines telling me I made a mistake, which I didn't 😢.

The ``_AutoRegister`` magic is purely internal to ``_base.py``. It seems like leftover to me, that is not useful for anything obvious anymore.

**Expected behavior**<br />

Can we just derive ``UaStatusCodeError`` from ``UaError`` in plain language?

``class UaStatusCodeError(UaError): ...``

**Version**<br />
Python-Version: 3.9<br />
opcua-asyncio Version (e.g. master branch, 0.9): 1.0.1

schroeder-, 2023-02-19:
The magic is need because every error code needs to be registered in a dict, before any instanced is created. If you have a better solution for this problem we can change it.

loehnertj, 2023-02-20:
Gave it a try - please review.

closed 2023-02-26 by schroeder-