FreeOpcUa/opcua-asyncio #1209: Impossible to stop the server when port is already in use

lbckmnn, 2023-02-15:
**Describe the bug** <br />
`server.stop()` raises, when server starting failed.
This is especially problematic when the sync server is used, because the process does somehow not end properly.

**To Reproduce**<br />
Consider the following code:
``` python
import time

from asyncua.sync import Server

if __name__ == "__main__":
    server = Server()
    server.set_endpoint("opc.tcp://127.0.0.1:4840")
    try:
        server.start()
        print("started server")
        while True:
            time.sleep(1)
    finally:
        print("stopping")
        server.stop()

```
Run this simultaneously in two processes.
The second process is obviously not able to bind the same port again, and `server.start()` raises an `OsError`.
However, in this case, the `server.stop()` in the finally block does also raise, and even worse, something seems to hang (threadloop is not ended?) and the process does only terminate when strg+c is pressed.

Output:
```
venv ❯ python minimal_opcua.py
Endpoints other than open requested but private key and certificate are not set.
OPC UA Server(opc.tcp://127.0.0.1:4840) error starting server
Traceback (most recent call last):
  File "/home/user/coding/voraus-system-control-py/venv/lib/python3.9/site-packages/asyncua/server/server.py", line 441, in start
    await self.bserver.start()
  File "/home/user/coding/voraus-system-control-py/venv/lib/python3.9/site-packages/asyncua/server/binary_server_asyncio.py", line 139, in start
    self._server = await asyncio.get_running_loop().create_server(self._make_protocol, self.hostname, self.port)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 1506, in create_server
    raise OSError(err.errno, 'error while attempting '
OSError: [Errno 98] error while attempting to bind on address ('127.0.0.1', 4840): address already in use
stopping
Traceback (most recent call last):
  File "/home/user/coding/voraus-system-control-py/minimal_opcua.py", line 9, in <module>
    server.start()
  File "/home/user/coding/voraus-system-control-py/venv/lib/python3.9/site-packages/asyncua/sync.py", line 94, in wrapper
    result = self.tloop.post(aio_func(*args, **kwargs))
  File "/home/user/coding/voraus-system-control-py/venv/lib/python3.9/site-packages/asyncua/sync.py", line 52, in post
    return futur.result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 446, in result
    return self.__get_result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 391, in __get_result
    raise self._exception
  File "/home/user/coding/voraus-system-control-py/venv/lib/python3.9/site-packages/asyncua/server/server.py", line 445, in start
    raise exp
  File "/home/user/coding/voraus-system-control-py/venv/lib/python3.9/site-packages/asyncua/server/server.py", line 441, in start
    await self.bserver.start()
  File "/home/user/coding/voraus-system-control-py/venv/lib/python3.9/site-packages/asyncua/server/binary_server_asyncio.py", line 139, in start
    self._server = await asyncio.get_running_loop().create_server(self._make_protocol, self.hostname, self.port)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 1506, in create_server
    raise OSError(err.errno, 'error while attempting '
OSError: [Errno 98] error while attempting to bind on address ('127.0.0.1', 4840): address already in use

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/user/coding/voraus-system-control-py/minimal_opcua.py", line 15, in <module>
    server.stop()
  File "/home/user/coding/voraus-system-control-py/venv/lib/python3.9/site-packages/asyncua/sync.py", line 332, in stop
    self.tloop.post(self.aio_obj.stop())
  File "/home/user/coding/voraus-system-control-py/venv/lib/python3.9/site-packages/asyncua/sync.py", line 52, in post
    return futur.result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 446, in result
    return self.__get_result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 391, in __get_result
    raise self._exception
  File "/home/user/coding/voraus-system-control-py/venv/lib/python3.9/site-packages/asyncua/server/server.py", line 463, in stop
    await self.bserver.stop()
  File "/home/user/coding/voraus-system-control-py/venv/lib/python3.9/site-packages/asyncua/server/binary_server_asyncio.py", line 159, in stop
    self.cleanup_task.cancel()
AttributeError: 'NoneType' object has no attribute 'cancel'
^CException ignored in: <module 'threading' from '/usr/lib/python3.9/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 1477, in _shutdown
    lock.acquire()
KeyboardInterrupt:
```

**Expected behavior**<br />
server.stop() should ideally not raise and the threadloop should be ended in any case.

**Version**<br />
Python-Version: 3.9<br />
opcua-asyncio Version (e.g. master branch, 0.9) 1.0.1/master:

AndreasHeine, 2023-02-15:
the OSError is correct you try to bind the port 4840 two times!

lbckmnn, 2023-02-15:
Yes of course.
But my point is, it is not possible to stop the server, end the thread loop and terminate the process in this case. 
The process just "hangs".

PR #1212 (2023-02-17): Fix/server stop fails when bind fails
Fixes #1209 
`server.stop()` does currently cause an exception (`AttributeError: 'NoneType' object has no attribute 'cancel`')  in `BinaryServer.stop()`, when `server.start()` failed, because `BinaryServer.cleanup_task` is None in this case. 
See linked issue for details. 

With this PR, the `cleanup_task.cancel()` is only called, when `cleanup_task` is actually valid (not None). 
It might be reasonable to move the creation of the cleanup task further up in `BinaryServer.start()` but I don`t have enough insight for that. 

This PR also adds a testcase that [... cut]
closed 2023-02-17 by oroulet