FreeOpcUa/opcua-asyncio #1225: Regarding Writing values on  a variable tag using opcua-asyncio python

rajnishnationfirst, 2023-02-22:
Hi Team,

I am trying to write values on to a variable node using opcua-asyncio python as below onto a variable .

The variable is of Type Int64 ( integer) 

```
import time
import logging

import asyncio
from asyncua import Client, Node, ua

async def main():
    async with Client(url='opc.tcp://192.168.0.1:4840') as client:
        while True:
            # Do something with client

           **#For Integer type i m using the below code to write 900 , but not working** 

            node2 = client.get_node('ns=2;s=RACK3_840D.production_raw.DOUBLE_TAG')
            await node2.write_value(900, varianttype=ua.VariantType.Int64)
           
           **#For Boolean type i m using the below code to write false , but not working** 

           node2 = client.get_node('ns=2;s=RACK3_840D.production_raw.BOOLEAN_TAG')
           await node2.write_value(0, varianttype=ua.VariantType.Int64)

           

            time.sleep(10)

if __name__ == "__main__":
    logging.basicConfig(level=logging.INFO)
    asyncio.run(main())

```

When i run the program i get the error / exceptions as below :

INFO:asyncua.client.client:connect
INFO:asyncua.client.ua_client.UaClient:opening connection
WARNING:asyncua.uaprotocol:updating client limits to: TransportLimits(max_recv_buffer=65535, max_send_buffer=65535, max_chunk_count=65, max_message_size=4194304)
INFO:asyncua.client.ua_client.UASocketProtocol:open_secure_channel
INFO:asyncua.client.ua_client.UaClient:create_session
INFO:asyncua.client.client:find_endpoint [EndpointDescription(EndpointUrl='opc.tcp://debian:4840/BFCGateway', Server=ApplicationDescription(ApplicationUri='urn:debian:Siemens:OPCUAServer:BFCGateway', ProductUri=None, ApplicationName=LocalizedText(Locale='en-US', Text='BFCGateway'), ApplicationType_=<ApplicationType.Server: 0>, GatewayServerUri=None, DiscoveryProfileUri=None, DiscoveryUrls=['opc.tcp://debian:4840/BFCGateway']), ServerCertificate=b'0\x82\x03\xbe0\x82\x02\xa6\xa0\x03\x02\x01\x02\x02\n\x0bJ\x1b\xb9\xe0\xb1\xd9\xe5\xe8\xac0\r\x06\t*\x86H\x86\xf7\r\x01\x01\x0b\x05\x000 1\x1e0\x1c\x06\x03U\x04\x03\x13\x15BFCGatewayOpcUaServer0\x1e\x17\r230221000000Z\x17\r240221000000Z0 1\x1e0\x1c\x06\x03U\x04\x03\x13\x15BFCGatewayOpcUaServer0\x82\x01"0\r\x06\t*\x86H\x86\xf7\r\x01\x01\x01\x05\x00\x03\x82\x01\x0f\x000\x82\x01\n\x02\x82\x01\x01\x00\xd8\xff\xad\xd1\xa5\xdb\xe0\x11\x1c\xd2\xcc\x9f9\xcaw\xb2\xc2\xe8\xb5\x0b\x9dr\xbak\xf6k\x02\x8d\xbd/x\xa6\x1bu\xbc\xbe\xaf\xf0\x19L\x0e\xd5\xe0\r\xe9\x06]\xa1 \xdc<\xd7\x1b\x98\xd1\xd6\xa6\xf7\xd4\xb5\x07\xd9\x92\x96\n\xb0\xb9\xa7\x80\'5\xf5~\xf6%\xf5\x18\xbe\x9f_\xb8\x0b\x16\xce\xba \xd5H\x8aF\xec\x07\x1ak\xc7wIw\x1c\xbe}m\xf8rq\xfe\x19\xa6`\x97\x1c\xbe3*\x13r\x1fe\x1e\xf6o\x82\x1bRq\x11+\xcf\xe8<E\xc9\xc2\xd8\xf2E\x1c6dC\xd9zj_\xc3\t\xc3\x08\xbd\xa0\x06\xa7\xdag\xf9\x02\x92U\r;\ng\x82\xb5\xf7\xaf)\xd3\xee8x\\\x9do\x84\xc5\xc3v\x94W\x0b\x17\xc1.\x07\xf8x\x95\xa0\xa1\xdf\xb8=\xf9\xc9\tV4N\xcbh\x813E\xf3o4\x828\xdaC\x17\\\x87\xd3H\x82\xe6g\x86\xea\xdd\xec-\xae\xd1Z\xe2\xff\xa00n\x1c\x91\xc8:\xaa\xfdc\xb5\r-R\xd5\x1f\xc6\xc5%\xa6\xe4bV\x9e\x7fu\x9b\x02\x03\x01\x00\x01\xa3\x81\xf90\x81\xf60\x12\x06\x03U\x1d\x13\x01\x01\xff\x04\x080\x06\x01\x01\xff\x02\x01\x000\x1d\x06\x03U\x1d\x0e\x04\x16\x04\x14-?\x9e\xef\x0f\x8d]\x16\xa6\x1fm+\xdf\xe5\x85\xd7\xa3\xba\xa6O0Q\x06\x03U\x1d#\x04J0H\x80\x14-?\x9e\xef\x0f\x8d]\x16\xa6\x1fm+\xdf\xe5\x85\xd7\xa3\xba\xa6O\xa1$\xa4"0 1\x1e0\x1c\x06\x03U\x04\x03\x13\x15BFCGatewayOpcUaServer\x82\n\x0bJ\x1b\xb9\xe0\xb1\xd9\xe5\xe8\xac0\x0e\x06\x03U\x1d\x0f\x01\x01\xff\x04\x04\x03\x02\x02\xf40 \x06\x03U\x1d%\x01\x01\xff\x04\x160\x14\x06\x08+\x06\x01\x05\x05\x07\x03\x01\x06\x08+\x06\x01\x05\x05\x07\x03\x020<\x06\x03U\x1d\x11\x04503\x86)urn:debian:Siemens:OPCUAServer:BFCGateway\x82\x06debian0\r\x06\t*\x86H\x86\xf7\r\x01\x01\x0b\x05\x00\x03\x82\x01\x01\x00r\xcd\xe3\xc7\xe8R\xdb`\xe8\xae\xdc\xa3\\M\x93\xa6|UG\x11\xda\x86\\\xc4%\x9f\x9f\x9a\'\xc0\xb5\x1f\xdd\x7f\x8f\x08\x93\xd2\xcd\x0bl\x06@\xa4\x1e\x92\xd985G\x0b\xcd\xb3\x03\xb0\xc0n\x1d+\xbe\x1cJ_dq{.\xb7\x0c(\xf26En\x80\x10l\x19z\x88\x814\xa5\xc4\x13\xe3\x8dlh\xa5W\xcc\xf7\xcc\x99\x18\xa6\xe2\xad\xbc+Ra\x13\x8a\xa8;\xbf\x0b\x0e\xabJ=\xa8\x08W5\x92\x84\xd50g\xaf\xf9\x8d\xb9\xc7\x10\x86\xa0/\x00\xf0\x10\x06\x8c:q\xc2\x9c$\xa3\xfa\xf8\xc8r\x14\xf6\x82+\xf1\x1d3\x16Y,M\xdf\xaf\x0c\xfb\x1d\xb8\xc7\x0e\xfc\xa1H1\xa1P\xb3\x9e\xa9+\'>\xfa\xe2\xd9\xa4\xd3\xa2p\xd7\xfdE\x1d\x86\xfbw\xbf\x1cz2\xbbK\x1b2=2\xdbI\'\xc3\xa8\xf5\xd0\xd0\xea\xcf\xc2Zo\x80R2\xbd\xb1\xd8\x89\x8fv\x18\xb4\xd8\xa83\xc1\xc0Z\xef\xc4\x12[\x1fw\xc21\xc4\xadc\x02\xef\x06\x00\x81l)\x88b\x04/\x19D\xec', SecurityMode=<MessageSecurityMode.Sign: 2>, SecurityPolicyUri='http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256', UserIdentityTokens=[UserTokenPolicy(PolicyId='1', TokenType=<UserTokenType.UserName: 1>, IssuedTokenType=None, IssuerEndpointUrl=None, SecurityPolicyUri='http://opcfoundation.org/UA/SecurityPolicy#Basic256'), UserTokenPolicy(PolicyId='2', TokenType=<UserTokenType.UserName: 1>, IssuedTokenType=None, IssuerEndpointUrl=None, SecurityPolicyUri='http://opcfoundation.org/UA/SecurityPolicy#None'), UserTokenPolicy(PolicyId='3', TokenType=<UserTokenType.UserName: 1>, IssuedTokenType=None, IssuerEndpointUrl=None, SecurityPolicyUri='http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256'), UserTokenPolicy(PolicyId='4', TokenType=<UserTokenType.UserName: 1>, IssuedTokenType=None, IssuerEndpointUrl=None, SecurityPolicyUri='http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15'), UserTokenPolicy(PolicyId='5', TokenType=<UserTokenType.Certificate: 2>, IssuedTokenType=None, IssuerEndpointUrl=None, SecurityPolicyUri='http://opcfoundation.org/UA/SecurityPolicy#Basic256'), UserTokenPolicy(PolicyId='6', TokenType=<UserTokenType.Certificate: 2>, IssuedTokenType=None, IssuerEndpointUrl=None, SecurityPolicyUri='http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256'), UserTokenPolicy(PolicyId='7', TokenType=<UserTokenType.Certificate: 2>, IssuedTokenType=None, IssuerEndpointUrl=None, SecurityPolicyUri='http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15'), UserTokenPolicy(PolicyId='8', TokenType=<UserTokenType.Anonymous: 0>, IssuedTokenType=None, IssuerEndpointUrl=None, SecurityPolicyUri='http://opcfoundation.org/UA/SecurityPolicy#None')], TransportProfileUri='http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary', SecurityLevel=6), EndpointDescription(EndpointUrl='opc.tcp://debian:4840/BFCGateway', Server=ApplicationDescription(ApplicationUri='urn:debian:Siemens:OPCUAServer:BFCGateway', ProductUri=None, ApplicationName=LocalizedText(Locale='en-US', Text='BFCGateway'), ApplicationType_=<ApplicationType.Server: 0>, GatewayServerUri=None, DiscoveryProfileUri=None, DiscoveryUrls=['opc.tcp://debian:4840/BFCGateway']), ServerCertificate=b'0\x82\x03\xbe0\x82\x02\xa6\xa0\x03\x02\x01\x02\x02\n\x0bJ\x1b\xb9\xe0\xb1\xd9\xe5\xe8\xac0\r\x06\t*\x86H\x86\xf7\r\x01\x01\x0b\x05\x000 1\x1e0\x1c\x06\x03U\x04\x03\x13\x15BFCGatewayOpcUaServer0\x1e\x17\r230221000000Z\x17\r240221000000Z0 1\x1e0\x1c\x06\x03U\x04\x03\x13\x15BFCGatewayOpcUaServer0\x82\x01"0\r\x06\t*\x86H\x86\xf7\r\x01\x01\x01\x05\x00\x03\x82\x01\x0f\x000\x82\x01\n\x02\x82\x01\x01\x00\xd8\xff\xad\xd1\xa5\xdb\xe0\x11\x1c\xd2\xcc\x9f9\xcaw\xb2\xc2\xe8\xb5\x0b\x9dr\xbak\xf6k\x02\x8d\xbd/x\xa6\x1bu\xbc\xbe\xaf\xf0\x19L\x0e\xd5\xe0\r\xe9\x06]\xa1 \xdc<\xd7\x1b\x98\xd1\xd6\xa6\xf7\xd4\xb5\x07\xd9\x92\x96\n\xb0\xb9\xa7\x80\'5\xf5~\xf6%\xf5\x18\xbe\x9f_\xb8\x0b\x16\xce\xba \xd5H\x8aF\xec\x07\x1ak\xc7wIw\x1c\xbe}m\xf8rq\xfe\x19\xa6`\x97\x1c\xbe3*\x13r\x1fe\x1e\xf6o\x82\x1bRq\x11+\xcf\xe8<E\xc9\xc2\xd8\xf2E\x1c6dC\xd9zj_\xc3\t\xc3\x08\xbd\xa0\x06\xa7\xdag\xf9\x02\x92U\r;\ng\x82\xb5\xf7\xaf)\xd3\xee8x\\\x9do\x84\xc5\xc3v\x94W\x0b\x17\xc1.\x07\xf8x\x95\xa0\xa1\xdf\xb8=\xf9\xc9\tV4N\xcbh\x813E\xf3o4\x828\xdaC\x17\\\x87\xd3H\x82\xe6g\x86\xea\xdd\xec-\xae\xd1Z\xe2\xff\xa00n\x1c\x91\xc8:\xaa\xfdc\xb5\r-R\xd5\x1f\xc6\xc5%\xa6\xe4bV\x9e\x7fu\x9b\x02\x03\x01\x00\x01\xa3\x81\xf90\x81\xf60\x12\x06\x03U\x1d\x13\x01\x01\xff\x04\x080\x06\x01\x01\xff\x02\x01\x000\x1d\x06\x03U\x1d\x0e\x04\x16\x04\x14-?\x9e\xef\x0f\x8d]\x16\xa6\x1fm+\xdf\xe5\x85\xd7\xa3\xba\xa6O0Q\x06\x03U\x1d#\x04J0H\x80\x14-?\x9e\xef\x0f\x8d]\x16\xa6\x1fm+\xdf\xe5\x85\xd7\xa3\xba\xa6O\xa1$\xa4"0 1\x1e0\x1c\x06\x03U\x04\x03\x13\x15BFCGatewayOpcUaServer\x82\n\x0bJ\x1b\xb9\xe0\xb1\xd9\xe5\xe8\xac0\x0e\x06\x03U\x1d\x0f\x01\x01\xff\x04\x04\x03\x02\x02\xf40 \x06\x03U\x1d%\x01\x01\xff\x04\x160\x14\x06\x08+\x06\x01\x05\x05\x07\x03 [... cut ...]
  File "C:\Users\SIEMENS_Hotline\AppData\Local\Programs\Python\Python310\lib\site-packages\asyncua\common\node.py", line 285, in write_attribute
    result[0].check()
  File "C:\Users\SIEMENS_Hotline\AppData\Local\Programs\Python\Python310\lib\site-packages\asyncua\ua\uatypes.py", line 328, in check
    raise UaStatusCodeError(self.value)
asyncua.ua.uaerrors._auto.Bad: "The operation failed."(Bad)

**SNAP for reference below :**

[image]

Request you to pls help me as i m exploring a POC to be shown to my management.

It will be great if any of you can help me for the same.

Thanks

Rajnish Vishwakarma

AndreasHeine, 2023-02-22:
>             node2 = client.get_node('ns=2;s=RACK3_840D.production_raw.DOUBLE_TAG')
>             await node2.write_value(900, varianttype=ua.VariantType.Int64)

"DOUBLE_TAG" with type "Int64" ???

rajnishnationfirst, 2023-02-22:
Thanks Andreas for your kind reply.

DOUBLE_TAG ( just a tag name ) which  is of type INT64

Pls check the below snap of UA expert

[image]

Thanks

Rajnish Vishwakarma

AndreasHeine, 2023-02-22:
does the write from UaExpert work?

rajnishnationfirst, 2023-02-24:
Hi Andreas thanks for the reply.. We just tested by writing via UA Expert client and i think we got same error what i am getting in Python OPCUA Custom client code written by me and post above .

**UA Expert Issue while writing value onto a tag - SNAP** 

[image]

Python Code error / exception **Snap**

[image]

It think the python console and the UA Expert console produces same error - Bad-Operation failed 

Thanks

Rajnish Vishwakarma

closed 2023-03-14 by AndreasHeine