FreeOpcUa/opcua-asyncio #1234: await server.load_data_type_definitions()

vhavard, 2023-03-02:
### Please read, before you post!

This is a _**BUG REPORT for issues in the existing code**_. 

If you have general questions, code handling problems or ideas, please use the:

Discussionsboard: https://github.com/FreeOpcUa/opcua-asyncio/discussions <br/> or <br />
Gitter-Channel: https://gitter.im/FreeOpcUa/opcua-asyncio

---

To save some time, please provide us following informations, if possible:

**Describe the bug** <br />
A clear and concise description of what the bug is.

**To Reproduce**<br />
Steps to reproduce the behavior incl code:

**Expected behavior**<br />
A clear and concise description of what you expected to happen.

**Screenshots**<br />
If applicable, add screenshots to help explain your problem. <br />
You can post Screenshots directly to github, please don't use 3rd Party webpages

**Version**<br />
Python-Version:<br />
opcua-asyncio Version (e.g. master branch, 0.9):

schroeder-, 2023-03-02:
No bug, import_xml already generated the datatypes. So the second call doesn't yield any datatypes.

vhavard, 2023-03-02:
Thank you, I found the solution to access my personnal datatype
````python
ua.__dict__["QuaternionDataType"](1, 5, 8, 1)
QuaternionDataType(x=1, y=5, z=8, w=1)
````
I leave the answer to help people

closed 2023-03-02 by vhavard

vhavard, 2023-03-03:
> No bug, import_xml already generated the datatypes. So the second call doesn't yield any datatypes.

By the way, is there another mean to get the classes generated via the import_xml() method ?

For the moment, I am using this:
```python
ua.__dict__["QuaternionDataType"](1, 5, 8, 1)
# return QuaternionDataType(x=1, y=5, z=8, w=1)
```

schroeder-, 2023-03-03:
can also do this:
ua.QuaternionDataType(x=1, y=5, z=8, w=1)
ua.QuaternionDataType(1, 5, 8, 1)