FreeOpcUa/opcua-asyncio #1245: Client does not support certificate chain received from server

piotrek204, 2023-03-15:
**Describe the bug** <br />
In case when the server has a certificate signed by CA root while the client is opening a secure connection to a server, the server sends a certificate chain to the client. The `opcua-asyncio` client is unable to parse such cefrtyficate chain and fails. 

This issue was already reported here: [https://github.com/FreeOpcUa/opcua-asyncio/issues/1148](https://github.com/FreeOpcUa/opcua-asyncio/issues/1148)
and as a continuation here (I added my observations there):  [https://github.com/pyca/cryptography/issues/7901](https://github.com/pyca/cryptography/issues/7901)

**To Reproduce**<br />
  - as a server is used `C++ SDK Demo Server` with the certificate "signed by" `UAGDS security management` (server contains on certificate signed by CA root and CA root certificate)
  - the following code is used:
  ```
import asyncio
import logging
import sys
sys.path.insert(0, "..")
from asyncua import Client, Node, ua
from asyncua.crypto.security_policies import SecurityPolicyBasic256Sha256

logging.basicConfig(level=logging.INFO)
_logger = logging.getLogger("asyncua")

async def task(loop):
    url = "opc.tcp://@2V43YK2:48010"  # UA Demo Server Cpp
    client = Client(url=url)
    client.set_user('root')
    client.set_password('secret')
    await client.set_security(
        SecurityPolicyBasic256Sha256,
        certificate=r"C:\other\opcua-asyncio\examples\certs_pisz\cert.der",
        private_key=r"C:\other\opcua-asyncio\examples\certs_pisz\pkey.pem",
    )
    async with client:
        objects = client.nodes.objects
        child = await objects.get_children()
        print(child)

def main():
    loop = asyncio.get_event_loop()
    loop.set_debug(True)
    loop.run_until_complete(task(loop))
    loop.close()

if __name__ == "__main__":
    main()
```
it results:
``` 
C:\Users\PSznapk\AppData\Local\Programs\Python\Python310-32\python.exe C:\other\opcua-asyncio\examples\client-with-encryption.py 
C:\Users\PSznapk\AppData\Local\Programs\Python\Python310-32\lib\site-packages\cryptography\hazmat\backends\openssl\backend.py:61: UserWarning: You are using cryptography on a 32-bit Python on a 64-bit Windows Operating System. Cryptography will be significantly faster if you switch to using a 64-bit Python.
  from cryptography.hazmat.bindings.openssl import binding
C:\other\opcua-asyncio\examples\client-with-encryption.py:29: DeprecationWarning: There is no current event loop
  loop = asyncio.get_event_loop()
INFO:asyncua.client.ua_client.UaClient:opening connection
INFO:asyncua.uaprotocol:updating client limits to: TransportLimits(max_recv_buffer=65536, max_send_buffer=65536, max_chunk_count=256, max_message_size=16777216)
INFO:asyncua.client.ua_client.UASocketProtocol:open_secure_channel
INFO:asyncua.client.ua_client.UASocketProtocol:close_secure_channel
INFO:asyncua.client.ua_client.UASocketProtocol:Request to close socket received
INFO:asyncua.client.client:find_endpoint [EndpointDescription(EndpointUrl='opc.tcp://2V43YK2:48010', Server=ApplicationDescription(ApplicationUri='urn:2V43YK2:UnifiedAutomation:UaServerCpp', ProductUri='urn:UnifiedAutomation:UaServerCpp', ApplicationName=LocalizedText(Locale='en', Text='UaServerCpp@2V43YK2'), ApplicationType_=<ApplicationType.Server: 0>, GatewayServerUri=None, DiscoveryProfileUri=None, DiscoveryUrls=['opc.tcp://2V43YK2:48010']), ServerCertificate=b'0\x82\x04{0\x82\x03c\xa0\x03\x02\x01\x02\x02\x0f\x145\x87\x83pBq6C\x10\x00\x00\x00\x00\t0\r\x06\t*\x86H\x86\xf7\r\x01\x01\x0b\x05\x00011\x0b0\t\x06\x03U\x04\n\x0c\x02RA1\x0f0\r\x06\x03U\x04\x0b\x0c\x06RAPISZ1\x110\x0f\x06\x03U\x04\x03\x0c\x08testpush0\x1e\x17\r230314075636Z\x17\r240320075636Z0\x81\x811\x170\x15\x06\n\t\x92&\x89\x93\xf2,d\x01\x19\x16\x072V43YK21\x0b0\t\x06\x03U\x04\x06\x13\x02DE1\x150\x13\x06\x03U\x04\x07\x0c\x0cLocationName1\x150\x13\x06\x03U\x04\n\x0c\x0cOrganization1\r0\x0b\x06\x03U\x04\x0b\x0c\x04Unit1\x1c0\x1a\x06\x03U\x04\x03\x0c\x13UaServerCpp@2V43YK20\x82\x01"0\r\x06\t*\x86H\x86\xf7\r\x01\x01\x01\x05\x00\x03\x82\x01\x0f\x000\x82\x01\n\x02\x82\x01\x01\x00\xd3\x84\xeap\xbf.\x1c\xc3\xb7\xe2\xf8q\xae\xa1Z\x1b\\9\xfa\xf2\xb9D\x91C\xd6\xb3\xc3\xf7\x95d\x1a\xbanz\xd1}T\x07\xf7U\xa2T@\x94~^\x06\r\x87\xee\xc5\xeb\x0c[KE\xa2\xc3\xde\xe8\x17\r-xT\x98N\x98\xf2\x1c\xaf\xce~n\xdc7#\xb1gTU\xc0m\x80\xb1\xed\xf0\x92?F\x9fS4v\x88\xcf3G\x05xwI6\xeb\x03\x0c\x9b\xda\xb4\xd5\xf5i\xfepN\xf6!,\xd0*\xf3\x13UNd\xdc\x86+\x02\x8a;8\x87\x1f\x04\xc4\x90\x08\xbf\x855\x1e\x01s\xb8h\xa7\x84\xda\xc5\x1c\x145\xe5]\r\xe0+\xe2\xdf$\x0b,\xcfK\xa5ew\x10\r\x82\x1d\xaa\xec\x90\xa5\x7f\x07C\xc4+\x83\xd0\x9b\x0f\xfe\xb1:\x07\x08*\xbb\xa0{R\x17\x01\xee\x8c\xe3W\xf4^\xe2X\x1b(\xa7\xa0j\r\x13\xb1\x81\xcc\x15;\x9fLe\xf2:\xacy!5\x97\x08ZL\xd1\x84"\x89!H\x9buv>\x15\x10\xe3>\x0f\x85\xae\x7f\xb27>j\xb5\t\x96\x85\x02\x03\x01\x00\x01\xa3\x82\x01=0\x82\x0190\x0c\x06\x03U\x1d\x13\x01\x01\xff\x04\x020\x0000\x06\t`\x86H\x01\x86\xf8B\x01\r\x04#\x16!"Generated with OpenSSL backend."0\x1d\x06\x03U\x1d\x0e\x04\x16\x04\x14\xaf3\xc4\xe8\x82.\x17\xc0vz\xb0\x06\x98\x89c(\x11\x8d\x9a\x1c0g\x06\x03U\x1d#\x04`0^\x80\x14\xfc\xa8\x1c\xc9r`\xc1{)\xf7Zu\xf8\xa9\xaf\xcdA\x10\x831\xa15\xa43011\x0b0\t\x06\x03U\x04\n\x0c\x02RA1\x0f0\r\x06\x03U\x04\x0b\x0c\x06RAPISZ1\x110\x0f\x06\x03U\x04\x03\x0c\x08testpush\x82\x0f\x14wt0"5\t\t\x02C\x00\x00\x00\x00\x010\x0e\x06\x03U\x1d\x0f\x01\x01\xff\x04\x04\x03\x02\x04\xf00 \x06\x03U\x1d%\x01\x01\xff\x04\x160\x14\x06\x08+\x06\x01\x05\x05\x07\x03\x01\x06\x08+\x06\x01\x05\x05\x07\x03\x020=\x06\x03U\x1d\x11\x04604\x86)urn:2V43YK2:UnifiedAutomation:UaServerCpp\x82\x072V43YK20\r\x06\t*\x86H\x86\xf7\r\x01\x01\x0b\x05\x00\x03\x82\x01\x01\x00\x8e]9\x16\xbb\xa0\xa99\xa4\x9e\xd9_\xc9\xbb\xad\r\xb5\xff\xf94\xdd\x8d\x97\xa3y\'I\x17\xf5\x8c\x12\x02\xe5\xfc]\xe0\xa2\x8d>)\x9a\x08+\xf1\xe1\x0ew\xa6[\x10\x91\x9a\xae!\xf6\x0f\xa3\r\x88\x96\xc9\xe6\xee\xd5:\x98S\xfb\xc8\xf4h\x9d\x81\x83\x15q\xb7aG)J\x8d\x7f\xb6\xffZY\x13\xc6mPM\xc5\xbc\xd7\xc9%\x1aqK\xf1\xc9\xe3K\xd2\xfd\x93\xec\x05\xfd\xa4\x1d\x1d\x9e\xdeR\x96\x85@\xa9\xe5H\x98\xde\xbax\x01\x81\x01\xb9\xaf2\x04J`\xad\xdcD\xc7)\xd4V\x18\xf2kU\xd8\xe6\xb4\x06*b\x97\xf5\x85\xdb&T\x8b\xbec\x97XNP\xe6ng\x93\x8c\ne\x0ej\xe6\x96%\xa37\xab\x7fz\xfc\xe6\xe1\xcc\x1c\xcb\xa4\x9eck 7\xd9N\x08\x88\xfc\x8d\x9e\xdc(z09\x885u\xb5r\x94FN\x1e\xd6/\x19?\x18!\x0b\xe0\x89\x04/\x080\xa2\xefj\x08\rp\rl\xb9\x1b\x7f\xbb\xa3\x13\x8d\x1f\n\xa7r\xd5\xe0\x8e\xc5:\x96\x05q\x870\x82\x04"0\x82\x03\n\xa0\x03\x02\x01\x02\x02\x0f\x14wt0"5\t\t\x02C\x00\x00\x00\x00\x010\r\x06\t*\x86H\x86\xf7\r\x01\x01\x0b\x05\x00011\x0b0\t\x06\x03U\x04\n\x0c\x02RA1\x0f0\r\x06\x03U\x04\x0b\x0c\x06RAPISZ1\x110\x0f\x06\x03U\x04\x03\x0c\x08testpush0\x1e\x17\r230301105339Z\x17\r250228105339Z011\x0b0\t\x06\x03U\x04\n\x0c\x02RA1\x0f0\r\x06\x03U\x04\x0b\x0c\x06RAPISZ1\x110\x0f\x06\x03U\x04\x03\x0c\x08testpush0\x82\x01"0\r\x06\t*\x86H\x86\xf7\r\x01\x01\x01\x05\x00\x03\x82\x01\x0f\x000\x82\x01\n\x02\x82\x01\x01\x00\xec\xfbu\x15\x17$\xcb\xd8\xed%E\x8d\xad8\x8a\xc8\x87\xb0F(\x93\x12\xa4WXPsZ\xd0E\xfd\x8c\x0c\xd5\x979\xe1\xc1[\xfe\x85*k\xf1\xfe\xc6\x91\xc5\x07\xf6\xeep6p.\xb64z\xd5\x1c\xa6\xc6\x8bx\xee\x03\x7f\x94\xe6#c2\x16l\x06\x17F\xc1\xca:\x10\xecdL\x85.\xc8\x8a\x00v\xe6\x16\x19\xe9:\xa0\x0c\xfd\x86^\x83*B\xca)\xd1\x8b\xa7\xb9\xe2\x18$\xe62\x96\xc4#\x87\xdc\xc9\\k\x0f)0\xd7\xc6\x9aT5\x11\x89\xdet\x10^\xd3\x9a\xefB\x8ff,\xfeZI8~\xd7\xa5;\x1d\xe3\xab7\xbd\xf2\x9c{0\xb7:\xd9qf\xe7\xc3p\xb9\x97o9\xe0\xe2\xa5\x9b\xdb 3\x08\x90\xfc\xfeI\x97\x7f\x9c\xad\n\xc4\x9d\x9e\xfe\x04\xad\rK\xfbD\xcc\x81\xaf\x9f\xd2s\xc8R|ZN\xc78X\x04o\xea\xd3-\xa4\xa5\xf7*\xb1\xe6+\x89\x9d\xcb\xb9W\xa0{\xef\xe5\x08\xf4\x16)8\xfa\x14\xf3f4@t\xa99\x9dVdI3G\xfe\xf3\x02\x03\x01\x00\x01\xa3\x82\x0150\x82\x0110\x0f\x06\x03U\x1d\x13\x01\x01\xff\x04\x050\x03\x01\x01\xff00\x06\t`\x86H\x01\x86\xf8B\x01\r\x04#\x16!"Generated with OpenSSL backend."0\x1d\x06\x03U\x1d\x0e\x04\x16\x04\x14\xfc\xa8\x1c\xc9r`\xc1{)\xf7Zu\xf8\xa9\xaf\xcdA\x10\x8310g\x06\x03U\x1d#\x04`0^\x80\x14\xfc\xa8\x1c\xc9r`\xc1{)\xf7Zu\xf8\xa9\xaf\xcdA\x10\x831\xa15\xa43011\x0b0\t\x06\x03U\x04\n\x0c\x02RA1\x0f0\r\x06\x03U\x04\x0b\x0c\x06RAPISZ1\x110\x0f\x06\x03U\x04\x03\x0c\x08testpush\x82\x0f\x14wt0"5\t\t\x02C\x00\x00\x00\x00\x010\x0e\x06\x03U\x1d\x0f\x01\x01\xff\x04\x04\x03\x02\x01\xe60T\x06\x03U\x1d\x11\x04M0K\x865urn:UnifiedAutomation:UaGds:Server:2V43YK2.ra-int.com\x82\x122V43YK2.ra-int.com0\r\x06\t*\x86H\x86\xf7\r\x01\x01\x0b\x05\x00\x03\x82\x01\x01\x00L\xe5%WBr\x1914\xd0%\xb2\x8a\x80\xbd\xccLR\x9f|\xd3\xabS!~\x83\xe0\xc23\x1e\xdc\xc8Yb\xce\xc9@\x83\x9a\x [... cut ...]
as proof of that there are two certificates we can try to unpack them:
```
>>> uacrypto.x509_from_der(endpoint.ServerCertificate[:1151])
<Certificate(subject=<Name(DC=2V43YK2,C=DE,L=LocationName,O=Organization,OU=Unit,CN=UaServerCpp@2V43YK2)>, ...)>
>>> uacrypto.x509_from_der(endpoint.ServerCertificate[1151:])
<Certificate(subject=<Name(O=RA,OU=RAPISZ,CN=testpush)>, ...)>
```

**Expected behavior**<br />
Connection to server is established with out any errors.

**Version**<br />
Python-Version: 3.10.5<br />
opcua-asyncio Version: master branch

piotrek204, 2023-03-23:
@schroeder- I checked the mentioned change on the master branch [https://github.com/FreeOpcUa/opcua-asyncio/pull/1250](https://github.com/FreeOpcUa/opcua-asyncio/pull/1250) but it still does not work. I get the following exception:
```
Traceback (most recent call last):
  File "C:\other\opcua-asyncio\examples\client-with-encryption.py", line 38, in task
    async with client:
  File "C:\other\opcua-asyncio\asyncua\client\client.py", line 86, in __aenter__
    await self.connect()
  File "C:\other\opcua-asyncio\asyncua\client\client.py", line 286, in connect
    await self.create_session()
  File "C:\other\opcua-asyncio\asyncua\client\client.py", line 474, in create_session
    raise ua.UaError("Server certificate mismatch")
asyncua.ua.uaerrors._base.UaError: Server certificate mismatch

```
When a session is created the entire chain of certs is still compared

schroeder-, 2023-03-23:
Ah ok i only got one place. Will create expanded patch.

piotrek204, 2023-03-31:
@schroeder- I tested your change and for my use cases it works fine

closed 2023-03-31 by piotrek204