FreeOpcUa/opcua-asyncio #1252: Client is dying after receiving little data

venkatp-co, 2023-03-20:
Hi My server run 24/7, but my client is dying after receiving little data from server, my client should also run 24/7. Please find my client below.

```
import asyncio

from asyncua import ua

from asyncua import Client

import requests

import json

class SubHandler(object):
    """Subscription Handler. To receive events from server for a subscription"""
    
    def datachange_notification(self, node, val, data):
        csv_url = "http://localhost:3001/pushprocesseddata"
        """This functions gets he notifications when data is changed"""
        sdt = data.monitored_item.Value.SourceTimestamp.strftime("%Y-%m-%d %H:%M:%S:%f")[:-3]
        row = {"RECORDED_DTTM": sdt, "POINT_ID": str(node).split("s=")[2], "RECORDED_VALUE": val}  
        print(row)
       

async def main():
    """This is the main function for the client"""
    url = "opc.tcp://admin@localhost:4849"
    async with Client(url=url) as client:

        print("Root node is: %r", client.nodes.root)
        print("Objects node is: %r", client.nodes.objects)

        print("Children of root are: %r", await client.nodes.root.get_children())
        uri = "Softsensor_OPC_Server_2"
        idx = await client.get_namespace_index(uri)
        print("idx",idx)
        myvar = await client.get_node("ns=2;i=1").get_children()
        # myvar_predicted = await client.get_node("ns=3;i=1").get_children()

        handler = SubHandler()
        sub = await client.create_subscription(10, handler) #When I give 10, I'm receving ~1400 records. When using 100, I'm receving 14000 records. When I give 200, no result and client is just terminating with out any errors.
        handle = await sub.subscribe_data_change(myvar)
        print("handle",handle)
        await asyncio.sleep(0.1)

       
        await sub.subscribe_events()
```

AndreasHeine, 2023-03-20:
can you explain what you mean with "dying"?

venkatp-co, 2023-03-20:
sub = await client.create_subscription(10, handler) 
When I give 10, I'm receiving ~1400 records. When using 100, I'm receiving 14000 records. When I give 200, no result and client is just terminating with out any errors

Below is the screenshot when 10 is given as argument. The program is just terminating after receiving few records.

[image]

AndreasHeine, 2023-03-20:
10 means in your case publishing interval 10ms so each 10 ms the client asks for changes! (with python you should not do less the 50ms... instead use the monitored item sampling interval!)

You should read the following parts of spec. to understand how the subscription work:
https://reference.opcfoundation.org/v104/Core/docs/Part4/5.13.1/
https://reference.opcfoundation.org/v104/Core/docs/Part4/5.12.1/

In simple words:
A OPC UA Subscription works like a "Mailbox" which gets emptied in a defined interval by the Client.
The OPC UA Client sends a publishrequest, the server takes all notifications since the last publishrequest.
and send them in the publishresponse to the client.
If there is no Notification in the "Mailbox", after a while the Client will get a "keep-alive" back and then the client will send another publishrequest.
The OPC UA Client can add "MonitoredItems" to the Subscription which will generate Notifications.

AndreasHeine, 2023-03-20:
for better debugging you should set the logger to "DEBUG" and post the output! otherwise its hard to guess whats failing!?

oroulet, 2023-03-20:
You are quiting the loop yourself. You need to do something to keep it open. Look at the examples!

venkatp-co, 2023-03-21:
I updated the code but still the same issue, please find the below updated code. @AndreasHeine please help.

```
import asyncio

from asyncua import ua

from asyncua import Client

import requests

import json

class SubHandler(object):
    """Subscription Handler. To receive events from server for a subscription"""
    
    def datachange_notification(self, node, val, data):
        csv_url = "http://localhost:3001/pushprocesseddata"
        """This functions gets he notifications when data is changed"""
        sdt = data.monitored_item.Value.SourceTimestamp.strftime("%Y-%m-%d %H:%M:%S:%f")[:-3]
        row = {"RECORDED_DTTM": sdt, "POINT_ID": str(node).split("s=")[2], "RECORDED_VALUE": val}  
        print(row)
       

async def main():
    """This is the main function for the client"""
    url = "opc.tcp://admin@localhost:4849"
    async with Client(url=url) as client:

        print("Root node is: %r", client.nodes.root)
        print("Objects node is: %r", client.nodes.objects)

        print("Children of root are: %r", await client.nodes.root.get_children())
        uri = "Softsensor_OPC_Server_2"
        idx = await client.get_namespace_index(uri)
        print("idx",idx)
        myvar = await client.get_node("ns=2;i=1").get_children()
        # myvar_predicted = await client.get_node("ns=3;i=1").get_children()

        handler = SubHandler()
        sub = await client.create_subscription(10, handler) **#When I give 10, I'm receving ~1400 records. When using 100, I'm receving 14000 records. When I give 200, no result and client is just terminating with out any errors.**
        handle = await sub.subscribe_data_change(myvar)
        print("handle",handle)
        await asyncio.sleep(0.1)

       
        await sub.subscribe_events()
        while True:
               await asyncio.sleep(0.01)
if __name__ == "__main__":
    # logging.basicConfig(level=logging.INFO)
    asyncio.run(main())
```

AndreasHeine, 2023-03-21:
@venkatp-co 
looks like you do to much stuff in the datachangecallback... you should never do requests in the SubHandler-Callbacks which will block the eventloop for to long...

you need to make the datachange-callback async and use a asyncio http library within the callback or even better use a asyncio.queue in the callback and process the queue in a different asyncio.task with an asyncio http library!

venkatp-co, 2023-03-21:
I commented the post call, but still the same issue. Below is my subhandler class.

```
class SubHandler(object):
    """Subscription Handler. To receive events from server for a subscription"""
    
    def datachange_notification(self, node, val, data):
       
        """This functions gets he notifications when data is changed"""
        sdt = data.monitored_item.Value.SourceTimestamp.strftime("%Y-%m-%d %H:%M:%S:%f")[:-3]
        row = {"RECORDED_DTTM": sdt, "POINT_ID": str(node).split("s=")[2], "RECORDED_VALUE": val}  
        print(row)
```

AndreasHeine, 2023-03-21:
so its only if you set the publishing-intervall to 200ms right? and your on the latest version of asyncua?

can you post a debug log of the connecting? containing the whole connect and subscribing!

AndreasHeine, 2023-03-21:
can you post it as text? hard to read the screenshots and i cant use search to find the keywords...

venkatp-co, 2023-03-21:
Client is dying in seconds ~30secs.

I will check subscription settings.

venkatp-co, 2023-03-21:
I am using default subscription settings.

venkatp-co, 2023-03-21:
I tried 1000ms, it's dying immediately. Anything more than 100ms, program is dying immediately.

AndreasHeine, 2023-03-21:
a common issue of subscriptions dieing is if the settings are not according to:
SessionTimout > (MaxKeepAlive * PublishingInterval)

from your log:

## Session:
RequestedSessionTimeout=3600000
RevisedSessionTimeout=3600000.0
so the session should last 1 hour (requested = revised so that's fine)

## Subscription:
RequestedPublishingInterval=100
RequestedLifetimeCount=10000
RequestedMaxKeepAliveCount=27000

unfortunately there is no Response in the log so i don't know the revised params...

## MonitoredItems:
RequestedSamplingInterval=0.0 -> The value 0 indicates that the Server should use the fastest practical rate.
RevisedamplingInterval=100.0 -> means the server samples with 100ms
Queuesize=0 -> means unlimited

so that all means 27000 * 100 ms = 2 700 000 ms -> 2700 s -> 45min (means that there can be a <45min gap with no notifications before the subscription got terminated from client side)
which is smaller as 60min RevisedSessionTimeout
should be fine... 

according to the RequestedLifetimeCount:
it can be that there are to many changes (+10000 according to the RequestedLifetimeCount) within the 100ms interval so the LifeTimeCounter terminates the Subscription serverside...
that would explain why the the longer publishinterval terminates faster (immediately)

the only thing you can try is to increase the RequestedLifetimeCount to the same as RequestedMaxKeepAliveCount and test if the time (~30s) is still the same!?

can you check that?

venkatp-co, 2023-03-21:
sure, i will check it. where can i increase RequestedLifetimeCount?

AndreasHeine, 2023-03-21:
client.py -> create_subscription
params.RequestedLifetimeCount = 10000

is probably the fastest way to test

venkatp-co, 2023-03-21:
I see it is already 10000, i tried changing it to 27000 but still same issue.

[image]

venkatp-co, 2023-03-21:
data changes are around ~150 for 100ms. I subscribed to only 1 node which has 10 variables, each variable changes ~15 times per 100ms. I will try to modify the code as per your suggestions.

schroeder-, 2023-03-23:
``` Python
import asyncio

from asyncua import ua

from asyncua import Client

import requests

import json

import logging

logging.basicConfig(level=logging.DEBUG)
_logger = logging.getLogger('asyncua')

class SubHandler(object):
    """Subscription Handler. To receive events from server for a subscription"""
    
    def datachange_notification(self, node, val, data):
        
        """This functions gets he notifications when data is changed"""
        sdt = data.monitored_item.Value.SourceTimestamp.strftime("%Y-%m-%d %H:%M:%S:%f")[:-3]
        row = {"RECORDED_DTTM": sdt, "POINT_ID": str(node).split("s=")[2], "RECORDED_VALUE": val}  
        print(row)
        

async def main():
    """This is the main function for the client"""
    url = "opc.tcp://admin@localhost:4849"
    async with Client(url=url) as client:

        print("Root node is: %r", client.nodes.root)
        print("Objects node is: %r", client.nodes.objects)

        print("Children of root are: %r", await client.nodes.root.get_children())
        uri = "Softsensor_OPC_Server_2"
        idx = await client.get_namespace_index(uri)
        print("idx",idx)
        myvar = await client.get_node("ns=2;i=1").get_children()
        # myvar_predicted = await client.get_node("ns=3;i=1").get_children()

        handler = SubHandler()
        sub = await client.create_subscription(100, handler)
        handle = await sub.subscribe_data_change(myvar)
        print("handle",handle)
        await asyncio.sleep(0.1)

        # we can also subscribe to events from server
        await sub.subscribe_events()
         # !!!!! Ident here otherwise you disconnect
        while True:        # Ident here otherwise you disconnect
             await asyncio.sleep(0.01)

if __name__ == "__main__":
    # logging.basicConfig(level=logging.INFO)
    asyncio.run(main()) 
```

If the code you posted is rly used, then disconnect your self. I correct the posted example and added an ident to the sleep loop!

venkatp-co, 2023-03-23:
I was able to solve the issue by implementing queues but getting a new issue.

```
import asyncio
import requests
import json
from asyncua import Client, Node, ua
from asyncua.common.subscription import DataChangeNotif, SubHandler
import httpx
ENDPOINT = "opc.tcp://127.0.0.1:4840"
NAMESPACE = "Softsensor_OPC_Server_5"

file = open("file.txt","w")
class MyHandler(SubHandler):
    def __init__(self):
        self._queue = asyncio.Queue()

    def datachange_notification(self, node: Node, value, data: DataChangeNotif) -> None:
        self._queue.put_nowait([node, value, data])
        print(f'Data change notification was received and queued.')

    async def process(self) -> None:
        csv_url = "http://localhost:3001/pushprocesseddata"
        try:
            while True:
                # Get data in a queue.
                [node, value, data] = self._queue.get_nowait()
                #path = await node.get_path(as_string=True)
                sdt = data.monitored_item.Value.SourceTimestamp.strftime("%Y-%m-%d %H:%M:%S:%f")[:-3]
                row = {"RECORDED_DTTM": sdt, "POINT_ID": str(node).split("s=")[2], "RECORDED_VALUE": value}  
                print(row)
                with open("file.txt","a") as f:
                    f.write("True"+"\n")
                async with httpx.AsyncClient() as client:
                    r = await client.post(csv_url, data=json.dumps(row))
                    print(r.text)
                #print("New data change event", node, value)
                # r = requests.post(csv_url, data=json.dumps(row))
                # print(r)

        except asyncio.QueueEmpty:
            pass

async def main() -> None:
    handler = MyHandler()
    while True:
        try:
            async with Client(url=ENDPOINT) as client:
                # Get a variable node.
                idx = await client.get_namespace_index(NAMESPACE)
                node = await client.get_node("ns=2;i=1").get_children()

                # Subscribe data change.
                
                subscription = await client.create_subscription(period=0, handler=handler)
                await subscription.subscribe_data_change(node)

                # Process data change every 100ms
                while True:
                    await asyncio.sleep(1)
                    await client.check_connection() 
                    await handler.process()
        except (ConnectionError, ua.UaError):
            await asyncio.sleep(2)
            

if __name__ == '__main__':
    asyncio.run(main())

```
[... 44 lines cut ...]
    await self.pre_request_hook()
  File "/home/venkat/.local/lib/python3.8/site-packages/asyncua/client/client.py", line 443, in check_connection
    await self._monitor_server_task
  File "/home/venkat/.local/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 566, in _publish_loop
    response = await self.publish([ack] if ack else [])
  File "/home/venkat/.local/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 549, in publish
    data = await self.protocol.send_request(request, timeout=0)
  File "/home/venkat/.local/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 158, in send_request
    await self.pre_request_hook()
  File "/home/venkat/.local/lib/python3.8/site-packages/asyncua/client/client.py", line 443, in check_connection
    await self._monitor_server_task
  File "/home/venkat/.local/lib/python3.8/site-packages/asyncua/client/client.py", line 457, in _monitor_server_loop
    _ = await self.nodes.server_state.read_value()
  File "/home/venkat/.local/lib/python3.8/site-packages/asyncua/common/node.py", line 179, in read_value
    result = await self.read_data_value()
  File "/home/venkat/.local/lib/python3.8/site-packages/asyncua/common/node.py", line 190, in read_data_value
    return await self.read_attribute(ua.AttributeIds.Value, None, raise_on_bad_status)
  File "/home/venkat/.local/lib/python3.8/site-packages/asyncua/common/node.py", line 304, in read_attribute
    result = await self.session.read(params)
  File "/home/venkat/.local/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 397, in read
    data = await self.protocol.send_request(request)
  File "/home/venkat/.local/lib/python3.8/site-packages/asyncua/client/ua_client.py", line 160, in send_request
    data = await asyncio.wait_for(self._send_request(request, timeout, message_type), timeout if timeout else None)
  File "/usr/lib/python3.8/asyncio/tasks.py", line 501, in wait_for
    raise exceptions.TimeoutError()