FreeOpcUa/opcua-asyncio #1256: Error in xmlimporter.py? Can not import the xml file of the Scales Companion Specification

MM-MII, 2023-03-23:
Hi, 
I am currently facing an issue with my opc ua server. I am trying to import the xml file of several Companion Specifications, as my information model relies on DataTypes that are specified within these specifications. 
Loading the xml File of DI, Machinery, PackML and IA seems to work fine. Loading the Scales nodeset does not work - my server script and the error message follows. 
All Nodesets come from the OPCF Github: https://github.com/OPCFoundation/UA-Nodeset

Server:
import asyncio
from asyncua import ua, Server
import os

# Define the path to the XML file containing the information model
endpoint_url = "opc.tcp://localhost:4840/"
DI_path = os.path.abspath("C:/Users/X/Documents/Demonstrator_Software/Server/Opc.Ua.Di.NodeSet2.xml")
PackML_path = os.path.abspath("C:/Users/X/Documents/Demonstrator_Software/Server/Opc.Ua.PackML.NodeSet2.xml")
IA_path = os.path.abspath("C:/Users/X/Documents/Demonstrator_Software/Server/Opc.Ua.IA.NodeSet2.xml")
machinery_path = os.path.abspath("C:/Users/X/Documents/Demonstrator_Software/Server/Opc.Ua.Machinery.NodeSet2.xml")
scales_path = os.path.abspath("C:/Users/X/Documents/Demonstrator_Software/Server/Opc.Ua.Scales.NodeSet2.xml")
#nodeset_path = os.path.abspath("C:/Users/X/Documents/Demonstrator_Software/Server/opcuacs_demonstrator.xml")

async def main():
    # Initialize the OPC UA server
    server = Server()
    await server.init()
    server.set_endpoint(endpoint_url)
    server.set_server_name("Test Server")

    server.set_security_policy(
        [
            ua.SecurityPolicyType.NoSecurity,
            #ua.SecurityPolicyType.Basic256Sha256_SignAndEncrypt,
            #ua.SecurityPolicyType.Basic256Sha256_Sign,
        ]
    )

    # Load the information model from the XML file
      
    await server.import_xml(DI_path)
    print("DI loaded")
    await server.import_xml(PackML_path)
    print("PackML loaded")
    await server.import_xml(IA_path)
    print("IA loaded")
    await server.import_xml(machinery_path)
    print("Machinery loaded")
    await server.import_xml(scales_path)
    print("Scales loaded")

    print("Server started")
    
    async with server:
        while True:
            await asyncio.sleep(1)

if __name__ == "__main__":
    asyncio.run(main())

Error/Terminal output: 

TESTSERVER_demonstrator_2.py
qn: QualifiedName(NamespaceIndex=1, Name='Lock')
qn: QualifiedName(NamespaceIndex=1, Name='SoftwareUpdate')
NodeData(nodeid:NodeId(Identifier=333, NamespaceIndex=2, NodeIdType=<NodeIdType.Numeric: 2>)) has datatypedefinition and path [Node(NumericNodeId(Identifier=84, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>)), Node(NumericNodeId(Identifier=86, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>)), Node(NodeId(Identifier=90, NamespaceIndex=0, NodeIdType=<NodeIdType.TwoByte: 0>)), Node(NumericNodeId(Identifier=24, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>)), Node(NumericNodeId(Identifier=26, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>)), Node(NumericNodeId(Identifier=28, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>)), Node(NodeId(Identifier=7, NamespaceIndex=0, NodeIdType=<NodeIdType.Numeric: 2>))] but we could not find out if this is a struct
DI loaded
PackML loaded
IA loaded
qn: QualifiedName(NamespaceIndex=2, Name='Identification')
qn: QualifiedName(NamespaceIndex=2, Name='Identification')
qn: QualifiedName(NamespaceIndex=1, Name='Components')
qn: QualifiedName(NamespaceIndex=1, Name='MachineryItemState')
qn: QualifiedName(NamespaceIndex=1, Name='MachineryOperationMode')
Machinery loaded
failure adding node NodeData(nodeid:NodeId(Identifier=53, NamespaceIndex=6, NodeIdType=<NodeIdType.Numeric: 2>)) object of type 'ExtObj' has no len()
Traceback (most recent call last):
  File "c:\Users\X\Documents\Demonstrator_Software\Server\TESTSERVER_demonstrator_2.py", line 50, in <module>
unners.py", line 44, in run
    return loop.run_until_complete(main)  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.10_3.10.2800.0_x64__qbz5n2kfra8p0\lib\asyncio\base_events.py", line 649, in run_until_complete
    return future.result()  File "c:\Users\X\Documents\Demonstrator_Software\Server\TESTSERVER_demonstrator_2.py", line 39, in main
    await server.import_xml(scales_path)
  File "C:\Users\X\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\asyncua\server\server.py", line 616, in import_xml
    return await importer.import_xml(path, xmlstring)  File "C:\Users\X\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\asyncua\common\xmlimporter.py", line 138, in import_xml
    node = await self._add_node_data(nodedata, no_namespace_migration=True)  File "C:\Users\X\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\asyncua\common\xmlimporter.py", line 234, in _add_node_data
    node = await self.add_variable_type(nodedata, no_namespace_migration)  File "C:\Users\X\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.10_qbz5n2kfra8p0\LocalCache\local-packages\Python310\site-packages\asyncua\common\xmlimporter.py", line 512, in add_variable_type
    if obj.value and len(obj.value) == 1:TypeError: object of type 'ExtObj' has no len()

I hope i will be able to solve this as i am trying to get this to work before April. Thank you all in Advance.

schroeder-, 2023-03-23:
Our nodeset import is not complete and has a lot of problems with extension objects in nodesets. If you are not on the current master you could try to update.
In the future I will look into it, but no promises here. So if you need it before april, you have to dig in the xmlimport your self.

schroeder-, 2023-03-26:
Fixed in #1259

closed 2023-03-31 by schroeder-