FreeOpcUa/opcua-asyncio #1267: Subscription loop is not stopped when LifetimeCount is reached

vitrex, 2023-03-29:
There is the theoretical risk of an endless loop and memory leakage, when the socket is still open, but no more publish requests are received:

When lifetime limit is reached, the loop is not stopped. That leads to changes still queuing up and consuming more and more memory.
In python-opcua the loop was stopped, but changes were still queuing up (https://github.com/FreeOpcUa/python-opcua/blob/master/opcua/server/internal_subscription.py#L313)

If PublishingInterval is 0, trigger_statuschange even leads to an endless recursion.

According to the spec (https://reference.opcfoundation.org/Core/Part4/v105/docs/5.13):
> When this counter reaches the value calculated for the lifetime of a Subscription based on the MaxKeepAliveCount parameter in the CreateSubscription Service (5.13.2), the Subscription is closed.

cziebuhr, 2023-03-29:
This issue is unrelated to #1265, but has been found during testing.

Disclaimer: I'm working together with @vitrex on a project.

AndreasHeine, 2023-06-02:
i almost forgot that one...

https://github.com/FreeOpcUa/opcua-asyncio/blob/244f08ef48d199f19365d34f53247f012ee8f1c4/asyncua/server/internal_subscription.py#L113

not sure if this is the correct implementation of the "LifetimeCount", unfortunately the naming is not really spec. compliant ("_publish_cycles_count")

AndreasHeine, 2023-06-02:
@vitrex can you give me some more infos?

its high freq. data? and probably a slow publishing interval? and you do use a infinite queue?

chrisjbremner, 2024-06-12:
I believe this should be fixed in https://github.com/FreeOpcUa/opcua-asyncio/pull/1654, will close this issue, but you should re-open if it does not help.

closed 2024-06-12 by chrisjbremner