FreeOpcUa/opcua-asyncio #1281: open_secure_channel fails with TimeoutError with stripped credentials

rth, 2023-04-11:
In v0.9.96  https://github.com/FreeOpcUa/opcua-asyncio/pull/1005 introduced,

> Strip credentials from server_url afterwards, preventing them to be sent unencrypted in e.g. send_hello.

corresponding to [this line](https://github.com/FreeOpcUa/opcua-asyncio/blob/master/asyncua/client/client.py#L61). The problem with this change creating a connection to a [Ignition's OPCUA server](https://docs.inductiveautomation.com/display/DOC81/Ignition%27s+OPC+UA+Server) in `open_secure_channel` fails with a TimeoutError.

<details>

```py
Traceback (most recent call last):
  File "example.py", line 65, in main
    loop.run_until_complete(task(loop))
  File "example.py", line 649, in run_until_complete
    return future.result()
  File "example.py", line 39, in task
    async with client:
  File "/app/opcua-asyncio/asyncua/client/client.py", line 88, in __aenter__
    await self.connect()
  File "/app/opcua-asyncio/asyncua/client/client.py", line 287, in connect
    await self.open_secure_channel()
  File "/app/opcua-asyncio/asyncua/client/client.py", line 374, in open_secure_channel
    result = await self.uaclient.open_secure_channel(params)
  File "/app/opcua-asyncio/asyncua/client/ua_client.py", line 308, in open_secure_channel
    return await self.protocol.open_secure_channel(params)
  File "/app/opcua-asyncio/asyncua/client/ua_client.py", line 225, in open_secure_channel
    await asyncio.wait_for(self._send_request(request, message_type=ua.MessageType.SecureOpen), self.timeout)
  File "/usr/local/lib/python3.10/asyncio/tasks.py", line 458, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
```
</details>

If I remove that line, the connections works with `master` branch. With that line, I get a TimeoutError for any version later than v0.9.96 which introduced this change (including master)

If it helps debugging the `params` sent to open_secure_channel look as follows,
```
OpenSecureChannelParameters(ClientProtocolVersion=0, RequestType=<SecurityTokenRequestType.Issue: 0>, SecurityMode=<MessageSecurityMode.SignAndEncrypt: 3>, ClientNonce=b'<some-binary-string>', RequestedLifetime=3600000)
```

While I cannot share a workable minimal example, the connection configuration looks something like the following,

<details>

```py
import sys

from asyncua import Client, ua
from asyncua.crypto.security_policies import SecurityPolicyBasic256Sha256

async def task(loop):
    url = "opc.tcp://<user>:<password>@<host>:<port>/discovery"
    client = Client(url=url, timeout=10)
    await client.set_security(
        SecurityPolicyBasic256Sha256,
        certificate=cert,
        private_key=private_key,
        mode=ua.MessageSecurityMode.SignAndEncrypt,
        server_certificate=server_cert,
    )
    client.application_uri = "<some-urn>"

    async with client:
        ...    

def main():
    loop = asyncio.get_event_loop()
    loop.set_debug(True)
    loop.run_until_complete(task(loop))
    loop.close()
```

</details>

So my request is either to remove that line or allow disabling it via for instance an extra parameter `Client(..., strip_credentials=False)`.

**Version**
Python-Version: 3.10
opcua-asyncio Version: master, and any version >= v0.9.96 

cc @cziebuhr @oroulet

rth, 2023-05-02:
I can make a PR. I would just need a confirmation about the way you would prefer to handle this.

schroeder-, 2023-05-02:
You can create a pr with extra parameters. Because sending the credentials unecrypted per default, doens't sound right to me.

closed 2023-09-13 by oroulet