FreeOpcUa/opcua-asyncio #1283: python OPCUA  or  opcua-asyncio package installation failed in Linux ARM32V7

rowshan, 2023-04-14:
### Please read, before you post!

This is a _**BUG REPORT for issues in the existing code**_. 

If you have general questions, code handling problems or ideas, please use the:

Discussionsboard: https://github.com/FreeOpcUa/opcua-asyncio/discussions <br/> or <br />
Gitter-Channel: https://gitter.im/FreeOpcUa/opcua-asyncio

---

To save some time, please provide us following informations, if possible:

**Describe the bug** <br />
I have opened the similar ticket for [python-opcua](https://github.com/FreeOpcUa/python-opcua/issues/1516).
I am advised to use the opcua-asyncio. but same problem i cannot make it work also for opcua-asyncio since I am trying to build the image in arm32v7 docker container. but i found the error in installing cryptography for opcua-asyncio development for  arm32v7/python3.8 base image but it does not work .
 

**To Reproduce**<br />
here is my docker file: 
`FROM  arm32v7/ubuntu:20.04
WORKDIR /sinewave
RUN apt-get update \
    && DEBIAN_FRONTEND=noninteractive apt-get install -y --no-install-recommends \
        software-properties-common \
    && add-apt-repository -y ppa:deadsnakes \
    && DEBIAN_FRONTEND=noninteractive apt-get install -y --no-install-recommends \
        python3.8-venv \
    && apt-get clean \
    && rm -rf /var/lib/apt/lists/*
RUN python3.8 -m venv /venv
ENV PATH=/venv/bin:$PATH
#RUN apt-get install -y python3.8
RUN pip3 install asyncua

CMD [ "python3", "client.py" ]`

**Screenshots**<br />
[image]

**Version**<br />
Python-Version:<br />
opcua-asyncio Version (e.g. master branch, 0.9):

schroeder-, 2023-04-14:
This has nothing todo with your library, if you can't install a library please contact the library which is causing the error.

closed 2023-04-14 by schroeder-

rowshan, 2023-04-14:
Ok thanks. But here the problem is with opcua-asyncio package installation you can see in the error log:
[image]
here in order to work with  opcua-asyncio I have to install  **asyncua** package which is failing.

oroulet, 2023-04-14:
looks like it is cryptograpjy package that has some issue on arm. That is strange.
Try installaing cryptography directly and tell us what is happening. if it fails you need to report the bug to the cryptography people

oroulet, 2023-04-14:
also trey installa cffi directly and let us know if it works

rowshan, 2023-04-14:
Finally I made it work inside docker using arm32v7/debian base image.  My Docker file is as following:
`FROM  arm32v7/debian:buster-slim 
WORKDIR /sinewave
RUN apt-get update && apt-get install -y python3-pip python3-venv curl libxslt1.1  
RUN pip3 install --upgrade pip
RUN curl https://sh.rustup.rs -sSf | sh -s -- --default-toolchain stable -y PATH="/root/.cargo/bin:${PATH}"
RUN \
    if [ `dpkg --print-architecture` = "armhf" ]; then \
    printf "[global]\nextra-index-url=https://www.piwheels.org/simple\n" > /etc/pip.conf ; \
    fi
RUN pip3 install cffi==1.13.1 cryptography==2.1.4 
RUN python3 -m venv /venv
ENV PATH=/venv/bin:$PATH
ENV CRYPTOGRAPHY_DONT_BUILD_RUST=1
RUN pip3 install paramiko
RUN pip3 install asyncua
ENTRYPOINT [ "python3", "client.py" ]
`
I had to downgrade libxslt to libxslt1.1 and cffi==1.13.1 cryptography==2.1.4 . it works for python-opcua as well as opcua-asyncio. 
Thanks for your support.  :)