FreeOpcUa/opcua-asyncio #1284: encryption:User does not have permission

growgreat77, 2023-04-14:
I am using server-with-encryption.py and client-with-encryption.py for testing. First, I generated certificates and keys using generate_certificate.sh and only modified the IP addresses and .cer，.pemin these three files while leaving everything else unchanged.

However, when I ran the client, the server displayed the message "name='test_name') attempted to do something they are not permitted to do". So, I added a statement to client.py: client.set_user("test_name"), but it didn't work. What am I doing wrong? I read that the server-side does not support user and passwd.
the server info is:
`INFO:asyncua.server.internal_session:activate session
INFO:asyncua.server.internal_session:Activated internal session ('192.168.40.133', 44398) for user User(role=<UserRole.User: 3>, name='test_name')
INFO:asyncua.server.uaprocessor:translate browsepaths to nodeids request (User(role=<UserRole.User: 3>, name='test_name'))
INFO:asyncua.server.uaprocessor:Read request (User(role=<UserRole.User: 3>, name='test_name'))
WARNING:asyncua.server.uaprocessor:User(role=<UserRole.User: 3>, name='test_name') attempted to do something they are not permitted to do
INFO:asyncua.server.binary_server_asyncio:processor returned False, we close connection from ('192.168.40.133', 44398)
INFO:asyncua.server.binary_server_asyncio:Lost connection from ('192.168.40.133', 44398), None
INFO:asyncua.server.uaprocessor:Cleanup client connection: ('192.168.40.133', 44398)
`
the client info is :
`    result = await self.session.write(params)
  File "/home/lss/demo/python/opcua-asyncio/examples/../asyncua/client/ua_client.py", line 407, in write
    data = await self.protocol.send_request(request)
  File "/home/lss/demo/python/opcua-asyncio/examples/../asyncua/client/ua_client.py", line 165, in send_request
    self.check_answer(data, f" in response to {request.__class__.__name__}")
  File "/home/lss/demo/python/opcua-asyncio/examples/../asyncua/client/ua_client.py", line 174, in check_answer
    hdr.ServiceResult.check()
  File "/home/lss/demo/python/opcua-asyncio/examples/../asyncua/ua/uatypes.py", line 372, in check
    raise UaStatusCodeError(self.value)
asyncua.ua.uaerrors._auto.BadUserAccessDenied: "User does not have permission to perform the requested operation."(BadUserAccessDenied)
`

schroeder-, 2023-04-14:
You need the matching cert for authentication. There are two ways currently implemented either use username + password. This not in the example. Or the username + certificate. This certificate is not the client certificate! To set the authentication certificate use

`await server.set_security(..., server_certificate=auth_cert)`

growgreat77, 2023-04-15:
> 

thank your for your help.

And the code should be added to examples/server-with-encryption.py like this ?
```await server.set_security(..., server_certificate="my_cert.der")```
so The updated code for examples/server-with-encryption.py is as follows:
```    server.set_endpoint("opc.tcp://192.168.40.133:4840/freeopcua/server/")
    server.set_security_policy([ua.SecurityPolicyType.Basic256Sha256_SignAndEncrypt],
                               permission_ruleset=SimpleRoleRuleset())
    # load server certificate and private key. This enables endpoints
    # with signing and encryption.
    await server.set_security(..., server_certificate="my_cert.der")

    await server.load_certificate("my_cert.der")
    await server.load_private_key("my_private_key.pem")```

However, the server may output give an error that the function set_security() is not available:
```    await server.set_security(..., server_certificate="my_cert.der")
AttributeError: 'Server' object has no attribute 'set_security'. Did you mean: 'set_security_IDs' ```

schroeder-, 2023-04-15:
I always forget that our CertificateUserManager is something really strange and none standard. 
So forget everything. You don't have to set client.set_user("test_name"). But your client certificate must match the **test_name** certificate!

closed 2023-05-11 by schroeder-

mbuzoli, 2023-10-21:
> I always forget that our CertificateUserManager is something really strange and none standard. So forget everything. You don't have to set client.set_user("test_name"). But your client certificate must match the **test_name** certificate!

Sorry for my ignorance, but how do I make sure the client certificate matches the 'name="test_name"'? Is this info supposed to be declared in the client somewhere? Script? Certificate?