FreeOpcUa/opcua-asyncio #1286: How to get type of subscribed node?

unregistered33, 2023-04-17:
Hello!  Not issue, just question.
How to get the value ua type in  `datachange_notification(self, node: Node, val, data)` (int, float, bool etc)? I need to sort the monitored items received depending on their ua type
Thanks a lot!

schroeder-, 2023-04-17:
There are multiple ways. 
Either you read the datatype before hand and provide it in your SubscriptionHandlerInstance.

You can check the variant type in the `datachange_notification`.
``` python
var: ua.Variant = data.subscription_data.Value.Value 
type: ua.VariantType = var.VariantType
```

You can use the python way: 
``` Python
t = type(val)
if t == str:
    ....
elif t == bool:
   ....
```

closed 2023-04-18 by schroeder-

unregistered33, 2023-04-18:
Thanks, Alexander!

unregistered33, 2023-04-18:
a small clarification
var: ua.Variant = data.subscription_data.Value.Value 
need var: ua.Variant = data.**monitored_item**.Value.Value