FreeOpcUa/opcua-asyncio #1294: subscribe_alarms_and_conditions fails at CreateMonitoredItemResponse with BadMonitoredItemFilterInvalid

ts4iot, 2023-04-27:
I have two different versions of  Simatic WinCC server in operation. 
With the newer version the capturing of alarms and conditions works fine:

```
updating client limits to: TransportLimits(max_recv_buffer=65536, max_send_buffer=65536, max_chunk_count=0, max_message_size=16777216)
updating client limits to: TransportLimits(max_recv_buffer=65536, max_send_buffer=65536, max_chunk_count=0, max_message_size=16777216)
updating client limits to: TransportLimits(max_recv_buffer=65536, max_send_buffer=65536, max_chunk_count=0, max_message_size=16777216)
Revised values returned differ from subscription values: CreateSubscriptionResult(SubscriptionId=16, RevisedPublishingInterval=1000.0, RevisedLifetimeCount=120, RevisedMaxKeepAliveCount=360)
```

With the older version the capturing fails:

```
updating client limits to: TransportLimits(max_recv_buffer=65536, max_send_buffer=65536, max_chunk_count=0, max_message_size=0)
Revised values returned differ from subscription values: CreateSubscriptionResult(SubscriptionId=24, RevisedPublishingInterval=1000.0, RevisedLifetimeCount=120, RevisedMaxKeepAliveCount=360)
Error in watchdog loop
```

I checked the opc ua protocol of the subscription process with Wireshark. The problem occures  in the message CreateMonitoredItemResponse with the error code: StatusCode: 0x80430000 [BadMonitoredItemFilterInvalid]

The capturing program is exactly the same, only the server IP is different.
I don't want to update the WinCC server to a newer version. 

What can I do to find the root cause of the problem?

ts4iot, 2023-05-04:
Thanks, your suggestion solved my problem.

ts4iot, 2023-05-05:
After setting the filter manually the subscription seems to work correct, but I see the following error in the console log. 

```
Exception while calling user callback: %s
Traceback (most recent call last):
  File "C:\Daten\Python\Python310\lib\site-packages\asyncua\client\ua_client.py", line 603, in _publish_loop
    await callback(response.Parameters)
  File "C:\Daten\Python\Python310\lib\site-packages\asyncua\common\subscription.py", line 119, in publish_callback
    await self._call_event(notif)
  File "C:\Daten\Python\Python310\lib\site-packages\asyncua\common\subscription.py", line 157, in _call_event
    data = self._monitored_items[event.ClientHandle]
KeyError: 0
```

Can I ignore this exeption or is it critical?

schroeder-, 2023-05-08:
No client should send a ClientHandle of value zero. 

From the spec:
This primitive data type is a UInt32 that is used as an identifier, such as a handle. All values, except for 0, are valid.

If your aplication is working correctly, this should be fine.

Maybe we should add this to **_call_event**, so only a log message is generated:

```
      if item.ClientHandle not in self._monitored_items:
                self.logger.warning("Received a notification for unknown handle: %s", item.ClientHandle)
                continue
```

ts4iot, 2023-05-08:
The exception only occures with WinCC 07.04.20. With WinCC 07.05.20 I don't see this exception. 
My application works fine. I like your suggestion with log entry of level warning.

schroeder-, 2023-05-09:
Added the warning #1302

closed 2023-05-11 by schroeder-