FreeOpcUa/opcua-asyncio #1312: client.export_xml fails with BadAttributeIdInvalid

jmtatsch, 2023-05-19:
**Describe the bug** <br />
I have an interface on a production s7 system I would like to simulate and run tests against.
I tried to export the node layout via client.export_xml() but it fails with a not helpful error message.

**To Reproduce**<br />

Connect to a s7 and attempt to dump the node layout:

```
"""Dump the node layout from external S7."""
import asyncio
from pathlib import Path

from asyncua import Client
from asyncua.common import ua_utils
async def main():
    client = Client(url='opc.tcp://user:password@192.168.1.1:4840')
    async with client:
        idx = await client.get_namespace_index('http://test.org')

        # Warning a complete dump must not be run during production
        # target_node = client.nodes.objects
        target_node = await client.nodes.objects.get_child(["3:ServerInterfaces", "{}:test.org".format(idx)])
        nodes = await ua_utils.get_node_children(target_node)

        await client.export_xml(nodes, Path('opcua_connector/ua-export.xml'), export_values=False)

if __name__ == "__main__":
    asyncio.run(main())

```
**Expected behavior**<br />

I would expect the xml to be successfully written. Alternatively I would expect the xml to be written without the incompatible node. At least I would expect to get informed which node/attribute caused the issue.

**Screenshots**<br />

```
"The attribute is not supported for the specified Node."(BadAttributeIdInvalid)
  File "/home/jtatsch/.local/lib/python3.10/site-packages/asyncua/ua/uatypes.py", line 328, in check
    raise UaStatusCodeError(self.value)
  File "/home/jtatsch/.local/lib/python3.10/site-packages/asyncua/common/node.py", line 306, in read_attribute
    result[0].StatusCode.check()
  File "/home/jtatsch/.local/lib/python3.10/site-packages/asyncua/common/xmlexporter.py", line 271, in add_etree_variable
    var = await node.read_attribute(ua.AttributeIds.MinimumSamplingInterval)
  File "/home/jtatsch/.local/lib/python3.10/site-packages/asyncua/common/xmlexporter.py", line 160, in node_to_etree
    await self.add_etree_variable(node)
  File "/home/jtatsch/.local/lib/python3.10/site-packages/asyncua/common/xmlexporter.py", line 72, in build_etree
    await self.node_to_etree(node)
  File "/home/jtatsch/.local/lib/python3.10/site-packages/asyncua/client/client.py", line 736, in export_xml
    await exp.build_etree(nodes)
  File "/opt/opcua_connector/dump_xml.py", line 18, in main
    await client.export_xml(nodes, Path('opcua_connector/ua-export.xml'), export_values=False)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/opt/opcua_connector/dump_xml.py", line 21, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main (Current frame)
    return _run_code(code, main_globals, None,
asyncua.ua.uaerrors._auto.BadAttributeIdInvalid: "The attribute is not supported for the specified Node."(BadAttributeIdInvalid)

```
**Version**<br />
Python-Version: 3.10.6<br />
opcua-asyncio Version (e.g. master branch, 0.9): asyncua-1.0.2

der-joel, 2023-05-31:
Your code fails when trying to read the `MinimumSamplingInterval` attribute. According to the [docs](https://reference.opcfoundation.org/Core/Part4/v105/docs/5.12.1.2):
> If the behaviour by which the underlying system updates the item is known, it will be available via the [MinimumSamplingInterval Attribute](https://reference.opcfoundation.org/search/18?t=MinimumSamplingInterval%20Attribute) defined in [OPC 10000-3](https://reference.opcfoundation.org/Core/Part4/v105/docs/?r=UAPart3).

Please check if your server specifies this attribute. If not `BadAttributeIdInvalid` is raised.

jmtatsch, 2023-06-01:
Hello Joel,

I found only one node i=6010 that doesn't have a MinimumSamplingInterval:

[image]

It belongs to an icon that seems to be autoadded if one creates a new namespace with the s7 management software.
All others in the namespace have -1 as MinimumSamplingInterval.

Can this maybe just throw a warning instead of quitting?

jmtatsch, 2023-06-07:
Unfortunately this icon cannot be removed from the s7 management software side :(

jmtatsch, 2023-06-12:
would it suffice to just patch line 271 of xmlexporter as follows?
`var = await node.read_attribute(ua.AttributeIds.MinimumSamplingInterval, raise_on_bad_status=False)`

jmtatsch, 2023-07-23:
there is another one in  common/node.py  that needs patching for Siemens 

```
async def read_node_class(self):
        """
        get node class attribute of node
        """
        result = await self.read_attribute(ua.AttributeIds.NodeClass)
        return ua.NodeClass(result.Value.Value)
```

commit 1f8e88f7 (2023-07-24): Dont die on missing MinimumSamplingInterval Fixes #1312
closed 2023-07-24 by oroulet