FreeOpcUa/opcua-asyncio #1313: Server shelf_file not working

harriv, 2023-05-22:
**Describe the bug** <br />
A clear and concise description of what the bug is.

When using the shelf_file in server init, an exception is raised:

> Traceback (most recent call last):
>   File "test_server.py", line 20, in <module>
>     asyncio.run(main(), debug=True)
>   File "Python311\Lib\asyncio\runners.py", line 190, in run
>     return runner.run(main)
>            ^^^^^^^^^^^^^^^^
>   File "Python311\Lib\asyncio\runners.py", line 118, in run
>     return self._loop.run_until_complete(task)
>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "Python311\Lib\asyncio\base_events.py", line 653, in run_until_complete
>     return future.result()
>            ^^^^^^^^^^^^^^^
>   File "test_server.py", line 9, in main
>     await server.init(shelf_file="test")
>   File "opc_env\Lib\site-packages\asyncua\server\server.py", line 122, in init
>     await self.iserver.init(shelf_file)
>   File "opc_env\Lib\site-packages\asyncua\server\internal_server.py", line 80, in init
>     await self.load_standard_address_space(shelffile)
>   File "opc_env\Lib\site-packages\asyncua\server\internal_server.py", line 122, in load_standard_address_space   
>     is_file = await asyncio.get_running_loop().run_in_executor(
>               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "Python311\Lib\concurrent\futures\thread.py", line 58, in run
>     result = self.fn(*self.args, **self.kwargs)
>              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>   File "Python311\Lib\pathlib.py", line 1267, in is_file
>     return S_ISREG(self.stat().st_mode)
>                    ^^^^^^^^^
> AttributeError: 'str' object has no attribute 'stat'
> 

My code:

```
from asyncua import Server, ua
from asyncua.common.methods import uamethod
from asyncua.common.structures104 import new_enum

import asyncio

async def main():
    server = Server()    
    await server.init(shelf_file="test")
    server.set_endpoint("opc.tcp://0.0.0.0:54840/server/")
    server.set_server_name("Test server")
    uri = "http://opcua.local"
    idx = await server.register_namespace(uri)
    main = await server.nodes.objects.add_object(idx, "Main")
    await main.add_variable(ua.NodeId('TestVar', NamespaceIndex=idx), 'TestVar', False, ua.Boolean)
    async with server:
        while True:
            await asyncio.sleep(1)

asyncio.run(main(), debug=True)
```

This looks like it comes from #1179 : https://github.com/FreeOpcUa/opcua-asyncio/commit/9aad5a4d6d4c079f58c0050b97b683951edba5b3#diff-e3cf94bf54c801debe599da03a24c815720ce9b3f213a3474f145006ff352917

schroeder-, 2023-05-23:
Try this then:

`await server.init(shelf_file=Path("test"))`

AiyionPrime, 2024-02-13:
Alright. I opened two PRs, one to verify the problem and on to resolve it.
I'll undraft the latter once the former is merged.

This will then bring us to phase two of the problem:
It's not only the creation of the shelf file, which is currently not working,
but loading as well; as @rnestler pointed out in late april "22:

> [...]
> But on the second run it just hits the `NotImplementedError` when trying to load the shelf file again https://github.com/FreeOpcUa/opcua-asyncio/blob/f2f7722d945dbbc04ab20a87f6aff17496f7edf1/asyncua/server/address_space.py#L718

_Originally posted by @rnestler in https://github.com/FreeOpcUa/opcua-asyncio/issues/867#issuecomment-1109872414_

Running git blame against said line yields us the commit

a85b89ae8 `[ADD] wip` by @cbergmiller, if I'm not mistaken.

Does someone have an idea what his intention was?
Added below the `NotimplementedError` is the todo:

> \# ToDo: async friendly implementation - load all at once?