FreeOpcUa/opcua-asyncio #1319: Unknown datatype when calling load_data_type_definitions

tomj-99, 2023-05-31:
When I connect to an OPC UA server as a client and call `await client.load_data_type_definitions()`, I get this error:

`error:  Unknown datatype for field: StructureField(Name='MyCustomTypeArrayField', Description=LocalizedText(Locale=None, Text=None), DataType=NodeId(Identifier=3008, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>), ValueRank=1, ArrayDimensions=[0], MaxStringLength=0, IsOptional=False) in structure:MyCustomStructureType, please report`

When calling `await client.load_type_definitions() ` instead, it works fine but gives a warning `Deprecated since spec 1.04, call load_data_type_definitions`. 

The field 'MyCustomTypeArrayField' is an array of a custom data type (ns=3 i=3008, with only basic opc ua data types as fields) with an array size of 0 that is set dynamically at runtime.

[image]

The server that I connect to uses the .NET SDK v3.3.0 from UnifiedAutomation which should actually support spec 1.05.

versions:
Python 3.11.0
ayncua 1.0.2

tomj-99, 2023-05-31:
@schroeder- yes, that exists. I just had a closer look at the log and found this:

`Structure type StructureDefinition(DefaultEncodingId=NodeId(Identifier=5018, NamespaceIndex=3, NodeIdType=<NodeIdType.FourByte: 1>), BaseDataType=NodeId(Identifier=22, NamespaceIndex=0, NodeIdType=<NodeIdType.TwoByte: 0>), StructureType_=<StructureType.StructureWithSubtypedValues: 3>, Fields=[...]) not implemented` 

I shortened it a little bit, the important part seems to be `StructureType_=<StructureType.StructureWithSubtypedValues: 3>`, I just went through the code of `load_data_type_definitions` and found in `structures104.py` in the function `make_structure_code` these lines:

```
if sdef.StructureType not in (ua.StructureType.Structure, ua.StructureType.StructureWithOptionalFields, ua.StructureType.Union):
        raise NotImplementedError(f"Only StructureType implemented, not {ua.StructureType(sdef.StructureType).name} for node {struct_name} with DataTypdeDefinition {sdef}")
```

so `StructureWithSubtypedValues `does not seem to be supported

schroeder-, 2023-05-31:
I will add StructureWithSubtypedValues if i find the time. I add i to my backlog.

tomj-99, 2023-05-31:
After changing the data type of the field from Number to Double, calling `await client.load_data_type_definitions()` works fine 👍

mous16, 2024-05-10:
I follow this thread, because I've also the same need.
Yes, I know, it's an exotic feature, but I cannot control the custom types required by the server I'm connecting to.

Scheibengraber, 2024-08-29:
Dear @schroeder- I also follow this thread. Do you have already some estimation when you can implement this enhancement? I am also interested in the enhancement.