FreeOpcUa/opcua-asyncio #1322: Subscription notification not triggered by change of SourceTimestamp

der-joel, 2023-05-31:
I am trying to subscribe to data changes for multiple nodes. Currently, if the same value is published twice on the server (change of `SourceTimestamp`) `datachange_notification` is not called. According to the OPC-UA [docs](https://reference.opcfoundation.org/v105/Core/docs/Part4/7.10/) this is default behaviour. 

For my use case I want to receive a notification with the current value if the `SourceTimestamp` changes. So I tried changing the trigger option of the `MonitoringFilter` to `StatusValueTimestamp` (which should trigger notifications on timestamp change as mentioned [here](https://reference.opcfoundation.org/v105/Core/docs/Part4/7.10/)):
```python3
async def main():
    """
    Main task of this Client-Subscription example.
    """
    client = Client(url=UA_SERVER_URL)
    async with client:
        handler = SubscriptionHandler()
        # Create subscription
        subscription = await client.create_subscription(500, handler)
        # Retrieve nodes
        nodes = []
        for node in ["ns=2;i=100", "ns=2;i=101"]:
            nodes.append(client.get_node(node))
        # Subscribe data change
        mfilter = ua.DataChangeFilter(Trigger=ua.DataChangeTrigger.StatusValueTimestamp)
        await subscription._subscribe(nodes, attr=ua.AttributeIds.Value, mfilter=mfilter)
        # Run
        try:
            while True:
                _logger.info("waiting ...")
                await asyncio.sleep(30)
        except KeyboardInterrupt:
            await subscription.delete()

if __name__ == "__main__":
    asyncio.run(main())
```
However, there are still no incoming notifications if the `SourceTimestamp` of the subscribed node is changed. I used a third party client to confirm that the timestamp has changed.
I am currently using a server created with asyncua too. Maybe this is a server-side issue?

**System Specs**
Python-Version: 3.8
opcua-asyncio Version: pypi, 1.0.2

schroeder-, 2023-05-31:
The client side looks good [see](https://stackoverflow.com/questions/75995942/opcua-asyncio-subscribe-data-change-when-timestamp-changes/75996196#75996196).
On the server side #1253 should have fixed this issue. Is your server using version 1.0.2?

der-joel, 2023-06-14:
You were right. Upgrading the server from 0.9.92 to 1.0.2 fixed this. Thanks!

closed 2023-06-14 by der-joel