FreeOpcUa/opcua-asyncio #1323: Limitation for historical data read does not work

KallistoD, 2023-06-01:
FreeOpcUa / opcua-asyncio / asyncua / common / node.py / read_raw_history() has description:
> Read raw history of a node
> result code from server is checked and an exception is raised in case of error
> If numvalues is > 0 and number of events in period is > numvalues
> then result will be truncated

Expected behavior:
If non-zero value provided for numvalues then read_raw_history() should request from server and then return no more than numvalues historical values for a Node (OPC UA variable).
This option is useful to manage memory usage on client side then reading huge amount of historical data in chunks. It is no always possible to predict memory usage by setting starttime and endtime values. So, numvalues is the solution.

Actual behavior:
read_raw_history() reads and returns all the historical data regardless of numvalues option. It does not provide result truncation as declared.
Non-zero numvalues causes a series of history_read() requests instead of single request. But all of them performed within single read_raw_history() call, not allowing to control memory usage on client side with numvalues option.

Please, consider to fix this behavior. Possible solution is to break from 'while True' loop inside read_raw_history() if numvalues is non-zero. Single call of history_read() should be enough if numvalues is non-zero and passed into details.NumValuesPerNode field. Something like adding at the end of 'while True' loop the following lines:
`if numvalues > 0:`
`    break`

The above applicable to opcua-asyncio code in master branch on 1st of June, 2023.
Documentation reference: https://reference.opcfoundation.org/v104/Core/docs/Part11/6.4.3/#6.4.3.1

schroeder-, 2023-06-01:
Actually i wouldn't do that. Because it will not free the continuations points on the server. 
The problem is that the server doesn't respect NumValuesPerNode, so i would ask the server developer to fix that behaviour.
Otherwise you can use the low level client and use our custom read_raw_history()
```
def my_read_history_raw(client: Client,  starttime=None, endtime=None, numvalues=0, return_bounds=True):
        details = ua.ReadRawModifiedDetails()
        details.IsReadModified = False
        if starttime:
            details.StartTime = starttime
        else:
            details.StartTime = ua.get_win_epoch()
        if endtime:
            details.EndTime = endtime
        else:
            details.EndTime = ua.get_win_epoch()
        details.NumValuesPerNode = numvalues
        details.ReturnBounds = return_bounds
        history = []
        continuation_point = None
        while len(history) < numvalues:
            valueid = ua.HistoryReadValueId()
            valueid.NodeId = self.nodeid
            valueid.IndexRange = ''
            valueid.ContinuationPoint = continuation_point
            params = ua.HistoryReadParameters()
            params.HistoryReadDetails = details
            params.TimestampsToReturn = ua.TimestampsToReturn.Both
            params.ReleaseContinuationPoints = False
            params.NodesToRead.append(valueid)
            result = (await client.uaclient.history_read(params))[0]
            result.StatusCode.check()
            continuation_point = result.ContinuationPoint
            history.extend(result.HistoryData.DataValues)
            # No more data available
            if continuation_point is None:
                break

        return history
```

KallistoD, 2023-06-01:
Thank you for your comments and for code example.
I actually use the opcua-asyncio client to connect another server, which respects NumValuesPerNode value.
Regarding pending continuation points the [documentation](https://reference.opcfoundation.org/v104/Core/docs/Part11/6.4.3/#6.4.3.1) says:
>When a continuationPoint is returned, a [Client](https://reference.opcfoundation.org/search/45?t=Client) wanting the next numValuesPerNode values shall call ReadRaw again with the [continuationPoint](https://reference.opcfoundation.org/search/45?t=continuationPoint) set.

I read it this way: It is up to client to decide if it wants to send the next request. Otherwise, it is not clear the reason for numValuesPerNode option to exist.
Also, the server should be fool-proof anyway. Client code may have bugs, it may crash at any point, network may be broken. Those are not the reasons for the OPC UA server to leak through unused continuation points. It looks a good idea to implement continuation points expiration for such cases.

KallistoD, 2023-06-01:
To be more clear: the function read_raw_history() has parameter numvalues, which is useless in the current implementation. So, you probably want to:
- remove it,
- implement it according to function description
- or change the function description to let users know that the parameter does not work currently

schroeder-, 2023-06-01:
The parameter is only for the server if the server doesn't care you get all elments. This is how opc ua and the api is designed, if  a server doesn't behave to the standard, we do not handle this. You can always use the low level client and build a workaround as i showed you in my previous comment.

closed 2023-09-29 by schroeder-