FreeOpcUa/opcua-asyncio #1327: Trying to instansiate optionals

herculesp17, 2023-06-05:
This is how I instantiate my object in the server. so far so good. 
`
#### instantiate objects in server

        self.device = await instantiate(await self.server.nodes.objects.get_child('3:Machines'),
                                        await self.server.nodes.base_object_type.get_child('4:WwMachineType'),
                                        bname = 'testServer_SA_OPCUA',
                                        dname = ua.LocalizedText('Dickenhobel'),
                                        idx = 4, instantiate_optional = False)`

but i want to add some optional children of the node type such as variables and properties. I just don't know how to get there.

AndreasHeine, 2023-06-05:
instantiate_optional = True

herculesp17, 2023-06-05:
yes but that instatiates all optionals; i don't want all of the though. I just want some specific ones.

AndreasHeine, 2023-06-05:
thats currently not possible...

schroeder-, 2023-06-05:
Delete the unwanted ones afterwards.

herculesp17, 2023-06-05:
@schroeder- There are like 100 nodes more beeing instaniated than i actually want

herculesp17, 2023-06-05:
@AndreasHeine I agree

herculesp17, 2023-06-05:
@schroeder- Can you tell me exactly how to do that? I tried the following, but it didn't work. My programming skills are not very good, yet and it's hard for me to understand the documention perfectly.

`        self.device = await instantiate(await self.server.nodes.ob,
                                        ua.NodeIdType(3),
                                        ua.NodeId('ns=1;i=6028'),
                                        bname = '4:Location',
                                        dname = 'Biel',
                                        idx = 4, instantiate_optional = False)
`

schroeder-, 2023-06-05:
Started a pr #1328, this is missing a test.