FreeOpcUa/opcua-asyncio #1333: Session is closed after activation when the server is restarted

ljwun, 2023-06-12:
**Describe the bug** <br />
I have an experimental environment using [gopcua/opcua](https://github.com/gopcua/opcua) as the OPC UA client and [tools/uaserver](https://github.com/FreeOpcUa/opcua-asyncio/tree/master/tools/uaserver) as the server.

When the network is disconnected and reconnected or the server is restarted, the session is frantically re-established between the client and the server.

The following is a continuous logs(I split it according to the request type):

1. restore session(port not same, old authentication token is `1001`)
```
INFO:asyncua.server.binary_server_asyncio:New connection from ('172.26.192.1', 57784)
DEBUG:asyncua.server.binary_server_asyncio:_process_received_message 51 51
INFO:asyncua.uaprotocol:updating server limits to: TransportLimits(max_recv_buffer=65535, max_send_buffer=65535, max_chunk_count=1601, max_message_size=104857600)
DEBUG:asyncua.server.binary_server_asyncio:_process_received_message 120 120
DEBUG:asyncua.server.binary_server_asyncio:_process_received_message 102 102
DEBUG:asyncua.server.uaprocessor:process_message NodeId(Identifier=467, NamespaceIndex=0, NodeIdType=<NodeIdType.FourByte: 1>) RequestHeader(AuthenticationToken=NodeId(Identifier=1001, NamespaceIndex=0, NodeIdType=<NodeIdType.FourByte: 1>), Timestamp=datetime.datetime(2023, 6, 9, 2, 52, 57, 863188), RequestHandle=2, ReturnDiagnostics=0, AuditEntryId=None, TimeoutHint=10000, AdditionalHeader=ExtensionObject(TypeId=NodeId(Identifier=0, NamespaceIndex=0, NodeIdType=<NodeIdType.TwoByte: 0>), Body=None))
INFO:asyncua.server.uaprocessor:Activate session request (None)
INFO:asyncua.server.uaprocessor:request to activate non-existing session (None)
ERROR:asyncua.server.uaprocessor:sending service fault response: "The session id is not valid." (BadSessionIdInvalid)
```
2. re-create session(new auhentication token is `1002`)
```
DEBUG:asyncua.server.binary_server_asyncio:_process_received_message 254 254
DEBUG:asyncua.server.uaprocessor:process_message NodeId(Identifier=461, NamespaceIndex=0, NodeIdType=<NodeIdType.FourByte: 1>) RequestHeader(AuthenticationToken=NodeId(Identifier=0, NamespaceIndex=0, NodeIdType=<NodeIdType.TwoByte: 0>), Timestamp=datetime.datetime(2023, 6, 9, 2, 52, 57, 865479), RequestHandle=3, ReturnDiagnostics=0, AuditEntryId=None, TimeoutHint=10000, AdditionalHeader=ExtensionObject(TypeId=NodeId(Identifier=0, NamespaceIndex=0, NodeIdType=<NodeIdType.TwoByte: 0>), Body=None))
INFO:asyncua.server.uaprocessor:Create session request (None)
INFO:asyncua.server.internal_session:Created internal session ('172.26.192.1', 57784)
INFO:asyncua.server.internal_session:Create session request
INFO:asyncua.server.internal_server:get endpoint
```
3. activate session
```
DEBUG:asyncua.server.binary_server_asyncio:_process_received_message 102 102
DEBUG:asyncua.server.uaprocessor:process_message NodeId(Identifier=467, NamespaceIndex=0, NodeIdType=<NodeIdType.FourByte: 1>) RequestHeader(AuthenticationToken=NodeId(Identifier=1002, NamespaceIndex=0, NodeIdType=<NodeIdType.FourByte: 1>), Timestamp=datetime.datetime(2023, 6, 9, 2, 52, 57, 866984), RequestHandle=4, ReturnDiagnostics=0, AuditEntryId=None, TimeoutHint=10000, AdditionalHeader=ExtensionObject(TypeId=NodeId(Identifier=0, NamespaceIndex=0, NodeIdType=<NodeIdType.TwoByte: 0>), Body=None))
INFO:asyncua.server.uaprocessor:Activate session request (None)
INFO:asyncua.server.internal_session:activate session
INFO:asyncua.server.internal_session:Activated internal session ('172.26.192.1', 57784) for user User(role=<UserRole.User: 3>, name=None)
```
4. close old session(client wants to close `1001`, but server close `1002`)
```
DEBUG:asyncua.server.binary_server_asyncio:_process_received_message 48 48
DEBUG:asyncua.server.uaprocessor:process_message NodeId(Identifier=473, NamespaceIndex=0, NodeIdType=<NodeIdType.FourByte: 1>) RequestHeader(AuthenticationToken=NodeId(Identifier=1001, NamespaceIndex=0, NodeIdType=<NodeIdType.FourByte: 1>), Timestamp=datetime.datetime(2023, 6, 9, 2, 52, 57, 868298), RequestHandle=5, ReturnDiagnostics=0, AuditEntryId=None, TimeoutHint=10000, AdditionalHeader=ExtensionObject(TypeId=NodeId(Identifier=0, NamespaceIndex=0, NodeIdType=<NodeIdType.TwoByte: 0>), Body=None))
INFO:asyncua.server.uaprocessor:Close session request (None)
INFO:asyncua.server.internal_session:close session ('172.26.192.1', 57784)
INFO:asyncua.server.subscription_service:delete subscriptions: []
INFO:asyncua.server.uaprocessor:sending close session response (None)
```
5. read value with session `1002`
```
DEBUG:asyncua.server.binary_server_asyncio:_process_received_message 84 84
DEBUG:asyncua.server.uaprocessor:process_message NodeId(Identifier=631, NamespaceIndex=0, NodeIdType=<NodeIdType.FourByte: 1>) RequestHeader(AuthenticationToken=NodeId(Identifier=1002, NamespaceIndex=0, NodeIdType=<NodeIdType.FourByte: 1>), Timestamp=datetime.datetime(2023, 6, 9, 2, 52, 57, 869492), RequestHandle=6, ReturnDiagnostics=0, AuditEntryId=None, TimeoutHint=10000, AdditionalHeader=ExtensionObject(TypeId=NodeId(Identifier=0, NamespaceIndex=0, NodeIdType=<NodeIdType.TwoByte: 0>), Body=None))
INFO:asyncua.server.uaprocessor:Request service that needs a activated session (User(role=<UserRole.User: 3>, name=None))
ERROR:asyncua.server.uaprocessor:sending service fault response: "The session cannot be used because ActivateSession has not been called." (BadSessionNotActivated)
INFO:asyncua.server.binary_server_asyncio:Lost connection from ('172.26.192.1', 57784), None
INFO:asyncua.server.uaprocessor:Cleanup client connection: ('172.26.192.1', 57784)
INFO:asyncua.server.internal_session:close session ('172.26.192.1', 57784)(id=NodeId(Identifier=1002, NamespaceIndex=0, NodeIdType=<NodeIdType.FourByte: 1>))
INFO:asyncua.server.subscription_service:delete subscriptions: []
```

6. Then
```mermaid
sequenceDiagram
    participant client
    participant server
    loop infinite
        client->>server: activate session `n`
        server-->>client: BadSessionIdInvalid(non-existing session)
        client->>server: create session
        server-->>client: new session authentication token is `n+1`
        client->>server: activate session `n+1`
        server-->>client: success activate session(`n+1`)
        client->>server: close session `n`
        server-->>client: success close session(`n+1`)
        client->>server: read value using session `n+1`
        server-->>client: BadSessionNotActivated
    end
```

According to the logs, the issue is due to the wrong session being closed. It should be closed according to the authenticationToken in the request header.

I tried to avoid this issue by adding the following code to line 203 of `asyncua/server/uaprocessor.py`.
```python
if requesthdr.AuthenticationToken!=self.session.auth_token:
    _logger.info("request to close non-existing session (%s)", user)
    raise ServiceError(ua.StatusCodes.BadSessionIdInvalid)
```
I'm not sure this is a good solution, but the reconnection mechanism works.

**To Reproduce**<br />
1. run `uaserver` as server and gopcua as client(like [this](https://gist.github.com/ljwun/e728f642efea0585a4cc3dc79d623056))
2. Stop and restart `uaserver`. (Or disconnect and reconnect network)

**Expected behavior**<br />
Check authenticationToken before close session.

**Version**<br />
Python-Version: 3.8.16<br />
opcua-asyncio Version (e.g. master branch, 0.9): master branch

tomj-99, 2023-06-15:
I am currently facing the same issue, I have a C++ client that connects to my uaserver. When I restart the uaserver, my C++ SDK tries to automatically recover the connection and I get an exception. 

`ERROR:asyncua.server.uaprocessor:sending service fault response: "The session id is not valid." (BadSessionIdInvalid)`

There is no way to recover from this, I have to restart my client application.

**Version**
Python 3.11.0
asyncua 1.0.2

AndreasHeine, 2023-06-15:
if the uaserver restarts it looses all session data! so your client need to react correctly to the BadStatuscode.